grblHAL/core #175: Use of the ODrive system with the grblHAL

Tettraspere, 2022-08-20:
Hi to all!

I would like to know if it is possible to use the ODrive system with the grblHAL.

I am triyng to achive high speeds for long periods and i am worried about having issues with back EMF from the stepper motors.

Thanks.

terjeio, 2022-08-21:
[This](https://odriverobotics.com/)? If so it has a step/dir interface sou can be used.

Tettraspere, 2022-08-21:
I wasn't aware of this new board from them. The board that i had on my mind was the version 3.6, that received communication by the RX and TX signal.

I have to read a bit about this new board.

I was curious if someone had tried this hardware . It seems an interesting choise for very fast movements for a long period of time due to the way that it deals with the back EMF.

hanke-cnc, 2022-08-25:
> [This](https://odriverobotics.com/)? If so it has a step/dir interface so can be used.

> [This](https://odriverobotics.com/)? If so it has a step/dir interface so can be used.

hi, terjeio, this looks very interesting, looks like it can be used to make cheap servo motors? Makes our CNC machines more powerful, so cool!

closed 2024-04-16 by terjeio