grblHAL/core #177: M104Px doesn't seem to work when Default Spindle is PWM

jaymis, 2022-08-23:
Hey Terje. I'm batch-testing a bunch of GRBLHAL2000 boards to ship out and I've come across something which may be a bug.

I'm shipping out my boards with `Default Spindle: PWM` so new users don't have their boards come up with Alarm:14 when they're just getting started.

I'm testing RS485 spindle control though, so I'm changing to VFD spindle, making sure the RS485 communication works with my VFD, then changing back to PWM spindle control as default.

If `Settings: GRBL > Spindle > Default Spindle:` is `PWM` then `M104P0` and `M104P1` don't seem to do anything (no error in console, nothing changes on the board). But if I change `Default Spindle` to `VFD` then it works fine, switching between PWM and VFD and I see the corresponding LEDs come on the GRBLHAL2000 board.

`$$=395` gives me the same output whether it's set as VFD or PWM:
```
$$=395
    0 - PWM
    1 - VFD Spindle
Spindle selected on startup.
```

I've found this behaviour on multiple GRBLHAL2000 boards (Teensy 4.1 based). I don't have any other board types to test though.

terjeio, 2022-08-30:
Can you check the latest commit, it seems you are building with an older version than my previous commit as well.

jaymis, 2022-08-31:
Aah apologies, I completely forgot to tell you which firmware version I've been running!

I'm still struggling a bit with pulling from the various sources and keeping my build environment up to date, so I've just been using the builds that @andrewmarles releases for the grblhal2000 board.

I think he's rolling out some new builds soon, so I'll update and check back in. 

Thanks Terje.

andrewmarles, 2022-08-31:
Yes, the Hal2K build is slightly behind and needs to be updated.

closed 2023-04-29 by terjeio