grblHAL/core #179: Reducing the baud rate of the Keypad plug-in serial port will also reduce the communication speed of the USB virtual serial port

hanke-cnc, 2022-08-25:
hi, I tried to make an encoder version of the handwheel for grblhal, using the keyboard plug-in, the test is no problem at all, but I found that if I reduce the serial communication rate of the keyboard plug-in (hope to improve the noise immunity with a lower communication rate), then The communication rate of the USB virtual serial port will also be reduced, sending "?", the reply becomes very slow,

hanke-cnc, 2022-08-25:
[image]

terjeio, 2022-08-25:
This is expected behaviour. The slowest stream will block processing when its output buffer is full - increase the reporting interval enough to let the output buffer empty out between requests to avoid blocking.

hanke-cnc, 2022-08-25:
> This is expected behaviour. The slowest stream will block processing when its output buffer is full - increase the reporting interval enough to let the output buffer empty out between requests to avoid blocking.

My handwheel doesn't need to receive replies from grblhal, what should I comment out? This way it shouldn't block

hanke-cnc, 2022-08-25:
My handwheel only represents G91G1X1F500 by sending a single character, like "Q", it doesn't need to receive information from grblhal, I don't know where I should comment out so that grblhal doesn't send information to the keyboard serial port so it doesn't block

hanke-cnc, 2022-08-25:
google translate is a bit bad,

terjeio, 2022-08-25:
Try:

```
#elif MPG_MODE == 2
    hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), true, keypad_enqueue_keycode);
#elif KEYPAD_ENABLE == 2
```
second parameter to stream_mpg_register() changed from `false` to `true`.

hanke-cnc, 2022-08-25:
> true

After changing to "true" it doesn't send info but also doesn't receive my handwheel info☺

hanke-cnc, 2022-08-25:
> 

It has been tested 2 times, you may have misremembered the code?

> Try:
> 
> ```
> #elif MPG_MODE == 2
>     hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), true, keypad_enqueue_keycode);
> #elif KEYPAD_ENABLE == 2
> ```
> 
> second parameter to stream_mpg_register() changed from `false` to `true`.

It has been tested 2 times, you may have misremembered the code?

hanke-cnc, 2022-08-25:
> Try:
> 
> ```
> #elif MPG_MODE == 2
>     hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), true, keypad_enqueue_keycode);
> #elif KEYPAD_ENABLE == 2
> ```
> 
> second parameter to stream_mpg_register() changed from `false` to `true`.

I pass the comment // hal.stream.write_all = stream_write_all;
it got better, thanks

closed 2022-08-25 by hanke-cnc