grblHAL/core #180: I'm trying to add custom M code for switching IO pins and I'm having a problem

hanke-cnc, 2022-08-27:
hi, there are many pins on the f401, it looks a bit wasteful, so I'm going to add some custom M code for switching the io pins,
I know you have an example program for adding custom M-codes, but it looks a bit complicated to me,
So I changed it directly in the grblhal source code,
My status_code_t gc_execute_block(char *block) in gcode.c
Added my simple code in the function
:
             case 70:
//

             HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_SET);
break;
             case 71:
  //

              HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_RESET);
  break;

Very good, he can run correctly, but HAL_GPIO_WritePin() runs as soon as grblhal receives this code, I don't know where to add it to make grblhal run HAL_GPIO_WritePin() after actually executing the m70 code

hanke-cnc, 2022-08-27:
It seems a little rude for me to modify the program directly in the source code like this to your grblhal philosophy

terjeio, 2022-08-27:
You can add up to 8 outputs and 7 inputs in your map file (for the STM32F4xx driver), these can then be controlled by M62-M65 (outputs) and M66 (inputs) and they are also available for use by plugin code. E.g [like this](https://github.com/grblHAL/STM32F4xx/blob/effe8daafa71f0e466eb9ee8f5dda4b058436140/Inc/st_morpho_map.h#L160-L207).

If you really need custom M-codes the best way is to add them via a plugin or board specific code as this will not be overwritten on an update. Is it [this example](https://github.com/grblHAL/Templates/blob/master/mcodes.c) you found to be complicated?

hanke-cnc, 2022-08-28:
> HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_RESET);

> You can add up to 8 outputs and 7 inputs in your map file (for the STM32F4xx driver), these can then be controlled by M62-M65 (outputs) and M66 (inputs) and they are also available for use by plugin [...]

Well, I'm not a professional programmer, I downloaded mcodes.c and mcodes.h,
And added in user_mcode_t in gcode.h
my_off =998,
     my_on=999
Added switch(gc_block->user_mcode) to the execute function in mcodes.c {

         case 999:
         HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_SET);
             break;
         case 998:
         HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_RESET);
         break;

After compiling and downloading, enter m999, M999 < error:20
The mocodes.c code doesn't seem to be enabled correctly,
I noticed your comment about the mcodes_init() function:
// Set up HAL pointers for handling additional M-codes.
// Call this function on driver setup.
But I don't know where to call the mcodes_init() function, what should I do?

terjeio, 2022-08-28:
You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?

If you have a board map file the correct way to add board specific code is to create a c file with the same name as the board map and add [#define HAS_BOARD_INIT](https://github.com/grblHAL/STM32F4xx/blob/effe8daafa71f0e466eb9ee8f5dda4b058436140/Inc/st_morpho_map.h#L49) to the map file. In the c file check for the board name definition [like this](https://github.com/grblHAL/STM32F4xx/blob/effe8daafa71f0e466eb9ee8f5dda4b058436140/Src/st_morpho.c#L24) and name the init function [board_init()](https://github.com/grblHAL/STM32F4xx/blob/effe8daafa71f0e466eb9ee8f5dda4b058436140/Src/st_morpho.c#L217).

Additional M-codes has to be checked for in three places, [first telling the parser that they are available](https://github.com/grblHAL/Templates/blob/1907a0cc5ef9f8ef7e7bc70837a6dabbe5d89518/mcodes.c#L42-L47), then when[ validating and claiming the parameters](https://github.com/grblHAL/Templates/blob/1907a0cc5ef9f8ef7e7bc70837a6dabbe5d89518/mcodes.c#L49-L87) (if any) and finally when [executing the related code](https://github.com/grblHAL/Templates/blob/1907a0cc5ef9f8ef7e7bc70837a6dabbe5d89518/mcodes.c#L89-L112).

It is important to call the original functions ([via the saved pointers](https://github.com/grblHAL/Templates/blob/1907a0cc5ef9f8ef7e7bc70837a6dabbe5d89518/mcodes.c#L120)) if the M-code is not one of yours - this way other plugin code may also add M-codes.

hanke-cnc, 2022-08-28:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

closed 2022-08-28 by hanke-cnc

hanke-cnc, 2022-08-28:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

Haha 😄, reading grblhal code is a bit difficult for me, I am not very familiar with grblhal for the time being,

hanke-cnc, 2022-08-28:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

I would like to report to you that my grblhal control board based on stm32f401rct6 has sold 100 pieces, some of them are used to control diy cnc machines, some are used to study and learn grblhal,

hanke-cnc, 2022-08-28:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

Also, I would like to ask how you are doing with the grblhal automatic tool changer?

I think it should not be very complicated to implement for this kind of tool magazine in the picture,

[image]

hanke-cnc, 2022-08-28:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

For other types of tool magazines and tool change modes, grblhal should no longer consider them. The straight-line tool magazine in the picture is the easiest to implement and the cheapest.

hanke-cnc, 2022-08-28:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

> terjeio

hi, I got up this morning and successfully added my custom M code by the method you said,
no one knows grblhal better than terjeio, great terjeio

hanke-cnc, 2022-08-29:
> You cannot use the standard way by defining aux inputs/outputs in the map file and using the M62-M65 commands?
>
> If you have a board map file the correct way to add board specific code is to create a [...]

Somewhat disappointed, the M code added via mcodes.c is still grblhal executes as soon as it is received, not the real execution like M3, M8, I don't know what I'm doing wrong?

/*

  mcode.c - user defined M-codes template

  Part of grblHAL

  Copyright (c) 2019-2021 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.

*/

/*
 * NOTE: this template is also a bare bones example for adding M100 with two parameters: P and Q
 */

#include <math.h>
#include <string.h>

#ifdef ARDUINO
#include "../grbl/hal.h"
#else
#include "grbl/hal.h"
#endif
#include "main.h"
static user_mcode_ptrs_t user_mcode;
// 检查 - 检查是否在此处处理 M 代码。
// 参数： mcode - 要检查的 M 代码（有些是在 grbl/gcode.h 的 user_mcode_t 中预定义的），如果不是，则使用强制转换。
// 返回：如果处理则为 mcode，否则返回 UserMCode_Ignore（UserMCode_Ignore 在 grbl/gcode.h 中定义）。

static user_mcode_t check (user_mcode_t mcode)
{
	hal.stream.write("CCC\n");//用于debug
    return mcode == 999
                     ? mcode                                                            // Handled by us.
                     : (user_mcode.check ? user_mcode.check(mcode) : UserMCode_Ignore); // 如果存在另一个处理程序，则调用它或返回忽略。
}

// validate - 验证参数
// 参数： gc_block - 指向 parser_block_t 结构的指针（在 grbl/gcode.h 中定义）。
// gc_block->words - 保存一个可用参数字的位域。
// 如果浮点值是 NAN（非数字），这意味着它们不可用。
// 如果整数值的所有位都设置为 1，这意味着它们不可用。
// 返回：status_code_t 枚举（在 grbl/gcode.h 中定义）：如果验证成功，则 Status_OK，否则来自枚举的适当状态。
static status_code_t validate (parser_block_t *gc_block, parameter_words_t *deprecated)
{
    status_code_t state = Status_GcodeValueWordMissing;
    hal.stream.write("BBB\n");//用于debug
    switch(gc_block->user_mcode) {

        case 999:
        	   state = Status_OK;
        	   gc_block-> user_mcode_sync = 0 ;
            break;

        default:
            state = Status_Unhandled;
            break;
    }

    // 如果我们没有处理并且存在另一个处理程序，则调用它。
    return state == Status_Unhandled && user_mcode.validate ? user_mcode.validate(gc_block, deprecated) : state;
}

// 执行 - 执行 M 代码
// 参数：state - sys.state（位图，在system.h中定义）
// gc_block - 指向 parser_block_t 结构的指针（在 grbl/gcode.h 中定义）。
// 返回：-
static void execute (sys_state_t state, parser_block_t *gc_block) {
	hal.stream.write("AAA\n");//用于debug
    bool handled = true;

    switch(gc_block->user_mcode) {

        case 999:
        	 HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_SET);

            break;
        case 998:
        	HAL_GPIO_WritePin(GPIOB, GPIO_PIN_3, GPIO_PIN_RESET);
        	 break;
        default:
            handled = false;
            break;
    }

    if(!handled && user_mcode.execute)         // 如果没有由我们处理并且存在另一个处理程序
        user_mcode.execute(state, gc_block);    //然后调用它。
}

// 设置 HAL 指针以处理额外的 M 代码。
// 在驱动程序设置时调用此函数。
void mcodes_init (void)
{
	// 保存当前的 HAL 指针，以便我们可以使用它们来保存
	     // 任何完整的 M 代码处理程序链。
    memcpy(&user_mcode, &hal.user_mcode, sizeof(user_mcode_ptrs_t));

    // 将 HAL 指针重定向到我们的代码。
    hal.user_mcode.check = check;
    hal.user_mcode.validate = validate;
    hal.user_mcode.execute = execute;
}

terjeio, 2022-08-29:
> I would like to report to you that my grblhal control board based on stm32f401rct6 has sold 100 pieces, some of them are used to control diy cnc machines, some are used to study and learn grblhal,

Great, IMO you should create a pull request for your map file(s).

>Also, I would like to ask how you are doing with the grblhal automatic tool changer?

I have not been working on this for a long time.

> I think it should not be very complicated to implement for this kind of tool magazine in the picture,

It should be easy enough - you want to give it a try? [Here is the code](https://github.com/grblHAL/MSP432P401R/blob/master/atc.c) I have been working on, you may get some ideas from that.

> For other types of tool magazines and tool change modes, grblhal should no longer consider them. The straight-line tool magazine in the picture is the easiest to implement and the cheapest.

The grblHAL core does not care about which kind of toolchanger is in use, different changers can be supported via plugins, either hardware specific or generic.

> Somewhat disappointed, the M code added via mcodes.c is still grblhal executes as soon as it is received, not the real execution like M3, M8, I don't know what I'm doing wrong?

You want [synchronized execution](https://github.com/grblHAL/Templates/blob/1907a0cc5ef9f8ef7e7bc70837a6dabbe5d89518/mcodes.c#L76)? Setting this variable to true will delay execution until all queued commands has been completed.

hanke-cnc, 2022-08-29:
> It should be easy enough - you want to give it a try? [Here is the code](https://github.com/grblHAL/MSP432P401R/blob/master/atc.c) I have been working on, you may get some ideas from that.

Ok 👌, thanks, next year I will go buy a tool changer spindle and try it out.
Hi! I would like to share with you the embedded CNC system I am making based on the open source grblhal and candle. It has always been my dream to have my own CNC system. It is very difficult to make a CNC system from scratch, and based on the open source grblhal and candle much easier,

https://user-images.githubusercontent.com/108179425/187188388-4932450f-5a47-4d78-ad59-75a5f7088cd1.mp4