grblHAL/core #182: Teensy4.1 WebUI compile error

pzh11001, 2022-09-04:
I use Arduino ide 1.8.19. The MCU is teensy4.1

Download the latest grblHAL version, install the sdcard plug-in, networking plug-in, and webui plug-in. 
In "my_ machine.h" , i opening "define ethernet_enable 1" and "define sdcard_enable 1" in the "my_ machine.h" file, it can compile successfully. And connect grblhal normally through Ethernet. I can use "wschat.html" . and the connection is successful and the command is sent normally .

However, when I open "define webui_enable 1" in "my_ machine.h" file, a compilation error occurred. As follows:
(I have patched all the plug-in according to the instructions, but it seems that there is a folder in "teensy41_ethernet master"library need to be deleted. I forget which folder it is.)

Error:
[image]
[image]

After commenting out "/ / # define webui_enable 1",  it can compile normally, start wschat.html, connect normally, and send commands normally
[image]
[image]

terjeio, 2022-09-04:
Another one forgotten...
Change `LWIP_HTTPD_CUSTOM_FILES` in the patched _lwipopts.h_ to
`#define LWIP_HTTPD_CUSTOM_FILES        0`

Be aware that the latest [WebUI build is currently not working](https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/CNC/GRBLHal), I am looking into that now.

pzh11001, 2022-09-04:
I see, Thank you for your great work！
Don't be too tired：）

terjeio, 2022-09-12:
FYI [WebUI v3 support is improving](https://github.com/grblHAL/core/blob/master/changelog.md), flash filing system added etc...

closed 2022-09-12 by terjeio