grblHAL/core #183: How to install grblHAL?

dmtuniverse, 2022-09-04:
Not really an issue, more a call for help.

I don't normally get stuck on things like this, I'm quite comfortable with designing PCB's and writing my own software etc, but for some reason I cannot for the life of me figure out how to install grblHAL. I find the wiki instructions to be confusing. Gave up on my last attempt and stuck with GRBL, but I need a faster controller now.

Where am I supposed to put the driver files, inside the core folder? The logic on the instructions (as I interpret it, wouldn't surprise me to learn I missed something) goes as follows:

Download the drivers for the MCU of choice.
Upload.

There's some steps missing in between. Have I missed some other documentation?

terjeio, 2022-09-04:
>  Have I missed some other documentation?

[This](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL)?
Which MCU have you chosen?

dmtuniverse, 2022-09-06:
That's the page that confuses me. I don't see understand what to do with the core repo, only seems to mention the drivers.

Am I supposed to be uploading just the driver? Or do I merge it with the core before adding it as a library?

I cloned the repo's using GitHub Desktop.

I'm targeting SAM3X8E on an Arduino Due with Arduino IDE.

dmtuniverse, 2022-09-06:
I get this error when trying to compile the example using just the drivers as the library. Same error regardless of which port I use.

In file included from C:\Users\Aviotronics\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/syscalls.h:29:0,
                 from C:\Users\Aviotronics\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\variants\arduino_due_x/variant.h:50,
                 from C:\Users\Aviotronics\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:201,
                 from C:\Users\Aviotronics\Documents\Arduino\libraries\grblHAL_Due\src\grbl\grbl.h:60,
                 from C:\Users\Aviotronics\Documents\Arduino\libraries\grblHAL_Due\src\grbl\hal.h:29,
                 from C:\Users\Aviotronics\Documents\Arduino\libraries\grblHAL_Due\src\grbl\vfs.c:27:
C:\Users\Aviotronics\Documents\Arduino\libraries\grblHAL_Due\src\grbl\vfs.c: In function 'vfs_eof':
C:\Users\Aviotronics\Documents\Arduino\libraries\grblHAL_Due\src\grbl\vfs.c:231:35: error: expected identifier before '(' token
     return ((vfs_t *)(file->fs))->feof(file);
                                   ^
exit status 1
Error compiling for board Arduino Due (Native USB Port).

terjeio, 2022-09-06:
_grbl/vfs.c_ [should not be compiled](https://github.com/grblHAL/core/blob/e009f47116db4d21f94d2bd1a1b5e682c35ff493/vfs.c#L27) for the DUE. If `#ifndef ARDUINO_SAM_DUE` is not present you have a previous version - either update or remove all lines in the file.

dmtuniverse, 2022-09-08:
For some reason I had downloaded an older version which did not contain the "#ifndef". Not sure how that happened, all I did was repeat the exact same procedure I did previously on GitHub Desktop, but this time it downloaded the most recent release.

Firmware uploaded successfully. Board connects to LightBurn. Just about to install the board in the machine, hopefully all works ok.

Thanks for your help and sharing your work :) appreciate it.

dmtuniverse, 2022-09-08:
Apologies for having to reopen this. I'm still having issues, no doubt all of my own doing, but nonetheless.

So the board appears to connect to LightBurn but I don't think it is connecting. I downloaded IOSender and cannot get it to connect to the board. Closes itself every time saying "no response received from board. Exiting". Given that you wrote that software specifically for this firmware I'm going to assume I've still not uploaded the firmware correctly.

Tried both native USB and UART ports on the Due.

terjeio, 2022-09-08:
I'just tested with the default settings in _my_machine.h_ - and I can connect via the programming port.
If connection via the native USB port you have to uncomment `//#define USB_SERIAL_CDC 1` in _my_machine.h_ and recompile/reupload. If the native USB port is enabled and it shows up in Windows then it is likely that the firmware is running as it should.

dmtuniverse, 2022-09-08:
Perhaps I am just destined to not use grblHAL then :(

As I said I cannot connect via either programming port or native USB. Same result with both. Same result on multiple PC's.

Steps I took:

-Download the driver for SAM3X8E
-Add driver repo to Arduino IDE as library (IDE version 1.8.19).
-Double tap the reset button.
-Upload (think I ended up using the native USB to upload the firmware, can't remember now)

The upload flashes successfully.

Not sure what else I can even try to do at this point.

dmtuniverse, 2022-09-08:
Is there a driver needed specifically for the SAM3X8E by any chance? 

The board I have uses a CH340 UART, I have the drivers installed for the UART chip.

terjeio, 2022-09-08:
I've now tested with the USB port as well (`//#define USB_SERIAL_CDC 1` uncommented), works ok and the port shows up like this for me in ioSender:

[image]

If the port enumerates (shows up in the drop down) then there is a driver installed.
I flash via the programming port and do not double tap the reset button beforehand. And I have a genuine Arduino Due...

dmtuniverse, 2022-09-08:
So this is the result I get from my current setup. It will initiate connection without quitting IOsender, but then times out waiting for a response.

[image]

I tried to reflash the firmware using the programming port without double tap. Can't find the port.

`Sketch uses 137188 bytes (26%) of program storage space. Maximum is 524288 bytes.
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:152)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2061)
	at java.lang.Thread.run(Thread.java:748)
Caused by: processing.app.SerialException: Error touching serial port 'COM39'.
	at processing.app.Serial.touchForCDCReset(Serial.java:107)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:136)
	... 5 more
Caused by: jssc.SerialPortException: Port name - COM39; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.touchForCDCReset(Serial.java:101)
	... 6 more
`

Seems I cannot upload via the programming port with or without double tapping reset.

Uploading via the native USB works fine with and without double tap of reset.

I only double tapped reset because of the following line in the example sketch:

`// Double tap reset to enter bootloader mode - select bootloader port for programming`

terjeio, 2022-09-08:
Ok, so the programming port does not enumerate - this means you need a driver for the chip used on the Due?

If you want to communicate via native USB you have to edit _my_machine.h_ - did you do that?
Remove `//` in front of [#define USB_SERIAL_CDC 1](https://github.com/grblHAL/SAM3X8E/blob/0c0d760f637ecfce009344411dbcbc1d1ba54131/grblHAL_Due/src/my_machine.h#L36) and recompile/reupload.

dmtuniverse, 2022-09-08:
Brilliant. Works now. Thank you :)

I neglected to recompile after uncommenting that line.

Still doesn't work via the programming port, but native USB works. As long as it works one way or another that's all that matters to me.

Thanks again for the help!!!!

closed 2022-09-08 by dmtuniverse

Ramr93, 2023-01-01:
> For some reason I had downloaded an older version which did not contain the "#ifndef". Not sure how that happened, all I did was repeat the exact same procedure I did previously on GitHub Desktop, [...]

can you help with steps to compile?