grblHAL/core #184: PlatformIO compiling conflict with "limit.h"

dresco, 2022-09-07:
Hi Terje, would you consider a pull request to rename the grbl limit.c & limit.h files? 

Building with networking support exposes this same problem that the Grbl_Esp32 folks faced [here](https://github.com/bdring/Grbl_Esp32/issues/74), and fixed [here](https://github.com/bdring/Grbl_Esp32/pull/75).

I see this on my H7 port, am assuming it would be the same for F7. I've fudged around it for now in my [platformio.ini](https://github.com/dresco/STM32H7xx/blob/a609f9e4a08825df5074f4400c2ca4f1665bc6fe/platformio.ini#L124-L126)

terjeio, 2022-09-07:
> would you consider a pull request to rename the grbl limit.c & limit.h files

I do not think so as you will have to fix all code referencing the .h file, so better I do it. I have a HAL function signature change I want to implement as well - I can do this at the same time as I work my way through all the drivers.

I'll remame _limits.h_ and _limits_.c_ to _machine_limits.h_ and _machine_limits.c_ since I find _grbl_limits_ to be somewhat ambiguous...

terjeio, 2022-09-07:
Done. See [changelog.md](https://github.com/grblHAL/core/blob/master/changelog.md) for signature change++.

closed 2022-09-07 by terjeio

dresco, 2022-09-07:
> Done. See [changelog.md](https://github.com/grblHAL/core/blob/master/changelog.md) for signature change++.

Thanks!