grblHAL/core #186: Use PWM/TTL Laser

Demitrius, 2022-09-08:
Not really an issue, more a call for help.

Trying setup laser cutter/engraver.
Laser module have VCC, GND and PWM pins.
As I imagine, laser PWM pin can be wired with FAN or E0 (extruder) board port.

What is the easy/better way to use Laser with grblHAL?

dresco, 2022-09-08:
I believe the laser PWM input would be wired to the spindle PWM port.

On your BTT SKR3 board - the spindle output should be pin 9 on the EXP1 header.. 
```
// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE       GPIOB_BASE
#define SPINDLE_PWM_PIN             0                           // EXP1 - PB0, pin 9
```

(Note also that you can check pin assignments with the `$pins` command, but that doesn't really help you know what connector on the board it's wired to. I've tried to document them in the [map file](https://github.com/dresco/STM32H7xx/blob/master/Inc/btt_skr_v3.0_map.h)).

dresco, 2022-09-09:
@terjeio I don't have a laser module, but just wanted to check the outputs look sane on a logic analyser..

I found this [laser_mode](https://github.com/terjeio/grblHAL/blob/master/doc/markdown/laser_mode.md) page on the old repo, which explains the constant & dynamic PWM modes. Where does the PPI plugin fit into all this? Thanks..

terjeio, 2022-09-09:
[PPI mode](https://www.buildlog.net/blog/2011/12/getting-more-power-and-cutting-accuracy-out-of-your-home-built-laser-system/) is a third option for laser power control, it switches the laser on/off based on distance moved. IMO it provides the best power control for cutting as it has close to linear relationship between power delivered to the cut vs. feed rate.
CO2 lasers has a pretty non-linear relationship between PWM output voltage and power delivered, don't know how diode lasers perform.

dresco, 2022-09-11:
Thanks, have checked the output on a logic analyser & makes sense.. 

One thing I observed with the PPI plugin though, the specified pulses per inch is only correct at low feed rates. Haven't tested on anything but H7, but for me it starts to drop off somewhere between 400 & 500 mm/min?

Oh, and noticed a minor typo in the F7 code;
```
diff --git a/Src/driver.c b/Src/driver.c
index 947d3c2..e4a4480 100644
--- a/Src/driver.c
+++ b/Src/driver.c
@@ -1899,7 +1899,7 @@ bool driver_init (void)
         .get_pwm = spindleGetPWM,
         .update_pwm = spindle_set_speed,
   #if PPI_ENABLE
-        .pulse_on = spindlePulseOn;
+        .pulse_on = spindlePulseOn,
   #endif
  #endif
         .config = spindleConfig,
```

Demitrius, 2022-09-12:
> I believe the laser PWM input would be wired to the spindle PWM port.
>
> On your BTT SKR3 board - the spindle output should be pin 9 on the EXP1 header..
>
> ```
> // Define spindle PWM output pin.
> #define [...]

Jon, pin 9 is pwm with 3.3v max. Can be configured for 5v?

Laser module is 5v, right?

[image]

Demitrius, 2022-09-12:
I found extra information. Laser can be connected to any of HE0/HE1 (PB3/PB4) ports by manual from BTT.

[image]

I was trying to do this.

1 - Configure ports
```
#define SPINDLE_PWM_PORT_BASE       GPIOB_BASE
#define SPINDLE_PWM_PIN             3                           // HEAT0 - PB3

```
2 - Connect voltmeter instead laser to HE0

[image]

3 - Test HE0 output.

```
M3 S200;
G1 F750.0 X65.666083 Y168.686816;

```
   Way A - with jumper. Voltmeter show about 6V output for S200
   Way B - jumper removed. Voltmeter always show 0V output on HE0

Way B is right way by BTT manual, but not works or I have mistake in wire?

dresco, 2022-09-12:
> I found extra information. Laser can be connected to any of HE0/HE1 (PB3/PB4) ports by manual from BTT.

Good find!

> Way B is right way by BTT manual, but not works or I have mistake in wire?

Just a mistake in the wiring I think, try taking the signal from the lower connector where you removed the jumper (the one in the red square). That should be the PWM output once it's passed through the 5V buffer IC.

closed 2022-09-15 by Demitrius