grblHAL/core #188: ask 4 driver

irshadmgs011, 2022-09-10:
Is there support for trinamic drivers on the arduino due board?

terjeio, 2022-09-11:
Currently no, but it should not be too hard to add the needed lowlevel code for either SPI or UART comms (if not soft UART). You can do it?

irshadmgs011, 2022-09-12:
sorry, i don't understand low level coding

terjeio, 2022-09-12:
The Due is a fairly old board and not used by many - and there are no shields available that has support for Trinamic drivers?
So IMO you should instead consider bying a board that already has Trinamic support if you need this for your own machine.

irshadmgs011, 2022-09-12:
how stable is this firmware, how many marks be given in 5?

terjeio, 2022-09-12:
> how stable is this firmware,

Most parts are, IMO, rock solid provided that the installation is electrically sound. Some parts, such as the latest WebUI releases, might not be - but the code will be "hardened" as usage increases and any problems are reported/fixed.

> how many marks be given in 5?

In a range of 1-5? Perhaps some users can comment on that.