grblHAL/core #191: How to define an axis as a rotation axis

GrblGru, 2022-09-13:
Hi Terje,

first of all, thank you very much for your great work.
I've only recently become aware of your project, so I'm still a newbie with lots of (maybe stupid) questions.

I have developed a simple CAM program 'GrblGru' which can already communicate with some controllers (Grbl, TinyG, Mega-5X) via USB. Thereby a 3D machine model is visualized on the screen parallel to the execution of the calculated GCode.
Now I would like to extend this functionality to GrblHAL. First attempts look very promising.
But I can't find any information about how to define an axis as a rotation axis. Can you please help me and point me to the corresponding information ?

Thanks in advance
GrblGru

terjeio, 2022-09-13:
`$376` - but only for A, B and C axes - disables scaling if in inches mode. Setting max travel to 0 for an axis disables soft limits for it as well. Note that many settings will not show up in the `$$` report if configuration does not need them, you have to set [N_AXIS > 3](https://github.com/grblHAL/core/blob/b9f5aebf65168c71b105d71641c2dbc21fc913eb/config.h#L35-L41) to enable `$376`.

You may find useful info in the wiki, e.g. there is a page [intended for developers](https://github.com/grblHAL/core/wiki/For-sender-developers).

I have seen your program, impressive!

GrblGru, 2022-09-14:
Terje, thanks for the quick reply. 

Have now been able to configure the desired 5-axis XYZAB machine with "#define N_AXIS 5". :)
But I am still not clear how to set up for example a machine with 4 linear axes XYZU, like you need for a foam cutter.

So there is still a lot for me to learn. Is this the right place to ask questions ? Or is there a more appropriate place to discuss such interesting topics as "hand wheel jogging, G5, thread cutting, Constant Surface Speed mode etc" ?

If there is interest, I would like to contribute a small part to the further development of GrblHAL and e.g. test new functions.
My program might be suitable to test GrblHAL on the desktop before a real test on the machine because of the graphical possibilities.

Cheers
GrblGru

terjeio, 2022-09-14:
> But I am still not clear how to set up for example a machine with 4 linear axes XYZU, like you need for a foam cutter.

I have not added support for remapping axis letters ABC to UVW, there is basic support for up to 8 axes (adds UV as option) but this has to be completed.

> So there is still a lot for me to learn. Is this the right place to ask questions ? Or is there a more appropriate place to discuss such interesting topics as "hand wheel jogging, G5, thread cutting, Constant Surface Speed mode etc" ?

Start [discussions](https://github.com/grblHAL/core/discussions) instead of raising issues.

> My program might be suitable to test GrblHAL on the desktop before a real test on the machine because of the graphical possibilities.

Interesting, I have made a [hardware emulator](https://github.com/terjeio/CNC_Machine_Simulator) with a PSoC 5 that can simulate VFDs, spindle encoders, limit switches, counting step pulses, probing etc... Currently this has a simple CLI interface, it would be nice to have a graphical UI for it. Could be a useful combo with GrblGru?

GrblGru, 2022-09-16:
Thank you for your explanation. I will use the discussions section for further questions in future.
I will also try to make a short video about how GrblHAL can communicate with GrblGru in the next days.

terjeio, 2022-09-23:
>>  But I am still not clear how to set up for example a machine with 4 linear axes XYZU, like you need for a foam cutter.

> I have not added support for remapping axis letters ABC to UVW, there is basic support for up to 8 axes (adds UV as option) but this has to be completed.

Support for ABC axis letter remapping to UVW has been added in a [recent commit - 20220918](https://github.com/grblHAL/core/blob/master/changelog.md). How do you detect axisletters to use? Via the `[AXS:<n>:<letters]` option in `$I`?

evert4130, 2022-10-17:
Hi @terjeio , thank you for your fantastic work. I have a question regarding the rotational axis.

I run my machine in XYZA configuration, with an inductive sensor to home the A-axis. I have the max travel set to 99999 mm/units (degees). Soft limits enabled, A axis defined as rotational in $376=1. This works very well, but I can only rotate it in the one direction after homing due to the soft limits. I have not run into the other soft limit yet.

When I read your post above, "... setting max travel to 0 disables soft limits", I thought it might be a good solution. But unfortunately, when the machine is homing the A-axis it throws an alarm 9. "...Try increasing max travel..."

Am I missing something, or is there another setting that needs to be changed?

terjeio, 2022-10-17:
@evert4130 This is a "bug" or rather a side effect of setting max travel to zero as max travel is used for setting the distance to seek for the limit switch. A fix will be to set the seek distance to 360 (degrees) for rotational axes when max travel is 0?

evert4130, 2022-10-17:
@terjeio That certainly seems like a good fix. Thanks for the quick response.

terjeio, 2022-10-18:
@evert4130  Can you try with changing
https://github.com/grblHAL/core/blob/2f42b7226bc9c70b98d612ae1c46576b718153f5/machine_limits.c#L277
to
`            max_travel = max(max_travel, (-HOMING_AXIS_SEARCH_SCALAR) * (settings.axis[idx].max_travel < -0.0f ? settings.axis[idx].max_travel : 360.0f));`

I may add a check for the axis beeing configured as rotational as well.

evert4130, 2022-10-18:
Wow, I will do it tomorrow and report back to you here, thanks again.

evert4130, 2022-10-19:
@terjeio Fantastic! It works exactly as expected. Your code is perfect. I can now home all axis normally (my sequence is Z-A-XY) and can then rotate the A-axis in any direction and with no limits, perfect. Thank you so very, very much!