grblHAL/core #202: Proposal | Proposal: Should a wait delay time to implement spindle start should be added to the core? The user can set the time through $

hanke-cnc, 2022-10-16:
To be precise, it is proposed by the user who purchased the grblhal control board

dresco, 2022-10-17:
Hi, there is already a spindle at speed check, see [this discussion](https://github.com/grblHAL/core/discussions/198) for recent explanation..

Btw, you may want to raise a PR to get your map file upstream?

closed 2022-10-19 by hanke-cnc

hanke-cnc, 2022-10-19:
> Hi, there is already a spindle at speed check, see [this discussion](https://github.com/grblHAL/core/discussions/198) for recent explanation..
> 
> Btw, you may want to raise a PR to get your map file upstream?

terjeio is a great man

dresco, 2022-10-19:
> Do you mean to upload the pin definition file for the grblhal control board I sell?
> Oh, I made the PCB according to the cnc3040.map file in the stm32F4 driver file, so it is not necessary

Yes, that is what I meant.. Thanks.