grblHAL/core #209: Question: homing with X and y auto squaring

matthiasng, 2022-10-22:
https://github.com/grblHAL/core/blob/fbef120e947117f88b42fcea3716237d4c8babec/machine_limits.c#L491

The function noted above says it only can home one auto squared axis at a time. $H on the other side always tries to home all axis. This means if I have an X and y axis with auto squaring enabled there is no way I can use $H for homing. Is there any particular reason why the function does not support it ?

Furthermore I'm also confused about the return Boolean return value. It actually returns false in this case, but the calling function doesn't set any alarms. This means, the controller is only homing Z, stops then and sets the status of all axis to homes (which seems like a bug to me).

terjeio, 2022-10-22:
> $H on the other side always tries to home all axis.

Yes, but according to the homing passes defined by $44-46, each pass can home a single axis if so configured. If a pass fail the sequence is exited.

> This means, the controller is only homing Z, stops then and sets the status of all axis to homes (which seems like a bug to me).

Yep - it seems like it does if you try to home X and Y at the same time and both are auto-squared. An alarm should be raised? Or returning an error is better?

matthiasng, 2022-10-23:
Ty for the clarification. I must have overlooked the homing cycle setting.

I think the way you have done it now is perfect. This way users will get feedback when their configration is wrong (like mine) 👍.

closed 2022-10-23 by matthiasng