grblHAL/core #211: Add a kinematic structure

ztcg888, 2022-10-26:
I want to add another kinematics structure, such as URAM. I don't know how to add it, or directly modify the corexy kinematics implementation can do it? How to get started?

terjeio, 2022-10-26:
> How to get started?

Study the existing code?
There are two basic approaches, [corexy](https://github.com/grblHAL/core/blob/master/corexy.c) is the simplest with a straightforward transform, [wall plotter](https://github.com/grblHAL/core/blob/master/wall_plotter.c) is a bit more complicated as long lines has to be split up into small segments. [Maslow](https://github.com/grblHAL/core/blob/master/maslow.c) (untested) is similar to wall plotter with the addition of settings and $ commands.

ztcg888, 2022-10-27:
Thank you!  My friend, After seeing your reply, I have some confidence, I will try it!

dentikhval, 2023-01-30:
> Study the existing code? There are two basic approaches, [corexy](https://github.com/grblHAL/core/blob/master/corexy.c) is the simplest with a straightforward transform, [wall [...]

Dear Terje, 

I have a [working IK (Inverse Kinematics) solution](https://github.com/dentikhval/SmallInverseKinematics/tree/master/Simple6DoF_IK_Lite) with a clear input (Cartesian coord.) and output (joint angles), and I am sure many robotic arm builders would love it.

I do not fully understand the code structure of GrblHAL, maybe there is someone with a better understanding of all the existing features to implement this kinematic for GrblHAL as a full-featured solution rather than the minimal half-a$$ed implementation I will be able to come up with?

Please do not send me to study the code, I tried and it's too big.

ztcg888, 2023-02-01:
&nbsp;Thank you, my friend, and thank you for the way you provided. During this period of time, I am sorting out the process of the g code from receiving to turning into a single step of the motor. The coordinate conversion has not been found out yet. Here is Replacing the linear motion of the motor with an angle, the positive and negative kinematic functions may have been verified, but it is a bit messy when combined into the system. It looks a bit simple, and it is also difficult to debug, mainly because it is still not good for your entire system. Too familiar, but with your help, I think I will debug a little bit until I succeed!

Message ID: ***@***.***&gt;