grblHAL/core #215: Feature request: Inactivity timeout for driver enable

LoganFraser, 2022-11-01:
If this already exists, I couldn't find a reference to it, currently the "enable" pins to external drivers are setup to either disable after a very short delay or be permanently on. Since the drivers being disabled means they will lose position and require rehoming, a more lengthly delay in minutes, able to be set from 1-90 mins or something to that effect would be beneficial to save electricity in the case the machine is left on.

terjeio, 2022-11-01:
This is a good candidate for [writing a plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin)?

terjeio, 2026-02-27:
A [template plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin/Stepper_enable_control) was added some time ago, and $1 has been changed to accept delays up to 65s (65000ms).