grblHAL/core #219: Hello, terjeio, I seem to have found a small bug in grblhal? The S value of the gc status information is the last value, not the current value

hanke-cnc, 2022-11-06:
Hello, terjeio, I seem to have found a small bug in grblhal? The S value of the gc status information is the last value, not the current value

The F/S value obtained by ? is the correct value

[image]

hanke-cnc, 2022-11-06:
I don't know if you can understand what I mean?

terjeio, 2022-11-06:
> The S value of the gc status information is the last value, not the current value

Correct. The GC status is from the current parser state which is at the back of the planner queue, the ? report is the current value including any override. This is per design.

hanke-cnc, 2022-11-06:
> > The S value of the gc status information is the last value, not the current value
> 
> Correct. The GC status is from the current parser state which is at the back of the planner queue, the ? report is the current value including any override. This is per design.

????????????
God, in my cognition, the gc state is the current state of grbl executing a line of g code

hanke-cnc, 2022-11-06:
I don't know linuxcnc and fanuc are the same?

hanke-cnc, 2022-11-06:
Now I don't know how to implement the idea of continuing processing after the user shuts down. Originally, my idea was to save the gc state of the currently executing G code line, send the saved gc state when continuing to process, and then send the g code from the specified line, but now Knowing that the gc state is not the current execution g-code state, does the great terjeio have any ideas?

closed 2022-11-07 by hanke-cnc