grblHAL/core #220: Spindle on delay shouldn't require spindle sync

devknoll, 2022-11-12:
https://github.com/grblHAL/core/blob/fbef120e947117f88b42fcea3716237d4c8babec/settings.c#L1730

The general spindle delay seems to require spindle sync (via the `cap.at_speed`). Seems like this should be `!hal.spindle.cap.at_speed` instead, since spindle sync has its own way (i.e. speed tolerance) to delay.

(I'm not at a desktop, otherwise I would PR this)

terjeio, 2022-11-12:
`Setting_SpindleOnDelay` is the delay before an alarm is generated if the actual RPM is not within limits. It is dependent on _at speed_ capability of the spindle, derived from RPM read either from an encoder or by other means provided by the spindle driver. E.g. VFD spindles typically provides the RPM as [a readout from a (modbus) register](https://github.com/grblHAL/Plugins_spindle/blob/6b70b5cfe2f3573b0cb33bd178e08c4e29b76353/vfd/huanyang.c#L215-L218), no encoder is needed.
The default driver provided spindle code _can_ provide RPM readout in some form independent of whether spindle sync is enabled or not - but is mostly related to the spindle sync code (for now) since spindle sync requires it.

An example of RPM readout from an encoder when spindle sync is not enabled can be found in the [MSP432P401 driver](https://github.com/grblHAL/MSP432P401R/blob/master/driver.c), from an experiment with PID controlled RPM.

devknoll, 2022-11-12:
In particular, I was looking at this branch:

https://github.com/grblHAL/core/blob/fbef120e947117f88b42fcea3716237d4c8babec/spindle_control.c#L299

So the value looks like it will be used _without_ triggering an alarm, but the setting doesn't appear (because the same `at_speed` check fails in settings.c)

terjeio, 2022-11-12:
I use the same settings value (but different setting ids) for both the [alarm delay](https://github.com/grblHAL/core/blob/fbef120e947117f88b42fcea3716237d4c8babec/settings.c#L569) and [spindle spinup delay](https://github.com/grblHAL/core/blob/fbef120e947117f88b42fcea3716237d4c8babec/settings.c#L566) on door close when safety door is enabled - this since the functionality is more or less the same and having two independent setting values might be confusing.
If the spindle supports the _at speed_ capability then the delay is for the alarm else it is a plain spinup delay with no RPM verification.

devknoll, 2022-11-12:
What if I don't have at speed support and I don't have a safety door, and just want a plain spin up delay (with no RPM verification)? My spindle takes a moment to ramp up to e.g. 18k, and after a tool change it will frequently try to start cutting immediately, before the spindle has reached the correct speed.

terjeio, 2022-11-13:
You have to use [G4P<delay in secs>](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g4) to add a delay before continuing. Typically this is added to the post processor.

And for those who have a door switch: the spin up delay is only added when the door is closed.

closed 2022-11-13 by devknoll

devknoll, 2022-11-13:
Makes sense, thanks!