grblHAL/core #222: G2/G3 Error 32: No axis words in plane (too strict rule)

kimstik, 2022-11-28:
I found potential issue in vanilla GRBL https://github.com/gnea/grbl/issues/1178

grblHAL have similar in gcode.c

```
                    if (!(axis_words.mask & (bit(plane.axis_0)|bit(plane.axis_1))))
                        FAIL(Status_GcodeNoAxisWordsInPlane); // [No axis words in plane]
```

terjeio, 2022-11-30:
Change to
```
                    if (gc_block.words.r && !(axis_words.mask & (bit(plane.axis_0)|bit(plane.axis_1))))
                        FAIL(Status_GcodeNoAxisWordsInPlane); // [No axis words in plane]
```
to fix?

kimstik, 2022-11-30:
your fix will make sense for R-mode only ?
original check is correct in Arc mode and non-valid in helix mode where in-plane begin and end coordinates is same .

terjeio, 2022-11-30:
> your fix will make sense for R-mode only ?

No, the check will only be performed for R format arcs, from [LinuxCNC specs](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3):

"It is an error if:
* both of the axis words for the axes of the selected plane are omitted"

For center format arcs the check is not required as error 33 will be reported if parameters are incorrect?

terjeio, 2022-12-04:
Ref _3.5.3.2 Center Format Arc_ in [NIST RS247NGC](https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=823374):

"It is an error if:
• X and Y are both omitted,
• I and J are both omitted."

etc...

Thus Grbl and grblHAL is compliant with the NIST spec, LinuxCNC is not? I do not have LinuxCNC instance to check with so cannot verify this. So for now I will leave the code as-is.

closed 2026-02-27 by terjeio