grblHAL/core #229: Reset in iosender prevents RPM from being sent to GS20 driver

LoganFraser, 2022-12-10:
I'll be as detailed as I can with this as it's an odd one. I'm using a GS10 VFD (same command codes) with the GS20 VFD driver. Whenever I press reset in IOSender, the controller no longer sends anything but 600rpm to the VFD over rs485. I know this because the pendant shows what rpm command is being sent, even though I'm selecting 10000 and CW in the sender. 

The VFD says 66Hz on the display, and IOsender says 3980 which is correct as I've set minimum speed to 4000rpm.

The only way I have found to clear the error is remove power from the controller, restart and close/reopen iosender. 

Let me know if there's any further details I can provide.

terjeio, 2022-12-11:
Which versions are reported by $I command (or Help > About in ioSender)?
What happens if you send M3S10000 from the MDI?

>  If I try to change the override speed or manually adjust the speed of the VFD more than 2-3 times the controller locks up 

Does this happen with the GS20 driver too?

I am currently on holiday and do not have my simulator available so sadly I cannot do any testing before I am back home.

LoganFraser, 2022-12-11:
Looks like lowering to 9600 baud has fixed all my issues so maybe it's a timing issue?  Not sure but at least it works now.  Fixed my issues with GS20 driver anyway, VFDMOD seems entirely unresponsive at 9600 baud.

terjeio, 2022-12-12:
> maybe it's a timing issue?

The GS10 is not ModBus specs compliant (similarly as the Huanyang v1) and require a longer [silent period](https://github.com/grblHAL/Plugins_spindle/blob/6b70b5cfe2f3573b0cb33bd178e08c4e29b76353/modbus.c#L69-L77) between messages? You can with try increasing it?

Are you using MODVFD with the GS10 spindle? If so I am not so sure that is possible...

LoganFraser, 2022-12-12:
I'll leave this open for a bit and if I find a solution I'll add it and close this out. My vfd worked fine at 38.4 connected to my PC via usb to rs485 using the same cable so it's weird I'm having these issues with grblhal.

closed 2022-12-30 by LoganFraser