grblHAL/core #239: Controlling RC Servos?

DrKlipper, 2023-01-09:
Hi !
Maybe I missed it, but is there any way of controlling RC Servos? I´m planning an Pen Plotter which uses a rc servo to move the pen up and down. Is there any way to do this in grblHal using M3/M5 commands? And if not are there plans to include this? 

And if this is not possible ... Can I use a stepper for this purpose?

terjeio, 2023-01-09:
> Maybe I missed it, but is there any way of controlling RC Servos?

Most drivers supports this, see [note 13](https://github.com/grblHAL/drivers) at the bottom of the drivers list for how to for those that do.

> And if this is not possible ... Can I use a stepper for this purpose?

Yes, and you may also use a solenoid - [here is a plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin/Solenoid%20spindle) that reduces the current after a short delay.

DrKlipper, 2023-01-09:
> Most drivers supports this
Oh, that is great to know. 

And which gcode commands will control this? Is it M3, M4, M5 like for Lasers?

terjeio, 2023-01-09:
M3/M4 + S value to move, M5 for back to "home" position.

closed 2024-04-16 by terjeio