grblHAL/core #242: Can't get spindle enable signal with spindle/modbus enabled

axg20202, 2023-01-22:
Hi. I have Phil B's breakout board for Teensy. On that board, the dust extraction trigger pins are linked to the spindle enable pins, so that gcode M3 turns on dust extraction. However, I'm controlling my spindle using modbus, not the spindle connections. Could this be the reason why the spindle enable pins on my board don't go high when the spindle is turned on? I'm currently unable to trigger the dust extraction because of this. Phil recommended I ask here in case it's a firmware issue related to using modbus.

terjeio, 2023-01-23:
> I'm controlling my spindle using VFD and modbus (both enabled in firmware and working) instead of the spindle PMW/0-10v connections. Could this be the reason why the spindle enable pins on my board don't go high when the spindle is turned on?

Yes, because the ModBus controlled spindle uses commands rather than the PWM related spindle signals (PWM, enable and direction) for control. Some users are switching between the PWM spindle (typically a laser) and a VFD - this means I cannot just change the code to set the enable signal when the VFD is active. Either a compile time option or a plugin has to be added to "fix" this. I'll come back to this when I am back home.

axg20202, 2023-01-23:
Thanks Terje. Understood. If there is a tweak I can make at compile time to make the enable go high when I start the VFD, that would be ideal.

andrewmarles, 2023-01-23:
It should be pretty easy to make your VFD output a signal when the spindle is active and just use that.

axg20202, 2023-01-23:
Possibly, but I'd much prefer a firmware solution that uses no additional wiring and let's me use the vac controls on the BOB instead, which also has a header for a manual vac override switch for general cleanup without the VFD running.

phil-barrett, 2023-01-23:
It would be great to have the ability allow any number of relay outputs be controlled by M3/M5.  Dust Collection is the current request but also air assist and exhaust for lasers and flood/mist for mills.  I could see a grbl setting "Link to Spindle on/off" that would have bits for various outputs. 0 - Spindle (always set), 1 - Mist, 2 - Flood, 3 and up for digital outputs.

The suggestion to use VFD to control the DC is a good one short term fix though.

axg20202, 2023-01-23:
I'm currently using the vfd to trigger the spindle water cooling pump. Could get messy adding another thing to trigger. Could be doable I guess, but a line or two of of code to keep the spindle controls live along with the modbus would be better if possible.

terjeio, 2023-01-24:
> It would be great to have the ability allow any number of relay outputs be controlled by M3/M5. 

I think I'll try to add such an option to the [fans plugin](https://github.com/grblHAL/Plugin_fans), will look into this soon. Sitting in AMS now on my way home, need a few days to get rid of jet lag and clear the issue backlog...

phil-barrett, 2023-01-24:
That makes sense.  

Schiphol, as airports go, it's pretty good.  But then it is still an airport...   Bon Voyage.

terjeio, 2023-01-26:
The fans plugin has now been extended with setting $483 for linking spindle enable to fan(s). If building the iMXRT1062 driver with a single VFD spindle enable and direction signals are repurposed as aux outputs allowing the use of the dust extraction pin via mapping (use $386 for fan 0). Note that if the fan is turned on manually (via M106 or the real time command) before the spindle is started it will _not_ be turned off automatically when it is stopped.

I may allow the use of the spindle enable signal to switch the fan later for dual spindle machines - if I can make it safe...

axg20202, 2023-01-27:
Thanks for adding this feature. Alas, I don't think it solves my problem. I'm using the dust TTL signal outputs of the BOB to trigger the relay to switch on dust collection. These pins are linked to spindle enable. I also want to use the manual dust collection pins on the board, which when shorted by a physical switch, also make the dust TTL output go high. This works if not using MODBUS (but I am using MODBUS). Unfortunately, linking the fan Aux 0 to spindle enable looks as though it makes Aux 1 (and probably its associated TTL signal out) go high with spindle enable, but I really need it to make the dust TTL out go high. That way I can have a single point of connection for dust triggering, which can come on with the spindle OR if I throw the manual switch.

phil-barrett, 2023-01-27:
Assuming you are set up to build grblHAL, one approach would be to make spindle enable an aux output.   I'm not sure how grblHAL knows the number of outputs.  Is it as simple as adding 

`#define AUXOUTPUT3_PIN      (12u) // AUX3
`
to T41U5XBB_map.h at line 102 (or so)?

Or, you could just changeAux0 to 12u in T41U5XBB_map.h like this?
`#define AUXOUTPUT0_PIN      (12u) // AUX0`

You'd lose Aux0 as an Output, though.

axg20202, 2023-01-27:
Hi Phil. Yes, I'm compiling my own.  My head is spinning tbh :-) I'm not understanding how the pin assignments for aux and enable relate to their physical outs, and also how the manual dust pins link to that. I don't get an enable signal using MODBUS, so thinking the above wouldn't work....but no idea.

terjeio, 2023-01-27:
If only one VFD spindle is selected for build spindle enable and spindle dir are remapped as [aux outputs](https://github.com/grblHAL/iMXRT1062/blob/96c34f701b3d5c9d64e9b4a5fbb1980c472b8441/grblHAL_Teensy4/src/T41U5XBB_map.h#L112-L115). Then fan 0 can be mapped to Aux3 (which is routed to the dust output if I am not mistaken) with `$386`.
Use the `$pins` command to see the active pin mappings.

axg20202, 2023-01-27:
$pins
[PIN:14,Emergency stop]
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:23,Y limit min 2]
[PIN:22,Z limit min]
[PIN:36,Aux input 0,P0]
[PIN:30,Aux input 1,P1]
[PIN:34,Aux input 2,P2]
[PIN:35,Aux input 3,P3]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:8,Y2 step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:9,Y2 dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:38,Y enable]
[PIN:19,Flood]
[PIN:18,Mist]
[PIN:31,Aux out 0,Fan 0]
[PIN:32,Aux out 1,P0]
[PIN:33,Aux out 2,P1]
[PIN:12,Aux out 3,P2]
[PIN:11,Aux out 4,P3]
[PIN:0,RX,Modbus]
[PIN:1,TX,Modbus]

axg20202, 2023-01-27:
Pins above. I then changed fan 0 out to AUX 3, hit save, and did a hard reset. It sent the board back to 'unable to communicate'

axg20202, 2023-01-27:
Calling it a day today. Will revisit this tomorrow. Thanks for your help guys.

phil-barrett, 2023-01-27:
You probably need to go and reset your input inversions in grbl settings.  

The manual DC header signal and the spindle enable signal (same one that goes to the En screw terminal) go to the inputs of an OR gate. The output of that gate goes high if either input is high.  The output of that gate goes to the VAC/DC output headers.

In the map file (T41U5XBB_map.h) in src, you can change actual pin assignments. Spindle Enable is on the Teensy 4.1 pin 12.  The 3 Aux outputs are on pins 31, 32 and 33.  So, if you change one of those to 12 it will show up on the spindle en and VAC/DC outputs.

terjeio, 2023-01-27:
I have now added a checkbox for _Add PWM spindle_ to the _General tab_  in the Web Builder. If a VFD spindle is selected in the dropdown above and _Add PWM spindle_ is left unchecked pin 12 (spindle enable) is remapped as Aux out 3 and pin 11 (spindle dir) as Aux out 4.
Mapping fan 0 to Aux out 3 and linking it to spindle enable (the internal signal, not the pin) with `$483` (in the Spindle settings group in ioSender) works for me. 

[image]

If _Add PWM spindle_ is checked no remapping is done and the PWM spindle can still be used.

phil-barrett, 2023-01-27:
That's great!  Thank you!

axg20202, 2023-01-28:
Brilliant! Is this relatively easy to accomplish when self-compiling too?

terjeio, 2023-01-28:
> Is this relatively easy to accomplish when self-compiling too?

Yes, enable the VFD you want with uncommenting `VFD_ENABLE` and set it to the [appropriate id](https://github.com/grblHAL/Plugins_spindle/blob/master/shared.h) in _my_machine.h_ (do not set it to -1) and leave `N_SPINDLE` at 1 in [grbl/config.h](https://github.com/grblHAL/core/blob/dbc170a571054f7019a1f10840591b7ff9f1b4cd/config.h#L58).

axg20202, 2023-01-28:
Great. Thanks very much for your help with this.

axg20202, 2023-01-28:
Just to check, I just need to recompile with the above. Fans plugin not needed? I've just recompiled and the dust TTL out isn't going high with spindle start, and now it also doesn't go high when I close the manual switch either. I have updated the modified submodules.

terjeio, 2023-01-28:
> Fans plugin not needed?

Still needed, it provides the $-commands and the binding to the internal spindle enable signal.

axg20202, 2023-01-28:
Aha, thanks. Probably needs a conditional setting in the Web builder - I don't think ticking the add spindle box forces the fan plugin selection.

terjeio, 2023-01-28:
> I don't think ticking the add spindle box forces the fan plugin selection.

Ticking the _Add PWM spindle_ box enables the "normal" PWM signals even when a VFD is selected, and remapping to Aux outputs is _not_ done. The fans plugin is enabled in the _Plugins_ tab:

[image]

axg20202, 2023-01-28:
OK, got it. I think it just killed my Teensy. This was after compiling with add spindle ticked but no fans plugin. I am guessing that closing my manual dust switch sent 5v to it. Pretty miffed.

phil-barrett, 2023-01-28:
Hmmm, I don't see how that would kill your teensy.   The VAC/DC header is actually powered via 3.3V.

Is it completely dead?  If you try to talk to it via a terminal app does it do anything?  Also, what happens when you press the button on the teensy?

axg20202, 2023-01-28:
OK. Maybe it was a static jolt from taking it in and out of my enclosure. I guess they are quite sensitive to esd. Looks pretty dead. Doesn't get picked up on usb, and no red led. Pressing or holding button does nothing. doesn't work in the BOB.

phil-barrett, 2023-01-28:
Ugh. Sorry to hear that. It sucks. I've blown several Teensies. Learned the hard way they don't like 12V or getting shorted out.

axg20202, 2023-01-29:
New Teensy shoved in with the new firmware. Fan checkbox ticks on and off with Spindle on/off. TTL out for dust collection still isn't going high though. Is there another grbl setting I need to make to link things up?

terjeio, 2023-01-29:
`$386` is set to map Fan 0 to pin 12? Check with `$pins` command.

axg20202, 2023-01-29:
[PIN:12,Aux out 3,P2]

terjeio, 2023-01-30:
> [PIN:31,Aux out 0,Fan 0]

This is incorrect, map it to Pin12 with `$386=3`

axg20202, 2023-01-30:
Thanks Terje.  I just made the change with hard reset. Still not working unfortunately.

[PIN:14,Emergency stop]
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:23,Y limit min 2]
[PIN:22,Z limit min]
[PIN:36,Aux input 0,P0]
[PIN:30,Aux input 1,P1]
[PIN:34,Aux input 2,P2]
[PIN:35,Aux input 3,P3]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:8,Y2 step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:9,Y2 dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:38,Y enable]
[PIN:19,Flood]
[PIN:18,Mist]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:12,Aux out 3,Fan 0]
[PIN:11,Aux out 4,P3]
[PIN:0,RX,Modbus]
[PIN:1,TX,Modbus]

axg20202, 2023-01-30:
OK, will do further testing. I'm really hoping that something on the BOB didn't get taken out when my Teensy died.

axg20202, 2023-01-30:
Following some testing, looks like spindle enable pins don't go high when spindle is turned on, and nothing goes to the vac TTL out either. Manual dust trigger doesn't cause vac TTL out to go high when switched either,

axg20202, 2023-01-30:
Just as an experiment, I tried the webbuilder firmware. Same result but different pin assignments.
[image]
[image]

[PIN:14,Emergency stop]
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:22,Z limit min]
[PIN:36,Aux input 0,P0]
[PIN:30,Aux input 1,P1]
[PIN:34,Aux input 2,P2]
[PIN:35,Aux input 3,P3]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:8,Y2 step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:9,Y2 dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:38,Y enable]
[PIN:12,Spindle on]
[PIN:11,Spindle direction]
[PIN:19,Flood]
[PIN:18,Mist]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,Fan 0]
[PIN:0,RX,Modbus]
[PIN:1,TX,Modbus]
[PIN:13,Spindle PWM]

axg20202, 2023-01-30:
Just tested another TTL out (flood) and that's working.

axg20202, 2023-01-31:
Still no joy getting this working. Hopefully it's just a pin assignment that's tripping me up.

axg20202, 2023-01-31:
@phil-barrett   Is there a fault condition that would damage only the vac/dc ttl output but leave the others working? I'm now wondering if this was taken out when my Teensy died. Closing the dust/vac manual trigger isn't making the vac TTL go high. I tried a different TTL out and that is working.

phil-barrett, 2023-01-31:
It is possible if vac output was shorted to ground at some point.  Also, inspect the board around the area of IC16, IC17 and IC19.  

If IC16 was damaged, that would prevent Vac from working at all.

axg20202, 2023-01-31:
Looking at the schematic, maybe my external relay board drew too much current from the TTL out?  I wonder if I should use the relay screw termals on the BOB instead?

phil-barrett, 2023-01-31:
It's possible but those relay boards draw very little current - usually around 2 mA. Typically, they use a 2700 ohm input resistor.

Unfortunately, if the TTL driver is blown, the relay coil output is too because the relay driver takes it input from the TTL driver.

axg20202, 2023-01-31:
Can't see any damage anywhere on the board.  So is replacing IC16 a likely candidate for a fix?

phil-barrett, 2023-01-31:
Maybe but if it was an output short, IC17 is probably the issue.  You can determine if IC16 is bad by looking at the output on pin 4 while shorting the Vac control header.  It should register something like 3.1-3.3 VDC.
[image]

If it doesn't, then it is bad. If you have soldering skill, you probably can replace it. But, I suspect the problem is IC17, 74ACHT541.  You might want to verify that.  

Rather than replacing IC17, you could just bodge to a different output pin, say aux1 (which is right next to vac) on IC17. First verify that aux1 works.  See the picture below.  Cut 2 wires (Vac and Aux1) and then connect IC16 output to the Aux1 input on IC17. Then you use aux1 in place of Vac.
[image]

axg20202, 2023-01-31:
Great. Thanks Phil. And sorry to you both for the million questions.

phil-barrett, 2023-01-31:
Not a problem.  Let me know what you find.

axg20202, 2023-02-01:
With the board on the bench, powered by 5v terminals, no Teensy installed, I get 0v at the Y terminal of IC16 when the vac header is shorted. I assume I don't need the Teensy installed for this test. I also see 0v at the Vcc of IC16.
The Y pin has a resistance of only 1.6 to ground. Was kind of expecting to see voltage at Vcc even if the IC is dead.

phil-barrett, 2023-02-01:
You need to have the teensy installed.  IC16 is powered from 3.3V which comes from the Teensy.

axg20202, 2023-02-01:
Hah, Doh!  Well, with Teensy powered up, looks like IC16 is dead. Vcc is getting 3.3v but 0v at Y with/without vac header shorted. Will replace it and restest.

axg20202, 2023-02-02:
Replaced IC16 - all working again :-)

axg20202, 2023-02-05:
I just wanted to report back that the TTL outs and external relays are now working perfectly after the fun of reparing and rewiring the board back into my enclosure. I added the EPROM while I was at it. Vac now comes on with VFD, and I can also trigger a solenoid at my compressor to control air. Thanks both for your great support.

phil-barrett, 2023-02-05:
That is fantastic! Glad you have it up and running.

closed 2024-04-16 by terjeio

axg20202, 2024-11-27:
Sorry to revive this old thread, but I have since added a PWM spindle in addition to my VFD spindle. The dust extractor no longer comes on with the VFD spindle. Is there a workaround? I'm guessing it isn't working because I am actually using the PWM selection in the Webbuilder now. The dust extractor does come on with the PWM spindle, but not the VFD spindle

terjeio, 2024-11-28:
> Is there a workaround?

Yes, you can use the [fans plugin](https://github.com/grblHAL/Plugin_fans/) to control the output, pin 12 if I am not mistaken.

The new [eventout plugin](https://github.com/grblHAL/Plugins_misc/#bind-events-to-aux-output-ports) can also be used but it is not yet available from the Web Builder.

axg20202, 2025-04-12:
Sorry to revive this yet again, but I'm still having issues with this. I'm unable to link Fan and dust extraction (Aux output 3), because Aux output 3 is no longer showing in my pins, possibly because spindle enable is now holding onto pin 12 (I have 2 spindles, one is PWM (actually a laser), the other is Modbus VFD) - I want the spindle on/off linked to Aux output 3 using either spindle. The current pins assignment means that the triggering of Aux output 3 is erratic. It works when PWM spindle is selected, but when Modbus spindle is selected, it appears to be sending a PWM signal to Aux output 3, sending my dust extractor relay crazy. 

Is there a workaround?  I'm currently compiling myself rather than using Webbuilder, so I can tweak pins if needed, but not sure what to change for this issue.

[VER:1.1f.20250409:]
[OPT:VNMHSL2,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,EXPR,ATC,SED,YM,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:FLASH]
[FREE MEMORY:480K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:250411]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[AUX IO:7,4,0,0]
[CLUSTER:16]
[PLUGIN:LightBurn clusters v0.06]
[PLUGIN:FS stream v1.00]
[PLUGIN:FS macro plugin v0.17]
[PLUGIN:MODBUS v0.18]
[PLUGIN:HUANYANG VFD v0.14]
[PLUGIN:Fans v0.15]
[FANS:0]
[PLUGIN:Laser PPI v0.09]
[PLUGIN:SDCARD v1.23]
[SPINDLE:Huanyang v1]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[HOME:0.000,0.000,0.000:0]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[SPINDLE:1|0|2|SDRVE|Huanyang v1|1200.0,24000.0]
[SPINDLE:0|1|0||PWM]

$Pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:21,Y limit min]
[PIN:23,Y limit min 2]
[PIN:22,Z limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,P3]
[PIN:41,Aux in 4,P4]
[PIN:28,Aux in 5,P5]
[PIN:29,Aux in 6,P6]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:8,Y2 step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:9,Y2 dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:38,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:13,Spindle PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:31,Aux out 0,P0]
[PIN:32,Aux out 1,P1]
[PIN:33,Aux out 2,P2]
[PIN:11,Aux out 4,P3]
[PIN:0,RX,Modbus]
[PIN:1,TX,Modbus]

axg20202, 2025-04-13:
Never mind - found a solution. In the board map I changed the PWM spindle enable reference from AUX OUT 3 to AUX OUT 2 (which I don't use), which freed up AUX OUT 3 (dust collection), so that it now appears in $Pins. I then used the EventOut plugin to link Spindle enable with Aux Out 3.  Fans plugin no longer needed.