grblHAL/core #244: Computer dos not recognise board after activating SD-Card on SKR ProV1.1

asulonn, 2023-01-30:
Hello, 

I'm using the "Web Builder v0.9" for compiling firmware for an SKR ProV1.1 board.
When i configurate the firmware without an SD card, everything (testing without motors attahced with IOsender) seems to be ok. 
I can jog the axes through the IO-sender program.

However, when I build the firmware with "SD Card" enabled under "Plugins" in the configurator, the computer no longer recognises the board when it is plugged in. It shows the following message:
[image]

Is there something i can do about this? I would like to use the SD-card capability of the board.

This is the board configuration Json from the configurator (with SD Card enabled in this case): 
`{"driver":"STM32F4xx","URL":"https://github.com/grblHAL/STM32F4xx","board":"BOARD_BTT_SKR_PRO_1_2","symbols":["USE_HAL_DRIVER","STM32F407xx","HSE_VALUE=8000000","HAS_BOOTLOADER","USB_SERIAL_CDC=1","X_AUTO_SQUARE=1","PROBE_ENABLE=1","SDCARD_ENABLE=1"],"docker_instance":"","grblHAL":"webBuilder"}`

terjeio, 2023-01-30:
The pin assignments for the SD card are wrong - likely never tested by the user that posted the pull request for the board.
I'll attempt a fix soon, but since I do not have a board available I leave it to you to test.

terjeio, 2023-01-30:
Looking closer: the pins are mapped to the EXP2 header - for an external SD card breakout?

asulonn, 2023-01-31:
Thanks for your quick reply!

At first it seemed obvious to me that it would have been the SD-card reader that was present on the board itself, but after some research (for instance: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/35 ) it seems that there is something quircky about that on board SD-card slot.  Maybe because the SD-card slot is also used for loading new firmware on the processor....
It seems that most people use an external SD-card breakout for storing G-code files.
Like many other Marlin boards, the SKR Pro board usually has a display with SD-card reader connected to it, meaby that's why it is like that...

I will try and report back what happens when i hook up an external SD-card breakout.

asulonn, 2023-02-01:
I tried connecting the SD card extension that is on a "reprapdiscount smart controller" (the one that is normally connected to the SKR Pro), but it makes no difference. After that i tried to connect another separate SD card extension, but that still the same message on the computer when i connect the board. 

It seems like [http://svn.io-engineering.com:8080/](url) is down. (I wanted to create some other configuration to see if that made any difference)

terjeio, 2023-02-02:
The Web Builder is up again now - had a power outage yesterdat. There will be another one today.

> ... still the same message on the computer when i connect the board.

Could be something crashing the controller. Are you programming with a debug probe and can halt the program?

terjeio, 2023-02-02:
The pin mapping for SD card chip select is incorrect, it messes up a USB pin. I have committed a fix so you may try again now.

asulonn, 2023-02-03:
Progress: After your change, the board connects to computer and IO sender now when enabling the SD-card in the Web Builder!

However when i connect the "reprapdiscount smart controller"-SD card extension, the board does not seem to recognise it.
=> when using the command "$FM", the board returns "error:60 - SD Card mount failed."

I did not check if the pins of the "reprapdiscount smart controller"-SD card extension match the pins in the skr_pro_v1_1_map.h (i just plugged it in). 
What i want to try next is the separate SD card break out board again, but i'm having a hard time to figure out which pinnumbers mean what on the board.
I tried connecting the separate SD-card read earlier (before your fix), but since i did notice the incorrect pins that you found out, i must have not been doing it right.
What i find on the web for the SKR pro is this:
[image]

But the naming in your skr_pro_v1_1_map.h is this:
```
// Connect to the  EXP2 Header
#define SD_CS_PORT                  GPIOB
#define SD_CS_PIN                   12
#define SPI_PORT                    2 // GPIOB, SCK_PIN = 13, MISO_PIN = 14, MOSI_PIN = 15
```

Where do you find the translation between the two? Does "GPIOB" mean all the pins in the picture with "PB" in front of the number?

terjeio, 2023-02-04:
> Where do you find the translation between the two?

Between GPIOB and PB? I guess that is by convention. GPIOB is used by the STM32 framework to reference port B, PB as a shorthand for labeling pins in the port.

> Does "GPIOB" mean all the pins in the picture with "PB" in front of the number?

Yes.

asulonn, 2023-02-05:
> > Does "GPIOB" mean all the pins in the picture with "PB" in front of the number?
> 
> Yes.

This seems very logical! :-)
What threw me off i think, was the fact that is tused to be 
`#define SD_CS_PORT                  GPIOA`
But there is no GPIOA on EXP2
Which should have triggered "this is wrong", but instead triggered: "i don't think i understand pinout numbers' :-)

I tested the external SD card reader, but it doesn't work either. The board returns "error:60 - SD Card mount failed."

Do you have tips about what i can do for more debugging?
(i'm using PlatformIO atm)

asulonn, 2023-02-05:
I checked the pin out for the "reprapdiscount smart controller", it seems the same as described in the driver

[image]
 
While testing I also connected the second connector, because it held the 3.3V
[image]

To make sure the hardware was ok, i flashed Marlin on the board, the marlin firmware does read the SD-card files.

So the problem seems to be something in the firmware. As i said earlier, if there is something i can do, let me know!

terjeio, 2023-02-05:
Are you compiling with more than 3 axes? If so PB15 is remapped as cycle start pin at line 147-148 - likely making problems for the SPI port.
I am using a F446 ST Morpho controller for testing - SD card is working with that, via SPI1, so the SD card code is ok. I cannot see anything wrong with setup for SPI2 except that the cycle start pin may cause problems.

asulonn, 2023-02-08:
I only have three Axis. I had Y-axis set to "Auto-squared", which makes it 4 motors, but when i changed Y to "Single motor", the effect was the same.
I also tried without connecting the "EXP1" cable, to make sure the pins on that connector did not interfere with anything. (I just connected the GND and 3.3V for power), but that did not make any difference either.

Is there any way i can debug this? Have some read out of what happens? Do i need to install the "STM32Cube" software to do this?

terjeio, 2023-02-08:
Do you have access to an oscilloscope?

asulonn, 2023-02-08:
No, atm I don't.
I only have a usb 24MHz8Channel logic analyser. (which I have never used, and don't really know how it works yet ^^) If I could see something with that, i'm willing to give that a try and learn how to work with it!

terjeio, 2023-02-08:
Ok, even better with a logic analyzer. At startup there should be a long train of clock pulses (SCK) with the MOSI and CS pins held high. If these are not there the controller has likely crashed for some reason earlier in the startup sequence.

asulonn, 2023-02-13:
Sorry for the late answer, it took a while before i had time to figure out how the logic anayliser (and software) worked.
I have managed to make some traces.

The image below shows the signals when i try to access the SD-card in IO-sender). A long pulse train on SCK, MOSI stays high, but it seems that CSEL goes low instead of goes high (or should this signal be low to be active?)
[image]

I also made a trace with MISO connected to the analyser, but this seemed to influence the signals. The signals are simular to what they were in the previous trace, but the pulse train seems to be a lot shorter. The image below, also shows the moment at which the board was plugged in. The only thing that happens at power up than is "MISO" is set high.
[image]

Each time i run the "$FM" command, this signal flow happens.
I only checked the signals with the "reprapdiscount smart controller"-reader, not yet with the other SD-card reader i have

It seems as though the controller tries to talk to the SD-card reader. Is it normal that the CSEL signal is inverted?

terjeio, 2023-02-13:
CS is active low, so that is correct. When $FM is issued, a [long clock train with CS and MOSI high](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Src/diskio.c#L116-L132) is sent to the card to initialize it to single lane mode. Then [some commands](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Src/diskio.c#L341-L384) are sent to it to check what type of card it is and proceed from there if it responds in a sane way. It is in this function I have done most of my debugging, if this fails you have communication issues.

asulonn, 2023-02-15:
Ok thank you for the information! I will have to do some more debugging.
I will also try use logic analyser when the Marlin firmware is loaded. Check if the communication with the SD-card reader is different.

This is off topic, but I am plannig to use this board/machine stand alone. I don't want to have to have a computer hooked up all the time. Is it possible to hook up an ESP32 runnig ESP3D to the board? I read that the new unreleased version of ESP3D (V3.0) supports GrblHal. But when i hooked up an ESP32 running ESP3D v3.0 to the UART port, the ESP3D indicated it did not find the GrblHal controller.. 
[image]
Or should i hook it up to the "wifi" connector (I did not try that yet)
[image]
If i should open a new topic for this, let me know!

terjeio, 2023-02-15:
> But when i hooked up an ESP32 running ESP3D v3.0 to the UART port, the ESP3D indicated it did not find the GrblHal controller..

Did you [disable the USB port](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Inc/my_machine.h#L56) by commenting out this line? And is the UART port routed to the correct pins? You can check with the `$pins` command.

asulonn, 2023-02-15:
> CS is active low, so that is correct. When $FM is issued, a [long clock train with CS and MOSI [...]

I tried the Marlin firmware again, and with that firmware the SD-card reader actually works. 
I tried to compare the code Marlin uses to[ init the SD-card reader ](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/849cd760735c36141cc6839d5bb1e8c0bb869ff2/firmware/SKR%20ProV1.1-Marlin_v2.0-Demo/Marlin/src/sd/Sd2Card.cpp#L259) to the code link you sent, and it looks similar to your code. It is written differently, but it seems the same things happen. (as far as my inexperienced eye can see...) Maybe you spot something different...

Like i said i will also try to analyse the communication with the logic analyser later.

asulonn, 2023-02-15:
> Did you [disable the USB port](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Inc/my_machine.h#L56) by commenting out this line?

No I didn't. I have only built firmware for the board by using the web builder. 
Can I do it in the web builder by setting "Connection" to "(Virtual) UART"? 
[image]
Does this mean I can no longer connect it to the computer (IOSender) over USB? This would be ok if I can get the ESP3D to work.

 > And is the UART port routed to the correct pins? You can check with the `$pins` command.

I will check!

I'm exited to try and find out! It would be awesome if that could work!

PS: thanks for all your replies! It is much appreciated! I try my best to find out what I can, but I'm kind of out of my depth for many of the things :-|

terjeio, 2023-02-15:
> Can I do it in the web builder by setting "Connection" to "(Virtual) UART"?

Yes.

> Does this mean I can no longer connect it to the computer (IOSender) over USB?

Correct. However, if ESP3D is updated to support the [MPG & DRO interfacel](https://github.com/grblHAL/core/wiki/MPG-and-DRO-interfaces) both can be used.

> I try my best to find out what I can, but I'm kind of out of my depth for many of the things :-|

Hopefully we can resolve the SD card issue if you are persistent and do not give up. And it is great to learn some new stuff, no?

asulonn, 2023-02-17:
On connecting to the ESP3D:

> Did you [disable the USB port](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Inc/my_machine.h#L56) by commenting out this line?

I built a version with the web builder with the "(Virtual) UART" active. It no longer connects to usb (computer), but it does not connect to the ESP3D either.
This might have to do with the UART port:
> You can check with the `$pins` command.

It does not show UART pins.
[image]

I executed the `$pins` command while still on USB mode, since I could not connect on UART. Do the pins it only show in the list on UART mode? I tried to look in the software for what pins could be the UART, but the i could only find UART pins for the trinamic drivers.
I tried connecting the ESP32 to the UART-labeled pins on the board, but as said earlier, it did not want to connect. This might be an ESP3D problem, but I don't know with what else I can let the board communicate.
[image]

> Hopefully we can resolve the SD card issue if you are persistent and do not give up. And it is great to learn some new stuff, no?

I agree, it's just that sometimes you want a little win to stay motivated to keep going :-)

terjeio, 2023-02-18:
The default UART is routed to J50/TFT display ([PA9/PA10](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Src/serial.c#L287-L317)), can be changed to the J48/UART3 header ([PD8/PD9](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Src/serial.c#L351-L381)) by uncommenting [this line](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Inc/btt_skr_pro_v1_1_map.h#L39) - the UART3 header cannot be selected in the Web Builder though.

>>    You can check with the $pins command.

> It does not show UART pins.

My bad, it is a chicken/egg situation - when USB is compiled in the UART port/pins are excluded from the build.
To have both at the same time MPG mode support has to be added to the board map. An interrupt capable pin is needed for that, [here is an example](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Inc/st_morpho_map.h#L150-L158) for how to do that. The W1/ESP8266 header is a candidate for MPG mode - if _serial.c_ is expanded for UART6 (PC6/PC7).

terjeio, 2023-02-19:
FYI commit with option to use UART6 just up.

asulonn, 2023-02-19:
Hi, I didn't have very much time today, so i only did a quick test:

When I try to build the firmware with the web builder now: (all settings to default, except "Connection" set to "Uart")
[image]

I tried to connect to the UART on PA9/PA10, but ESP3D gave no connection error
[image]
I can do more testing later when I have more time.

terjeio, 2023-02-19:
Web Builder error is due to me commiting a config file with number of axes set to 4, fixed now.

> I tried to connect to the UART on PA9/PA10, but ESP3D gave no connection error

Did you connect RX to TX and TX to RX?
If you have an UART <> USB breakout you may try with that and connect with a terminal program to check the port.
Or check the controller TX pin for output - the welcome message is sent on startup or a reset.

asulonn, 2023-02-20:
Ok, so I did some more testing!

I followed your advice to check with a UART to USB and a logger. The problem seems to be in the ESP32. The GrblHal board outputs the welcome message as you mentioned. I tried some more interaction like the "help" and the "$$" and that all seems to work! Which is nice!
I will try to find out what is wrong with the ESP32 and/or open a ticket on the ESP3D Github repo. 
I also connected toe UART-to-USB to the ESP32, but it only output some data about the WIFI connection status. So maybe I need to connect to other pins... As I said, I will try to find out :-)
I'm going to do that first and afterwards continue the search for the SD-card issue.

BTW: I did these tests with a new built from the web builder. Your fix worked, the builder no longer stopped on an error! (y)

terjeio, 2023-02-24:
FYI I just got a Nucleo-144 with a F446 that has the same pins as the SKR uses for the SD card broken out. The SD card works with USB or UART 3 enabled (compiled with the SKR map), IMO this means the code is ok.

asulonn, 2023-03-01:
Ok. After your message I was triggered to do some more testing. :) 
I tried the SPI sd card break out board with an esp32. To make sure it worked. The reader worked. I could read and write from/to the sd card.
After that I connected the sd card break out board with the same sd card in it to the skr pro board, but it stil did not work. I checked the pins again, but all seems to to be connected ok.

terjeio, 2023-03-03:
Are you able to set a breakpoint [here](https://github.com/grblHAL/STM32F4xx/blob/b9d0c7d5d50198463084a748f48048679d8fe270/Src/diskio.c#L377)? Or modify the code to output a message?

At startup the card is probed at a low clock frequency - if this succesful, indicated by a non-zero `ty` value, then this will help deciding how to proceed.

closed 2023-08-14 by terjeio