grblHAL/core #245: Помогите с кодом.

logvinov-ivan, 2023-02-01:
Хочу сделать швейную машинку для вышивания.. 
 как подключить (датчик  индуктивности или датчик холла )для установки поднятия иглы ? и прописать его в прошивке 
  чтоб по команде в gcode m555 двигатель остановился и провернулся до замыкания датчика . ?!?
-----------------------------------------------------
I want to make a sewing machine for embroidery..
  how to connect (inductance sensor or hall sensor) to set the needle lift? and register it in the firmware
   so that, at the command in gcode m555, the engine stops and turns until the sensor closes. ?!?

terjeio, 2023-02-02:
You will have to write a plugin for that.
If your board has registered auxillary inputs  (preferably interrupt capable?) you can use the [ioports API](http://svn.io-engineering.com/grblHAL/html/ioports_8h.html) to read the input. [This template](https://github.com/grblHAL/Templates/blob/master/my_plugin/Macros%20bound%20to%20aux%20inputs/my_plugin.c) reads inputs via interrupts, you may also poll the input while waiting.
Add M555 handling to the plugin in a similar way as [[this example](https://github.com/grblHAL/Templates/blob/master/mcodes.c](https://github.com/grblHAL/Templates/blob/master/my_plugin/Stepper%20enable%20control/my_plugin.c)).

logvinov-ivan, 2023-02-02:
stm32 3 axes are used, there, there are also outputs for 2 axes. Is there any instructions on how to use and configure these plugins. I'm not good at programming. it turns out I take a pin with stm32, enter an additional "M * command and when executing Gcode, the command will be executed according to the written conditions and after that it will start executing the following lines of gcode

terjeio, 2023-02-02:
> Is there any instructions on how to use and configure these plugins. 

For the templates instructions are usually in the .c file header. For "active" plugins in hosted separate repositories in the associated readme. Some general informatin can be found in the [Wiki](https://github.com/grblHAL/plugins#example-and-template-plugins).
For the templates and any you write yourself just copy (or create) a file in the source folder (where _main.c_ is located) and recompile - the only requirement is that a function named _my_plugin_init_ is present, it will be called on startup.

`M555` is new for me, is this generally used by embroidery machines? If so do you have any links to information?

logvinov-ivan, 2023-02-02:
no, I wrote m555 as an example to make it clear in gcode , otherwise you can use any command

terjeio, 2023-02-02:
> I wrote m555 as an example

Ok, then you can first try with [M66](https://linuxcnc.org/docs/2.5/html/gcode/m-code.html#sec:M66-Input-Control) - wait for input, use the `$pins` command to check if P0 (Aux0) is available.
E.g. `M66P0L2Q1800` will stop execution and wait for a falling edge with a timeout of 30 minutes.

logvinov-ivan, 2023-02-02:
by gcode, the engine will turn off and m66 will then wait for the sensor to work on the pin ?! so the engine should reduce to 10 revolutions and rotate until it finds a signal from the sensor, after which it continues to move .. according to the gcode program

terjeio, 2023-02-02:
> by gcode, the engine will turn off

Motion will stop and wait for the sensor. If you are using the spindle signal for something and want to stop that then M66 will _not_ disable the spindle signal.

> so the engine should reduce to 10 revolutions and rotate until it finds a signal from the sensor,

Is this in the gcode? And is it the same sensor as for the first step?
Do you have example gcode that shows the sequence? With comments if possible.

logvinov-ivan, 2023-02-02:
there is no gcode, it will just execute commands, after the execution has passed, the engine should spin slowly to find the sensor and stop, it turns out that the needle will come out of the fabric, and will not tear the fabric after gcode continues to work.

terjeio, 2023-02-02:
> there is no gcode, it will just execute commands

You lost me there - execute what kind of commands? grblHAL understands a number of G- and M-codes plus some $ system commands.
It _is_ possible to add parsers for other command sets, as I did for the[ HPGL plotting language](https://github.com/grblHAL/Templates/tree/master/my_plugin/hpgl).

How are the commands generated? By a program? Made by you or generally available?

logvinov-ivan, 2023-02-02:
publicly available.
yes, everything is like a plotter, just add parsers so that the electric motor can tighten the needle, that is, it will turn until the sensor is triggered, stop and then begin to execute gcode., this is in order not to take off the needle and tear the fabric, you need to track when moving the axes, the upper position of the needle.

logvinov-ivan, 2023-02-02:
What will happen.
x-axis
y-axis
hall sensor motor
through Ink / Stitch I will generate a code for embroidery.
I will run the code through grblhal
the axes will start moving along the program. after the program is completed, you need to move, but since the needle may remain at the bottom and when moving to another point starts, it will lead to a tear in the fabric and a broken needle. that's why I want to track the position of the needle before moving to another place. To allow movement, the sensor must allow movement, that it is closed and you can perform movement or further execute gcode.

terjeio, 2023-02-02:
Do you have any links?

> just add parsers so that the electric motor can tighten the needle, that is, it will turn until the sensor is triggered, stop and then begin to execute gcode

So there is defined sequence of actions that has to be taken when executing the new M-code, not only a simple wait?
Perhaps a bit similar to the [toolchange](https://github.com/grblHAL/core/blob/master/tool_change.c) sequence(s)? Motions, wait for probe input, motions, wait for input (cycle start), motions...

logvinov-ivan, 2023-02-02:
> У вас есть ссылки?
> 
> > просто потом добавили парсеры, чтобы электродвигатель мог подтянуть иглу, то есть она крутилась до срабатывания датчика, останавливалась и начинала запускать gcode
> 
> То есть определена последовательность действий, которые необходимы при выполнении нового М-кода, а не простое ожидание? Возможно, это немного похоже на последовательность(и) [смены инструмента](https://github.com/grblHAL/core/blob/master/tool_change.c) ? Движение, ожидание ввода датчика, движение, ожидание ввода (начало цикла), движение...

what you need, but here the movement is performed by raising the axis of the stepper motor, and I have an asynchronous one, to find the sensor you need low engine speeds, PWM control.

logvinov-ivan, 2023-02-02:
And how to collect all this in the firmware ?! is it possible with mega2560 and shield ramps 1.4?

terjeio, 2023-02-02:
Looking at [this video](https://www.youtube.com/watch?v=0lERsPc5gEk) it seems to me that want you want is to run the retract step (pulling the needle up) until a sensor is struck and then continue?
This could be achieved by using a [probing sequence](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38) for the retract?

> is it possible with mega2560

Sorry no, grblHAL does not run on 8 bit processors.

> shield ramps 1.4

[Some boards supported by grblHAL](https://github.com/grblHAL/Controllers) could use this if Ramps 1.4 is similar to 1.6?

logvinov-ivan, 2023-02-02:
I'll take stm32 well. yes, I watched the video, a stepper motor is used there, and with it there is no sick speed of work, for this the hall sensor will be used to determine and raise the needle to the top and continue working on gcode further

terjeio, 2023-02-02:
You are using translation software to english? I am having a bit of trouble understanding everything...

You refer to PWM above - does this mean you want to use a RC servo motor for driving the needle?

logvinov-ivan, 2023-02-02:
Yes, I use google translator.
asynchronous motor with a hall sensor. I don’t have a servo motor

terjeio, 2023-02-03:
You control the motor with the PWM and spindle direction signals - using M3/M4 and S for speed?
The Hall sensor detects when the needle is in the "home" position (moved up)? And another one when reaching the lower position?

logvinov-ivan, 2023-02-03:
Yes, I would like to use these M codes. I need only one sensor to determine the upper position of the needle, so that the axes move to another location and the code continues to execute.

terjeio, 2023-02-06:
Have you made the machine already or is it in the planning stage?
Is it a design used by others? If so do you have any links? Pictures?
What kind of driver is used for the asynchrounous motor?
Would you not require a sensor for the lower position as well?
...

These are some of the questions that comes to mind and that has to be answered before any attempt to write a plugin.

logvinov-ivan, 2023-02-08:
Yes, under development ... I'm waiting for details from China.
I still don’t really know what I need and what sensors,
I still have problems with the firmware
1.where to see driver connections to stm32 in grbl hal firmware
2. compiled via ide, included in the file define USB_SERIAL_CDC =1, but usb does not work, how to enable it? so that after the firmware it was possible to use stm32 via usb.
3. Through the site, the firmware is compiled well, everything is buzzing.
  4. ioSender 2.0.41 writes motion blocking
like all settings are off.
[VER:1.1f.20230129:]
[OPT:VNMSLW,35,512,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,TC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F103C8]
[DRIVER VERSION:230125]
[BOARD:CNC 3040]
[KINEMATICS:WallPlotter v2.00]
error:18 - Reset asserted\

error:8 - '$' command cannot be used unless controller state is IDLE. Ensures smooth operation during a job. -what should be done? I don't understand

terjeio, 2023-02-08:
> 1.where to see driver connections to stm32 in grbl hal firmware

In the [board map file](https://github.com/grblHAL/STM32F1xx/blob/master/Inc/cnc3040_map.h) or by using the `$pins` command

> 2. compiled via ide, included in the file define USB_SERIAL_CDC =1, but usb does not work, how to enable it?

Did you compile with the correct MCU selected and bootloader on/off version?

4. See [this Wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).

logvinov-ivan, 2023-02-08:
I don't know how to enable usb properly.

logvinov-ivan, 2023-02-08:
$14=78 set, but grblhal does not accept these values, an error is written

terjeio, 2023-02-13:
`$14` controls pin inversions and pin availablility is depending on the controller. Try `$14=15` or` $$=14` for information about which pins are available and the setting value assigned to each pin. Sum the values to get the setting value you need. Another option is to connect the pin(s) to ground.

terjeio, 2023-02-13:
> I don't know how to enable usb properly.

USB communication [is the default](https://github.com/grblHAL/STM32F1xx/blob/f529ddf79782ac87f7b86794a230608d3e42f8b9/Inc/my_machine.h#L34-L36) when available. If it is not working it could be due to flashing a version of the firmware that does not match the processor and/or the processor clock (crystal) on the board.

logvinov-ivan, 2023-02-13:
how to flash stm32f103c8t6 correctly, the assembly from isb works through the site, and if the assembly is done through ide, then it doesn’t work, what am I doing wrong? downloaded grbl hal opened, compiled received a file, format . bin via stm32 loaded the program into the board.

terjeio, 2023-02-13:
You are compiling the RC version instead of the unnamed one which is for the C8? Select the correct one here - from the dropdown presented when clicking on the down arrow.

[image]

If flashing via the debug tool ensure the correct binary is selected.

[image]

Ams-Laser, 2023-03-12:
> ## Хочу сделать швейную машинку для вышивания..
> как подключить (датчик индуктивности или датчик холла )для установки поднятия иглы ? и прописать его в прошивке
> чтоб по команде в gcode m555 двигатель [...]

https://user-images.githubusercontent.com/87819514/224558057-c2149635-8332-4305-a383-79770a65580f.mp4

i have what you want to do already working, i just want to makeit faster, at least 800 to 1200 STPM
M66 is the most correct way to go, however i have setup a joystic board wired to a optocoupler sensor to "read" whhen the neddle is in the up position, then the sender just sends one line each time the sensor is triggered

i will test with grblHal and see if i can get it to work with M66

Ed

terjeio, 2023-03-12:
@Ams-Laser A few questions:

Is the current controller running some Grbl version?
Can you zip and add an example gcode file if it gcode is used?
Is the sender a dedicated sender with special features? Like handling when the needle in the up position?
Is the threading (needle) motor under speed (RPM) control?
How are long jumps handled - they may take longer than the time for one stitch (or one revolution of the motor)?

> i will test with grblHal and see if i can get it to work with M66

You will have to add a definition for an Aux input to the board map, you can do that? Or should I do it?

Ams-Laser, 2023-03-12:
@terjeio 
> @Ams-Laser A few questions:
> 
> Is the current controller running some Grbl version?
yes its running grblmega-5x from here https://github.com/fra589/grbl-Mega-5X

 Can you zip and add an example gcode file if it gcode is used?
[sigzag.zip](https://github.com/grblHAL/core/files/10951922/sigzag.zip)
here is the input DST (tajima) file and the converted (by my sender) output GCODE file

 Is the sender a dedicated sender with special features?
YES

Like handling when the needle in the up position?
Yes, i call this servo mode, it waits for a joystick button to be pressed before sending the next G! line

 Is the threading (needle) motor under speed (RPM) control?
controlled by relay, speed is setup at the servo control box and can be varied without problem unless it gets so fast that the moves cannot keep up

 How are long jumps handled - they may take longer than the time for one stitch (or one revolution of the motor)?
the time we have for movement is half revolution (from the poin where the needle comes out the fabric to the point where its almost going into the fabric again

> 
> > i will test with grblHal and see if i can get it to work with M66
great!!!
 
> You will have to add a definition for an Aux input to the board map, you can do that? Or should I do it?
please doit, i failed at compiling grblHal for my MKS-dlc32 board

let me know if you need my sender to test and compare

Thanks again 
Ed

terjeio, 2023-03-12:
@Ams-Laser I will add an aux input tomorrow, it is getting late here. I came to think of another feature of grblHAL that could be used, it is possible to put the controller in single step mode, toggled on/off either via `$S` or real-time command `0x89`. Triggering the cycle start input will execute the next line (block) in the program. Using this will be simpler than adding M66 commands?
BTW you can compile by using the [Web Builder](http://svn.io-engineering.com:8080/).

Ams-Laser, 2023-03-12:
> @Ams-Laser I will add an aux input tomorrow, it is getting late here. I came to think of another feature of grblHAL that could be used, it is possible to put the controller in single step mode, [...]

y tried the web builder but the uploader has changed and i could not find instructions on new version
thanks
Ed

terjeio, 2023-03-13:
>  I will add an aux input tomorrow

Done, pin 33 (EXP_1 header, pin 8). Note that this pin does not have a pullup so you have to drive the pin by logic level or add a pullup resistor. I have also added feed hold and cycle start inputs on J2 (I2C header). 

You can test the aux input by sending `M66P0L2Q5`, if you trigger the pin (pull it low) you should get the `ok` response before the 5 second timeout. I have only tested with a bare dev board as I do not have a DLC32.

From looking at the sigzag gcode file I guess you are using the comments to control the needle motor and set up a wait for trigger. Correct? 

> the uploader has changed and i could not find instructions on new version

You are running on Linux? I've just tested the Windows version and that works fine using the current instruction page.

Ams-Laser, 2023-03-13:
Thanks, im still having truble compiling for my dlc32 board , once i get it working ill test the i/o

Again, thanks
Ed.

terjeio, 2023-03-13:
@Ams-Laser I've added a sensor on an old PFAFF machine to check timings on my scope - you must have pretty high acceleration and feedrate settings to achieve the speed I see in your video? May I ask what they are?

BTW I have added plugin support for Tajima DST files (streaming from SD card only) - I need to add motor control to the sewing machine to complete that, but it looks good. I can even get a 3D view in my sender from translating the DST format to gcode and upload that to the sender:

[image]

Ams-Laser, 2023-03-13:
Looks Great!!!!
Step/mm 10 on all axis
Máx rate 6000mm/min
Accel 4000mm/sec2

Motor speed control done by relay on output pin 

Can i please try on my machine

Tanks
Ed.

terjeio, 2023-03-14:
> Can i please try on my machine

Not yet - I have to figure out when to cut power the motor first. It should be cut a number of milliseconds before the trigger so that it comes to a stop just after the final encoder signal? Or is motor acceleration so slow that it does not matter for the first stitch after a start? At low stitch rates this might not matter at all, but at higher ones it might?

How do you handle this, if at all?

Anyway - I am going to the workshop to fetch a relay for the motor...

Ams-Laser, 2023-03-14:
@terjeio 

> > Can i please try on my machine
> 

> Not yet - I have to figure out when to cut power the motor first. It should be cut a number of milliseconds before the trigger so that it comes to a stop just after the final encoder signal? Or is [...]

Im usinga Servo motor for the sewing machine, it has its own encoder and automatically stops the machine with the neddle in the up position, there are cheaper than closed loop steppers or servos, i just cut off the relay and wait 2 seconds in case the servo was just passed the needle up position and slowly does one last stitch then is safe to move the axis

can you share your converted "CUADROS.dst" file so i can see how you are handling the positions, color changes, etc.

hope this helps
Ed.

terjeio, 2023-03-14:
> Im usinga Servo motor for the sewing machine

Ok, but I guess not everybody will have that. The machine I am testing with is old (50 years+) and uses a tapped resistor for speed control, so I believe I have to be able to calibrate when to cut power.

> can you share your converted "CUADROS.dst" file

I do not have such a file, you forgot to attach it? Anyway the conversion to gcode is available just for a sender to fetch (if it wants it) so that it can display the design. Execution is directly from the embroidery file (.dst or .pes) - the plugin reads stitch data from a SD card connected to the controller and calls the appropriate code in the grblHAL core to control the machine. This way I have a lot more control of what is going on than by translating to gcode first and sending that to the gcode parser.

Ams-Laser, 2023-03-14:
@terjeio as a solution for the sewin machine control, just use a relay to enable, and use the pedal to set the speed, upon stopping display a message to user asking tu put needle in upper position manualy then pres continue, 
anyway the neddle wont stop until a color change is needed so there shoudl not be a problem,
this way it will work wit ANY motor without need to adjust timings, as these can vary on tread and fabric being sewed
this is the file i use for test/debug, its small and has bot small and large stitches remove the TXT dummy extension and its a DST file, if you could catch your gcode output i could help you verify, i had one bug when i converted when the very first line after the neddle set command its in the grong position and should be sent before starting the motor.
Ed
[cuadros.dst.txt](https://github.com/grblHAL/core/files/10973437/cuadros.dst.txt)

Ams-Laser, 2023-03-14:
is there a way to enable bluethooth serial on esp32 with grblHal?
or setup a virtual port like i had with MSK original firmware using
[image]
Thanks. 
Ed.

terjeio, 2023-03-14:
> is there a way to enable bluethooth serial on esp32 with grblHal?

[Yes](https://github.com/grblHAL/ESP32/blob/6e2bb1196b4d74921ad322e2afeaac3ca925494e/main/CMakeLists.txt#L38) - but not yet from the Web Builder. You have to compile locally for now.

> or setup a virtual port like i had with MSK original firmware using

Telnet and Websocket communication is supported so yes?

> anyway the neddle wont stop until a color change is needed so there shoudl not be a problem,

There are other commands that stops the needle as well, at least trim and jump commands.

> if you could catch your gcode output i could help you verify, i had one bug when i converted when the very first line after the neddle set command its in the grong position and should be sent before starting the motor.

Give me a bit of time for that - saving is disabled in my sender for files loaded from the SD card.

BTW, I have [published the code](https://github.com/grblHAL/Plugin_embroidery) as it now stands, and opened [a discussion](https://github.com/grblHAL/Plugin_embroidery/discussions/1) where further comments should be made.

terjeio, 2023-03-14:
@Ams-Laser Here is the converted file. It is in a minimized format, in legal gcode, to increase upload speed:

[cuadros.txt](https://github.com/grblHAL/core/files/10974115/cuadros.txt)

pitergus, 2024-08-18:
@terjeio Hello, I have a question about the embroidery attachment. How is the motor controlled? Is it controlled by turning it on and off or by PWM?

terjeio, 2024-08-18:
> How is the motor controlled? Is it controlled by turning it on and off or by PWM?

You choose. A trigger signal is needed to detect needle position once for every stitch.

pitergus, 2024-09-12:
@terjeio  
Hello how will I be able to enable the PWM signal for the sewing machine motor, when I turn the pulley manually if the next xy axis movement passes
[image]

terjeio, 2024-09-13:
@pitergus I am not sure I understand what you want to achieve. You want to turn the motor manually while the program is running and at some point start it?

pitergus, 2024-09-13:
“Hello, when starting the embroidery, the motor does not activate in PWM mode. I am testing in off/on mode. How do you configure it to use PWM?

terjeio, 2024-09-15:
>  How do you configure it to use PWM?

Setting $30=1 may help, it will output a 100% PWM signal when turned on. $30=2 will output a 50% duty cycle signal since the spindle is turned on with a RPM of 1. Duty cycle is 1 / $30 value.

pitergus, 2024-09-19:
@terjeio "Hi, I'm using .pes files and it's - working great.  https://github.com/user-attachments/assets/9c6ca3d0-186a-4d95-bcb2-110b6d227a89

closed 2024-10-26 by terjeio

Ams-Laser, 2025-03-07:
https://youtube.com/shorts/ccW4PQBU1mk?si=Whr4XTzKCeQr84WK 
latest update