grblHAL/core #246: $22 error

Am0k-GIT, 2023-02-02:
Good afternoon
I am using the following configuration:
```C
#define DEFAULT_HOMING_ENABLE                             On          //$22 (bit 0)
#define HOMING_SINGLE_AXIS_COMMANDS                      Off          //$22 (bit 1)
#define DEFAULT_HOMING_INIT_LOCK                          On          //$22 (bit 2)
#define HOMING_FORCE_SET_ORIGIN                          Off          //$22 (bit 3)
#define DEFAULT_HOMING_ALLOW_MANUAL                       On          //$22 (bit 4)
#define DEFAULT_HOMING_OVERRIDE_LOCKS                     On          //$22 (bit 5)
#define DEFAULT_HOMING_KEEP_STATUS_ON_RESET              Off          //$22 (bit 6)
```
`$22=53 (0b00110101)`
However, after flashing, $RST=$, I see in EEPROM
`$22=101 (0b01100101)`

Next combination:
```C
#define DEFAULT_HOMING_ENABLE                             On          //$22 (bit 0)
#define HOMING_SINGLE_AXIS_COMMANDS                       On          //$22 (bit 1)
#define DEFAULT_HOMING_INIT_LOCK                          On          //$22 (bit 2)
#define HOMING_FORCE_SET_ORIGIN                           On          //$22 (bit 3)
#define DEFAULT_HOMING_ALLOW_MANUAL                       On          //$22 (bit 4)
#define DEFAULT_HOMING_OVERRIDE_LOCKS                     On          //$22 (bit 5)
#define DEFAULT_HOMING_KEEP_STATUS_ON_RESET               On          //$22 (bit 6)
```
`$22=127 (0b01111111)`
However, after flashing, $RST=$, I see in EEPROM
`$22=239 (0b11101111)`

terjeio, 2023-02-02:
The `$22` setting has a bit inserted from the limits flags at position 4 - so there is no 1:1 relation with the homing flags byte:

https://github.com/grblHAL/core/blob/e2ba6db69cda9d7ab664c3d4bbce0c1737b81768/settings.c#L979-L998

Other settings may also mix data from different sources, and as you can see from above also modify related settings.

Am0k-GIT, 2023-02-02:
Thank you.
`\internal Bit 4 in settings.homing.flags.`  in `config.h` phrase confused me

terjeio, 2023-02-02:
> \internal Bit 4 in settings.homing.flags. in config.h phrase confused me

Yep - that is a documentation bug - I'll have to proofread it again...

terjeio, 2023-02-02:
Two homing settings defines are missing the DEFAULT_ prefix, will fix in the next commit.
And I will move the DEFAULT_LIMITS_TWO_SWITCHES_ON_AXES setting into the $22 group.

BTW do you want me to add a link to [your board](https://github.com/Am0k-GIT/STM32F401CCU6_UNI) in the [controllers list](https://github.com/grblHAL/Controllers)? If so then if you are not using an existing board map then yours should be added to the repo as well.

Am0k-GIT, 2023-02-03:
Yes, I'm actively following your work, and I'm trying to make my board better as much as I can. Mail for honor if there is a link to it somewhere. 
Right now I'm using pinmap by simply overwriting [blackpill_map.h](https://github.com/Am0k-GIT/STM32F401CCU6_UNI/blob/master/GRBL/Inc/blackpill_map.h)

terjeio, 2023-02-03:
> Right now I'm using pinmap by simply overwriting [blackpill_map.h](https://github.com/Am0k-GIT/STM32F401CCU6_UNI/blob/master/GRBL/Inc/blackpill_map.h)

Don't do that if you want other people to use your board. Create a new map file and modify _my_machine.h_ and _driver.h_ to compile with it when enabled. Then create a pull request for the changed files. If you want the board to be available from the [Web Builder](http://svn.io-engineering.com:8080/) add it to _driver.json_ as well and include it in the pull request.

closed 2026-02-27 by terjeio