grblHAL/core #255: Error while compiling

zab-ig, 2023-02-20:
Good afternoon. A compilation error occurs when included VFD. The error does not occur in the previous version.
`
Arduino: 1.8.9 (Windows 8.1), TD: 1.57, Плата:"Teensy 4.1, Serial, 600 MHz, Faster, US English"

ВНИМАНИЕ: Категория 'Utility' в библиотеке FlexIO_t4 не является действительной. Установка на 'Uncategorized'
ВНИМАНИЕ: Категория 'Language' в библиотеке ArduinoStreaming не является действительной. Установка на 'Uncategorized'
C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\networking\networking.c:86:9: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]

 #pragma GCC diagnostic ignored "-Wstringop-overflow"

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c: In function 'fs_open':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c:92:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

             file->size = f_size((FIL *)&file->handle);

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c: In function 'fs_tell':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c:126:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     return f_tell((FIL *)&file->handle);

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c: In function 'fs_eof':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c:136:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     return f_eof((FIL *)&file->handle) != 0;

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_fatfs.c:136:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c: In function 'fs_read':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c:124:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     return lfs_file_read(&lfs, &((time_file_t *)&file->handle)->file, buffer, size * count);

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c: In function 'fs_tell':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c:138:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     return lfs_file_tell(&lfs, &((time_file_t *)&file->handle)->file);

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c: In function 'fs_seek':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c:143:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     return lfs_file_seek(&lfs, &((time_file_t *)&file->handle)->file, offset, LFS_SEEK_SET);

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c: In function 'fs_eof':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c:148:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     return lfs_file_tell(&lfs, &((time_file_t *)&file->handle)->file) == file->size;

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\spindle\vfd\spindle.c:157:13: warning: 'is_vfd_spindle' defined but not used [-Wunused-function]

 static bool is_vfd_spindle (const setting_detail_t *setting)

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\spindle\vfd\spindle.c:170:13: warning: 'is_modvfd_selected' defined but not used [-Wunused-function]

 static bool is_modvfd_selected (const setting_detail_t *setting)

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\utility\ff_utils.c: In function 'get_fattime':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\utility\ff_utils.c:107:29: warning: implicit declaration of function 'rtc_get' [-Wimplicit-function-declaration]

     struct tm tx=seconds2tm(rtc_get());

C:\TEMP\arduino_build_197975\libraries\grblHAL_Teensy4\spindle\vfd\spindle.c.o: In function `is_ysgl_selected':

C:\Program Files (x86)\Arduino\libraries\grblHAL_Teensy4\src\spindle\vfd/spindle.c:189: undefined reference to `spindle_select_get_binding'

collect2.exe: error: ld returned 1 exit status

Ошибка компиляции для платы Teensy 4.1.

`

Avatario34, 2023-02-20:
Don't know what the exact problem is. Only spotted that you were using Teensyduino 1.57. I can't compile GrblHAL with Teensyduino 1.57 either, with errors in respect to sd card handling (which got an overhaul in 1.57). So I use Teensyduino 1.56 instead as recommended in platformio.ini:
`#platform = ${common_teensy.platform} NOTE: the latest version is broken as of 2022-10-21
platform = ${common_teensy.platform}@4.16`
Maybe try to compile with 1.56 to eliminate options

terjeio, 2023-02-20:
Fix commited, thanks for reporting.
The 1.56 vs 1.57 comment is relevant, but not related to this issue.

zab-ig, 2023-02-21:
After the fix, the error no longer occurs. Thank you.