grblHAL/core #258: I don't understand, if I don't use the USB virtual serial port and enable the KeyPad plug -in, which serial port will the keypad plug -in?

hanke-cnc, 2023-02-23:
Compiled an error and can run normally, but I don't know that the default PA9 PA10 of KeyPad is occupied by GRBL. Which serial port will he use?

hanke-cnc, 2023-02-23:
I found a little trace in driver_opts.h and then no progress

#if MPG_ENABLE && !defined(MPG_STREAM)
#if USB_SERIAL_CDC
#define MPG_STREAM          0
#else
#define MPG_STREAM          1
#endif
#endif

terjeio, 2023-02-23:
`$pins` will tell you which pins are assigned. Be aware that the keypad plugin can use I2C or UART comms depending on the configuration in _my_machine.h__:

```
//#define MPG_ENABLE           1 // Enable MPG interface. Requires serial port and one handshake pin unless
                                 // KEYPAD_ENABLE is set to 2 when mode switching is done by the CMD_MPG_MODE_TOGGLE (0x8B)
                                 // command character. Set both MPG_ENABLE and KEYPAD_ENABLE to 2 to use a handshake pin anyway.
//#define KEYPAD_ENABLE        1 // Set to 1 for I2C keypad, 2 for other input such as serial data. If KEYPAD_ENABLE is set to 2
                                 // and MPG_ENABLE is uncommented then a serial stream is shared with the MPG.
```

hanke-cnc, 2023-02-23:
> `$pins` will tell you which pins are assigned. Be aware that the keypad plugin can use I2C or UART comms depending on the configuration in _my_machine.h__:
>
> ```
> //#define MPG_ENABLE           1 // [...]

> `$pins` will tell you which pins are assigned. Be aware that the keypad plugin can use I2C or UART comms depending on the configuration in _my_machine.h__:
>
> ```
> //#define MPG_ENABLE           1 // [...]

Thank you, I found it in serial.c,
Also using $pins with your help, same result
But this also caused a problem. I never knew the powerful command $pins. It seems that it is not written in the wiki. (Maybe I am the only one who does not know?????????), your source The code seems to have many system commands

closed 2023-02-23 by hanke-cnc