grblHAL/core #259: A strange demand from a user? He needs to save the mechanical coordinates after power off?

hanke-cnc, 2023-02-26:
This seems to be the usage habit of mach3?

terjeio, 2023-02-26:
Mach3 is a windows app that is closed manually or on Windows shutdown allowing it to save the position.
grblHAL is microcontroller firmware that does not have shutdown command and IMO never will get one.

However, a plugin can be written that writes the current position to EEPROM or FRAM on a transition to the _Idle_ state and restores it on startup. Not that if settings are stored in flash I would not contemplating doing so - it would be a sure way to wear out the flash...

closed 2023-02-26 by hanke-cnc

hanke-cnc, 2023-02-26:
Oh I know!!! Although grbl doesn't need a computer to run, usually people have to connect to a computer anyway, and when the user wants to shut down, save the mechanical coordinates in the computer, but I don't know how to change the mechanical of grbl coordinates, I don't know where its variable

terjeio, 2023-02-26:
... but I don't know how to change the mechanical of grblhal coordinates, I don't know where its variable

If you want to save and restore the position from a sender you would still need a plugin as there is no command to set the position. The position is stored in the [sys.postion](https://github.com/grblHAL/core/blob/0c6233d7096e352770be76c7524f14261931ce96/system.h#L264) array. If set after startup is complete other positions has to be synced with this.
If position is restored from EEPROM or FRAM on startup then the plugin can "hook into" [a function pointer](https://github.com/grblHAL/core/blob/0c6233d7096e352770be76c7524f14261931ce96/grbllib.c#L226-L227) to set it.

karoria, 2023-02-26:
@hanke-cnc @terjeio if you want to restore saved work coordinates from PC and apply them to machine without moving any axis, you can do it by sending G10L20P0X<workcoordinate> to the controller. That is standard Code. E.g. G10L20P0X107.848. Replace X with Y, Z or A for other axis. I use this command in my grblTouch webui to set coordinate to a specific digit.