grblHAL/core #260: I downloaded my_plugin.c, and added my_plugin_init() to board_init in st_morpho.c, grblhal seems to be dead, there will be no response

hanke-cnc, 2023-03-03:
The USB virtual serial port can be recognized, but GRBLHAL seems to be dead. If you don’t add my_plugin_init(); in board_init, everything works, but I can’t find the custom settings.

hanke-cnc, 2023-03-03:
The plugin I downloaded here, without any changes, https://github.com/grblHAL/Templates/blob/master/my_plugin/settings/my_plugin.c
, and I don't understand static plugin_settings_t my_settings;
What is the parameter value of my_settings? I mean how much is he $?

terjeio, 2023-03-03:
> and added my_plugin_init() to board_init

Don't do that, just add _my_plugin.c_ to the _Src_ folder and recompile. It will be automatically added.

> What is the parameter value of my_settings?

In the IDE just hover the mouse over `Setting_UserDefined_0` (and 1) and you will see the settings number. Or look them up in [grbl/settings.h](https://github.com/grblHAL/core/blob/6163f2e30a207174862382ccf51cc025e666e596/settings.h#L279-L280).

If using ioSender the new settings will also show up in the _Settings: Grbl_ tab:

[image]

I have not written documentation for how to add different setting types. FYI there are plenty of examples [here](https://github.com/grblHAL/core/blob/6163f2e30a207174862382ccf51cc025e666e596/settings.c#L406-L563) that should get you started.

hanke-cnc, 2023-03-04:
> Don't do that, just add _my_plugin.c_ to the _Src_ folder and recompile. It will be automatically added.
>
>
> In the IDE just hover the mouse over `Setting_UserDefined_0` (and 1) and you will see the [...]

Great, I see that so I can add a set value for my handwheel speed

closed 2023-03-04 by hanke-cnc