grblHAL/core #263: Can't compile 6-axis for Neculeo-64 STM32F4RE Board

MuhammadQaisarAli, 2023-03-08:
I followed the instructions given in "compiling girbalHAL section" to compile the driver for 6-axis. but after downlading biard, the driver is still 3 axis. 
I edited the cnc3040 file for needs, and saved the file with the name "my_machine_map.h" in Inc folder. then i uncommented the "#define  BOARD_MY_MACHINE " line. But didn't got success.

The webbuilder tool dont have 6 axis support for Nuclo64 boards, altho Bluepill F401 can have 5 axis.

terjeio, 2023-03-08:
Did you change [N_AXIS](https://github.com/grblHAL/core/blob/c7a6f049e489119e58f60c7585f977d71df39f33/config.h#L42) to 6?

MuhammadQaisarAli, 2023-03-08:
Yes.. changed.

MuhammadQaisarAli, 2023-03-08:
Here is the "MACROS"
[image]

 
Here is **my_machine.h** file
`/*
  my_machine.h - configuration for STM32F4xx ARM processors

  Part of grblHAL

  Copyright (c) 2020-2023 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used.
//#define BOARD_CNC3040
//#define BOARD_PROTONEER_3XX   // For use with Nucleo-F411RE and F446RE boards.
//#define BOARD_GENERIC_UNO     // For use with Nucleo-F411RE and F446RE boards.
//#define BOARD_CNC_BOOSTERPACK
//#define BOARD_MORPHO_CNC      // For use with Nucleo-F411RE and F446RE boards. Work in progress.
//#define BOARD_MORPHO_DAC_CNC  // For use with F446RE boards and a 32-bit I2S stereo DAC for X and Y output. **EXPERIMENTAL**
//#define BOARD_BLACKPILL       // For use with F411 BlackPill, with optional auto square support for one axis
//#define BOARD_MINI_BLACKPILL  // For use with F411 BlackPill, https://github.com/avizienis/Minimal-Black-Pill--STM32F4xx-BOB-for-grblHAL
//#define BOARD_BTT_SKR_PRO_1_1 // F407 based 3D Printer board
//#define BOARD_BTT_SKR_PRO_1_2 // F407 based 3D Printer board
//#define BOARD_BTT_SKR_20      // F407 based 3D Printer board
//#define BOARD_BTT_SKR_20_DAC  // F407 based 3D Printer board, uses analog output (DAC) for spindle speed control
//#define BOARD_FYSETC_S6       // F446 based 3D Printer board
//#define BOARD_FLEXI_HAL       // F446 CNC board
//#define BOARD_STM32F401_UNI   // F401 CNC board
#define BOARD_MY_MACHINE      // Add my_machine_map.h before enabling this!

// WARNING: BOARD_BTT_SKR_20 may fry your Trinamic drivers due to bad hardware design.
//          The risk goes away if Q1 (HY1904C2) is shorted between source (S) and drain (D).
//          See this issue for details: https://github.com/bigtreetech/SKR-2/issues/63
//          Note: grblHAL does not check for reversed drivers.

#if defined(NUCLEO_F401) || defined(NUCLEO_F411) || defined(NUCLEO_F446)
#define IS_NUCLEO_DEVKIT 1
#else
#define IS_NUCLEO_DEVKIT 0
#endif

// Configuration
// Uncomment to enable.

#if !IS_NUCLEO_DEVKIT && !defined(USB_SERIAL_CDC) // The Nucleo boards has an off-chip UART to USB interface.
#define USB_SERIAL_CDC         1 // Serial communication via native USB.
#endif
//#define SAFETY_DOOR_ENABLE   1 // Enable safety door input.
//#define BLUETOOTH_ENABLE     1 // Set to 1 for HC-05 module. Requires and claims one auxillary input pin.
//#define VFD_ENABLE           1 // Set to 1 or 2 for Huanyang VFD spindle. More here https://github.com/grblHAL/Plugins_spindle
//#define MODBUS_ENABLE        1 // Set to 1 for auto direction, 2 for direction signal on auxillary output pin.
//#define SDCARD_ENABLE        2 // Run gcode programs from SD card.
//#define MPG_ENABLE           1 // Enable MPG interface. Requires serial port and one handshake pin unless
                                 // KEYPAD_ENABLE is set to 2 when mode switching is done by the CMD_MPG_MODE_TOGGLE (0x8B)
                                 // command character. Set both MPG_ENABLE and KEYPAD_ENABLE to 2 to use a handshake pin anyway.
[... 134 lines cut ...]
#define COOLANT_FLOOD_PORT      GPIOB
#define COOLANT_FLOOD_PIN       3
#define COOLANT_MIST_PORT       GPIOB
#define COOLANT_MIST_PIN        4

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define CONTROL_PORT            GPIOB
#define RESET_PIN               5
#define FEED_HOLD_PIN           14
#define CYCLE_START_PIN         15
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         8
#endif
#define CONTROL_INMODE          GPIO_MAP

// Define probe switch input pin.
#define PROBE_PORT              GPIOB
#define PROBE_PIN               13
`

Here is main screen snap, 
[image]

While uploading, the following is selected as configuration file.
[image]

terjeio, 2023-03-08:
You have to select the debug build for _6. Debug F411 Nucleo 64_ from the build tool.

FYI _1. Debug_ has N_AXIS set to 5, this overrides the _config.h_ setting and the launch you chose do not use the generated binary.
[image]

This test in the map file will cause a compilation failure when N_AXIS is 6, remove it:

```
#if N_ABC_MOTORS > 2
#error "Axis configuration is not supported!"
#endif
```

There are no pin mappings for motor 5 (the C axis), you have to change the motor 4 definition from:
```
// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PORT STEP_PORT
#define M4_STEP_PIN 9
#define M4_DIRECTION_PORT DIRECTION_PORT
#define M4_DIRECTION_PIN 10
#endif
```
to 
```
// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS Z > 1
#define M4_AVAILABLE
#define M4_STEP_PORT STEP_PORT
#define M4_STEP_PIN 9
#define M4_DIRECTION_PORT DIRECTION_PORT
#define M4_DIRECTION_PIN 10
#endif
```
and add motor 5 definitions:
```
// Define ganged axis or C axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 3
#define M5_AVAILABLE
#define M5_STEP_PORT STEP_PORT
#define M5_STEP_PIN 9
#define M5_DIRECTION_PORT DIRECTION_PORT
#define M5_DIRECTION_PIN 10
#endif
```
Change the pin numbers and possibly the ports used to match the pins you want to use!
If pins are on different ports than STEP_PORT and/or DIRECTION_PORT more changes has to be made,  See the _btt_skr_pro_v1_1_map.h_ for how to, or perhaps modify that instead.

MuhammadQaisarAli, 2023-03-08:
As there was no definition for 3rd motor, so i was going with 5-axis in this example. Thenk you very much for adding the code for 3rd motor as well.

MuhammadQaisarAli, 2023-03-08:
Everything goes well. In iosender I can see 6-axis now, but there is E-stop error,
[image]

for the time being I uncommented the following line to avoid the E-stop error. (By default the E-stop should have to be disabled).

`#define ESTOP_ENABLE         0 // When enabled only real-time report requests will be executed when the reset pin is asserted.
`
But now reset is always asserted.
[image]

To fix it I disabled the pullup resistor on reset pin,
[image]

Or inverted the control pin.
[image]

One can see that its done via iosender app. How to do it via code?.

terjeio, 2023-03-08:
> One can see that its done via iosender app. How to do it via code?.

`$14=1` or you may change the setting default in [grbl/config.h](https://github.com/grblHAL/core/blob/c7a6f049e489119e58f60c7585f977d71df39f33/config.h#L800-L808).
or connect a normally closed (NC) switch to the input ([recommended](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings)).

MuhammadQaisarAli, 2023-03-16:
I modified added more code and was able to compile upto 8 axis (X,Y,Z,A,B,C,U,V,W). I tested the signals with oscilloscope as well, all 8 axis worked.
Now I started adding 9th axis (W),  there was a lot of modification and upgradation to code in almost all files.

When I added 9th axis , the Limit_Bit_mask macro raised a warning ⚠️ as  (1<<W_Axis) is (1<<9), which makes the mask a 9 bit value and can't be assigned to uint8_t variable ( which was defined to store mask value). I change that variable from uint8_t to uint16_t.. the warning fixed, but now the compiled code takes just 2% of flash, which was 25% for 8- Axis. Obviously there is something wrong. Any idea 💡 what I am missing?.

MuhammadQaisarAli, 2023-03-16:
Hmm..
Well, I changed the `nvs.h` file.
For 3 axis the `nvs_end` was `1023`, while for more than 6 axis, it was `1151`, which is `128` more than `1023` . As logical guess I added 256 to it, means I made `nvs_end` to be 1279. Although it was not calculated perfectly, but fortunately the code compiled and I was able to run 9 axis. The controller was checked with oscilloscope, ioSender and UGS as well.

I noticed two problems.
1: when the feed goes very high, the controller freezes and it's firmware settings are restored to default. But it happened only 2-3 times .

2: I created a random G-Code file for 9 axis,  Used `G1` command with 9 axis coordinates and a specific feed rate. I noticed that the feed rate on UGS is much higher than commanded feed rate, sometimes it goes to such a high feed rate, that controller can't generate pulses for, and as a result Controller freezes .

I updated the following  firmware settings.
2uS for step time
250 steps/mm for all axis,
500000 mm/min for all axis
500 mm/sec^2 for all axis.

Rest of settings are default.

I generated the following random G code having 500 lines.
On each line, there is `G1`  followed by each axis can have value between `-250` and `250`, followed by feed rate having random value between `50000` and `60000`.

terjeio, 2023-03-18:
Uncomment [this line](https://github.com/grblHAL/core/blob/cad1dc8476b9bdc8e751d2535dae7019f701a835/config.h#L163) to get a debug command, `$Q`, for the [NVS layout](https://github.com/grblHAL/core/blob/cad1dc8476b9bdc8e751d2535dae7019f701a835/nvs_buffer.c#L374-L423).

There should not be any ned to modify the NVS code itself, only the [layout spec](https://github.com/grblHAL/core/blob/cad1dc8476b9bdc8e751d2535dae7019f701a835/nvs.h#L33-L61).

closed 2023-08-26 by terjeio