grblHAL/core #265: Ganged axis does not work if defined in my_machine.h

muhammadelmogy, 2023-03-10:
I want to enable Ganged Y axis but when I define this in my_machine.h it doe not work 
```
#define Y_GANGED            1
#define Y_AUTO_SQUARE       1
```
In fact when I check the associated bins with the oscilloscope it seems like it is floating!
But when I copy those definitions to the map file it would work! and of course when I build the project it gives me warnings about definition duplications !

I have my whole project in the attachments
[Mlabs-Controller.zip](https://github.com/grblHAL/core/files/10944130/Mlabs-Controller.zip)

dresco, 2023-03-10:
At a guess, are you building with PlatformIO?

https://github.com/grblHAL/STM32F4xx/blob/2074a0efa842ced8a239888e47c9ddd4dd648f8c/platformio.ini#L19-L22

muhammadelmogy, 2023-03-10:
Yes you are right , but in that case what is the point of using my_machine.h
Also, when I disable the override I get a lot of redefine warnings of the values in my_machine.h!

dresco, 2023-03-10:
> Yes you are right , but in that case what is the point of using my_machine.h

If building with PlatformIO, the idea would be to define the settings in the .ini file instead of editing my_machine.h

Makes it much easier to support multiple configurations for different boards without any code changes. Typically you would create a new `[env:]` section for your board with the required definitions in it.

terjeio, 2023-03-10:
> Yes you are right , but in that case what is the point of using my_machine.h

Perhaps an additional platformio.ini file that do not override _my_machine.h_ should be added? Or the current one should be modified to allow that?

dresco, 2023-03-10:
> Perhaps an additional platformio.ini file that do not override _my_machine.h_ should be added? Or the current one should be modified to allow that?

From memory, it's not consistent across the drivers (i.e. F1 & F4 override my_machine.h, but iMXRT does not). 

Personally, I prefer the override if building with PlatformIO, keeps all the configuration in one place..

muhammadelmogy, 2023-03-10:
If I override my_machine.h, how do I set ganged Y axis in platform.ini file? is there a parameters list to be used in the platform.ini file?

dresco, 2023-03-10:
> If I override my_machine.h, how do I set ganged Y axis in platform.ini file? is there a parameters list to be used in the platform.ini file?

You would just add the extra definitions to your existing `[env:mlabs_controller]` section, they're the exact same symbols as in my_machine.h, but get passed into the build via the command line when building for that board..
```
  -D Y_GANGED=1
  -D Y_AUTO_SQUARE=1
```

muhammadelmogy, 2023-03-10:
Great, I prefer this way instead of modifying the source code.
I will try it later and give a feedback after that if something goes wrong!

closed 2023-03-10 by muhammadelmogy