grblHAL/core #274: Wiring limits and setup settings.

Demitrius, 2023-03-26:
Hello!
I want to try limits on my SKR-3 board and wiring similar attached scheme (NC). The wire schema taken from [Wiring Limit Switches](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches)

My goal is use limits for homing and emergency stop when one of limit trigger. What do I need to setup in GrblHAL settings?

$5=0 (or 1?)
$14=7 (or ?)
$20=1
$21=1
$22=1

Kindly guide me on defining configuration and wiring scheme if Im wrong.

[image]

terjeio, 2023-03-28:
`$5=0`
`$14` is for estop/reset and some other input signals. Set to 0 if using NC switches.
`$20` is for enabling soft limits, not related to switches.
`$21`=1
`$22=1` or odd values >=1, `$$=21` for info.
`$44=4` - home Z first
`$45=3` - then X and Y

More info [here.](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings)

Demitrius, 2023-03-28:
Thanks!

Im a bit confused, SKR 3 has 3.3V logic signal level.

If I wiring like this scheme, It will equals: 5V -> Resistor -> 3.3V signal pin.
Question, what kind of resistor I need? 1K, 4.7K or 10K ohms ?

dresco, 2023-03-28:
If you have a look at the top left of the [schematic](https://github.com/bigtreetech/SKR-3/blob/master/Hardware%20(SKR%203%20EZ)/BIGTREETECH%20SKR%203%20EZ%20V1.0-SCH.pdf) for the SKR3, you can see those pins are not directly connected to the MCU.

Am assuming their intent for passive switches is to place them between GND and the signal pin. The MCU will then either see 3.3v if open, or 0v if closed..

Note: looking at the H743 datasheet, the Y endstop (PC2) input is **not** 5v tolerant for the LQFP part.

terjeio, 2023-03-31:
> Are you sure $5=0 for Normal Close scheme?

[Yes](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings#limit-switches) - for best practice reasons (better EMI immunity and safer if a wire breaks) grblHAL defaults to normally closed switches for all inputs.

Demitrius, 2023-03-31:
_$14 is for estop/reset and some other input signals. Set to 0 if using NC switches._

I have no one connected sensors/buttons, only one Limit switch on X. Now $14=7 and board start success without warnings, any way, need I set $14=0 or not mandatory and leaves $14=7?

Demitrius, 2023-03-31:
Thanks all guys! Wired, connected, works!

Chears

closed 2023-03-31 by Demitrius

didecibel, 2025-02-27:
Hello, I am trying to wire min and max switches on a rpi pico. I cannot figure out if X and X2 are the min and max pins and what would be the pin to connect on the pico. I have read the driver.c file which seems to be the one to tweak but I am not very confortable with programming so.
I also have found this mod : 
[https://github.com/grblHAL/ESP32/commit/fa7b9196ec64ad6bec013a3ff3d5d29f24ba94e4](https://github.com/grblHAL/ESP32/commit/fa7b9196ec64ad6bec013a3ff3d5d29f24ba94e4
)

but the driver.c file is not the same. thanks a lot.

terjeio, 2025-02-27:
@didecibel The Pi Pico driver does not support max limit switches - I can add that. But you have to define your own board map since none of the current ones assign pins for them.

didecibel, 2025-02-28:
@terjeio Sorry to bother you. I would be glad if you could add the max limit switch, at least on the Y axe. I don't need the spindle commands and I can do the custom map file. thank you anyways for your answer.