grblHAL/core #276: auto Square issue

pachucristo, 2023-03-28:
Hi everyone, 
I'm working with a blackbox X32
I usualy work with cartesian machines (XYYZ) and I needed the autosquare function.
with default firmware(installed with OB CONTROL), when I do a home cycle, Y limitSw stops both motors without moving one at a time once detected at least one of 2 NC switches wired in series to the Y limit.
I assumed that I needed to change the firmware, and I found the web "wizzard" to configure a bin file to flash firmware, I configured as usualy. flashed it, and started configuring again the motors and switches.

first I noted that the Probe Pin now act as a Y2 limit so I re wired every limit sw to a dedicated port in the blackbox. BUT...
when I ask for a $HY homing cycle, the switches act as they were independent only if the Y2 Switch is Triggered  first and Y1 second. and if I ask for a $HY again, and press Y1 Limit first and then press Y2, both motors keep running until I touch the Y2 limit. 
if the motors where attached to the machine, the autosquare would only work if Y2 motor is ahead of Y1, if they where the other way arround the machine would crash Y1motor until Y2 reach Y2 limit.
the independent movement of Y1-2 motors is only effective if Y2 limit is triggered first.

i would appreciate any advise on this, thanks!

[firmware blackboxX32.zip](https://github.com/grblHAL/core/files/11093589/firmware.blackboxX32.zip)

terjeio, 2023-03-29:
Homing works ok with my CNC machine simulator.
Note that performing a successful homing cycle by manually triggering switches is hard for a single motor axis, for a dual motor axis close to impossible?

pachucristo, 2023-03-29:
here is my setup, from left to right: X Y1 Y2 Z
although I can set a % when homing to trip an alarm, in some cases the Y1 motor is going to crash with the end of the machine. 
[image]

https://user-images.githubusercontent.com/19658139/228560285-f3705ee1-5f05-45c5-a2c0-32acc88ef333.mp4

pachucristo, 2023-03-29:
> Homing works ok with my CNC machine simulator. Note that performing a successful homing cycle by manually triggering switches is hard for a single motor axis, for a dual motor axis close to impossible?

GRBL 1.1f  you can home even 3 motors at the same time, every one of them stoping right where the switches are in the machine space. ( thats because usualy the Y2 switch is wired parallel to the Z switch, so home Z first, then X, Y1and Y2 at the same time, because in that cycle the 3 limit pins are available to react and stop where it should. 
in this case, the home cycle only reacts stoping the motor when Y2 trigger first. if Y1 trigger first it keeps turning towards the Y1 switch and only stops when Y2 is triggered. 
its like the homing routine only keep going if one(exclusive) of the 2 ganged switch triggers first, and not either of the 2 ganged switch like should it be. like in 1.1f.

I hope i was clear with this observation, Thanks!

terjeio, 2023-03-29:
> thats because usualy the Y2 switch is wired parallel to the Z switch

This is not supported by grblHAL - each limit switch has its own input. `$pins` lists which:

> if Y1 trigger first it keeps turning towards the Y1 switch and only stops when Y2 is triggered.

Oops - this is a regression from last year - a line was commented out that should not be.

Tip: you may use the `$lim` command to check which switches are triggered - see [this discussion](https://github.com/grblHAL/core/discussions/261) for details.