grblHAL/core #277: Ethernet connection issue

AronAtGit, 2023-03-30:
Hey there, long story short but I have a 3 axis machine using a teensy 4.1 and a grblHAL breakout board with ethernet.  The connection between the PC and the breakout board has always been over ethernet, not USB.  The machine was purchased second hand and recently the controls were a bit buggy so I decided it was time to update both the version of IOsender on my windows computer.  After downloading the newest version (2.0.42) I couldn't keep the program open for very long before it crashed.  After reading some similar issues I decided to update the firmware on the teensy since I have no historic record of what it was previously or when it was flashed.

After updating both the firmware on the teensy (using the new web builder) and IOsender I am unable to connect via ethernet anymore.  I can connect fine over serial USB but when an ethernet cable is connected but the IP address in the console of IOsender is blank.  It appears as though the PC is sending packets but not receiving any in return from the Teensy.  For now I've left the grbl setting ($70 I think) to DHCP.

Any insight would be hugely appreciated, thanks!

terjeio, 2023-03-31:
Reset plugin settings with `$RST=&` and power cycle the controller. If it does not work after that then post your settings, Use _Help > Abount > To clipboard_ and then paste in a comment.

AronAtGit, 2023-03-31:
Thanks for the quick reply!  Resetting the plugin settings with $RST=& and power cycling the controller did not help.  Here's the settings:

%
; grblHAL
; 1.1f.20230320
; [OPT:VNMSL2,35,1024,3,0]
; [NEWOPT:ENUMS,RT+,ES,REBOOT,TC,SED,RTC,ETH,SD]
; [FIRMWARE:grblHAL]
; [NVS STORAGE:*FLASH]
; [DRIVER:iMXRT1062]
; [DRIVER VERSION:230130]
; [DRIVER OPTIONS:USB.1]
; [BOARD:T41U5XBB]
; [AUX IO:4,3,0,0]
; [IP:]
; [PLUGIN:SDCARD v1.08]
;
$N0=
$N1=
; 0 - Step pulse time
$0=10.0
; 1 - Step idle delay
$1=25
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=0
; 4 - Invert stepper enable pin(s)
$4=7
; 5 - Invert limit pins
$5=0
; 6 - Invert probe pin
$6=0
; 8 - Ganged axes direction invert
$8=0
; 10 - Status report options
$10=511
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 14 - Invert control pins
$14=0
; 15 - Invert coolant pins
$15=0
; 16 - Invert spindle signals
$16=0
; 17 - Pullup disable control pins
$17=0
; 18 - Pullup disable limit pins
$18=0
; 19 - Pullup disable probe pin
$19=0
; 20 - Soft limits enable
$20=0
; 21 - Hard limits enable
$21=0
; 22 - Homing cycle
$22=0
; 23 - Homing direction invert
$23=0
; 24 - Homing locate feed rate
$24=25.0
; 25 - Homing search seek rate
$25=500.0
; 26 - Homing switch debounce delay
$26=250
; 27 - Homing switch pull-off distance
$27=1.000
; 28 - G73 Retract distance
$28=0.100
; 29 - Pulse delay
$29=0.0
; 30 - Maximum spindle speed
$30=1000.000
; 31 - Minimum spindle speed
$31=0.000
; 32 - Mode of operation
$32=0
; 33 - Spindle PWM frequency
$33=5000
; 34 - Spindle PWM off value
$34=0.0
; 35 - Spindle PWM min value
$35=0.0
; 36 - Spindle PWM max value
$36=100.0
; 37 - Steppers deenergize
$37=0
; 39 - Enable legacy RT commands
$39=1
; 40 - Limit jog commands
$40=0
; 43 - Homing passes
$43=1
; 44 - Axes homing, first pass
$44=4
; 45 - Axes homing, second pass
$45=3
; 46 - Axes homing, third pass
$46=0
; 62 - Sleep enable
$62=0
; 63 - Feed hold actions
$63=2
; 64 - Force init alarm
$64=0
; 65 - Probing feed override
$65=0
; 70 - Network Services
$70=3
; 100 - X-axis travel resolution
$100=250.000
; 101 - Y-axis travel resolution
$101=250.000
; 102 - Z-axis travel resolution
$102=250.000
; 110 - X-axis maximum rate
$110=500.000
; 111 - Y-axis maximum rate
$111=500.000
; 112 - Z-axis maximum rate
$112=500.000
; 120 - X-axis acceleration
$120=10.000
; 121 - Y-axis acceleration
$121=10.000
; 122 - Z-axis acceleration
$122=10.000
; 130 - X-axis maximum travel
$130=200.000
; 131 - Y-axis maximum travel
$131=200.000
; 132 - Z-axis maximum travel
$132=200.000
; 300 - Hostname
$300=grblHAL
; 301 - IP Mode
$301=1
; 302 - IP Address
$302=192.168.5.1
; 303 - Gateway
$303=192.168.5.1
; 304 - Netmask
$304=255.255.255.0
; 305 - Telnet port
$305=23
; 307 - Websocket port
$307=80
; 341 - Tool change mode
$341=0
; 342 - Tool change probing distance
$342=30.0
; 343 - Tool change locate feed rate
$343=25.0
; 344 - Tool change search seek rate
$344=200.0
; 345 - Tool change probe pull-off rate
$345=200.0
; 346 - Restore position after M6
$346=1
; 370 - Invert I/O Port inputs
$370=0
; 372 - Invert I/O Port outputs
$372=0
; 384 - Disable G92 persistence
$384=0
; 398 - Planner buffer blocks
$398=35
; 481 - Autoreport interval
$481=0
%

terjeio, 2023-03-31:
Settings looks ok, $301=1 which enables DCHP.
Is the green LED in the ethernet socket blinking when the cable is connected?

AronAtGit, 2023-03-31:
Yes, the green light blinks when I power cycle the controller or unplug and then replug the ethernet cable but after about 5-10 seconds of blinking it stays solid green.  When I look at the ethernet device on the computer I see that the computer is sending packets but is never receiving any in return.  I'm not sure if that's a relevant bit of information or not.

AronAtGit, 2023-03-31:
[image]

terjeio, 2023-03-31:
Is the controller connected directly to an ethernet port on the PC?

terjeio, 2023-03-31:
ok - then you need a fixed IP address - not getting it via DHCP. Which IP address did you use earlier for connecting?
Set:
$301=0
$302=previous IP address
$303=PC IP address

PC IP address should be defined in the TCP/IPv4 properties, _Use the following IP address_ checked and fields filled in. I am not sure how to obtain DNS addresses - leave as-is?

[image]

AronAtGit, 2023-03-31:
I tried it this way, no dice unfortunately.  I set $301 to 0, $303 to my PC's IP address  and $302 to the old address and then a random new address that followed the pattern of 192.168.1.xxx I tried, The old IP address was 10.10.10.xxx but the addressing on my computer's network follows the 192.168.1.xxx protocol.  I did a power cycle of the controller after every setting change.  Strange but maybe i'm not allocating the static IP address correctly?

terjeio, 2023-03-31:
The controller cannot have the same address ($302) as the PC and it needs to be in the range (191.168.1.xxx) - as defined by the netmask ($304 - 255.255.255.0). 

>  Strange but maybe i'm not allocating the static IP address correctly?

It should be reported by the $I command (or in _Help > About_ in the sender). It isn't?

terjeio, 2023-03-31:
Can you ping the controller IP?
When testing here I can connect via websocket and the WebUI seems to work, but I cannot connect via telnet even with the firewall off.

AronAtGit, 2023-03-31:
I set the controller's static IP to a different address than the PC.  I did this first in the windows Network and Internet tab of the controller's ethernet settings and then in IOsender's GRBL settings.

After I did that I was able to see the static IP using the $I command but unable to use that Ip address for connecting the next time I open the program.  

$I output below:

%
; grblHAL
; 1.1f.20230320
; [OPT:VNMSL2,35,1024,3,0]
; [NEWOPT:ENUMS,RT+,ES,REBOOT,TC,SED,RTC,ETH,SD]
; [FIRMWARE:grblHAL]
; [NVS STORAGE:*FLASH]
; [DRIVER:iMXRT1062]
; [DRIVER VERSION:230130]
; [DRIVER OPTIONS:USB.1]
; [BOARD:T41U5XBB]
; [AUX IO:4,3,0,0]
; [IP:192.168.1.200]
; [PLUGIN:SDCARD v1.08]
;
$N0=
$N1=
; 0 - Step pulse time
$0=10.0
; 1 - Step idle delay
$1=25
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=0
; 4 - Invert stepper enable pin(s)
$4=7
; 5 - Invert limit pins
$5=0
; 6 - Invert probe pin
$6=0
; 8 - Ganged axes direction invert
$8=0
; 10 - Status report options
$10=511
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 14 - Invert control pins
$14=0
; 15 - Invert coolant pins
$15=0
; 16 - Invert spindle signals
$16=0
; 17 - Pullup disable control pins
$17=0
; 18 - Pullup disable limit pins
$18=0
; 19 - Pullup disable probe pin
$19=0
; 20 - Soft limits enable
$20=0
; 21 - Hard limits enable
$21=0
; 22 - Homing cycle
$22=0
; 23 - Homing direction invert
$23=0
; 24 - Homing locate feed rate
$24=25.0
; 25 - Homing search seek rate
$25=500.0
; 26 - Homing switch debounce delay
$26=250
; 27 - Homing switch pull-off distance
$27=1.000
; 28 - G73 Retract distance
$28=0.100
; 29 - Pulse delay
$29=0.0
; 30 - Maximum spindle speed
$30=1000.000
; 31 - Minimum spindle speed
$31=0.000
; 32 - Mode of operation
$32=0
; 33 - Spindle PWM frequency
$33=5000
; 34 - Spindle PWM off value
$34=0.0
; 35 - Spindle PWM min value
$35=0.0
; 36 - Spindle PWM max value
$36=100.0
; 37 - Steppers deenergize
$37=0
; 39 - Enable legacy RT commands
$39=1
; 40 - Limit jog commands
$40=0
; 43 - Homing passes
$43=1
; 44 - Axes homing, first pass
$44=4
; 45 - Axes homing, second pass
$45=3
; 46 - Axes homing, third pass
$46=0
; 62 - Sleep enable
$62=0
; 63 - Feed hold actions
$63=2
; 64 - Force init alarm
$64=0
; 65 - Probing feed override
$65=0
; 70 - Network Services
$70=3
; 100 - X-axis travel resolution
$100=250.000
; 101 - Y-axis travel resolution
$101=250.000
; 102 - Z-axis travel resolution
$102=250.000
; 110 - X-axis maximum rate
$110=500.000
; 111 - Y-axis maximum rate
$111=500.000
; 112 - Z-axis maximum rate
$112=500.000
; 120 - X-axis acceleration
$120=10.000
; 121 - Y-axis acceleration
$121=10.000
; 122 - Z-axis acceleration
$122=10.000
; 130 - X-axis maximum travel
$130=200.000
; 131 - Y-axis maximum travel
$131=200.000
; 132 - Z-axis maximum travel
$132=200.000
; 300 - Hostname
$300=grblHAL
; 301 - IP Mode
$301=0
; 302 - IP Address
$302=192.168.1.200
; 303 - Gateway
$303=192.168.1.156
; 304 - Netmask
$304=255.255.255.0
; 305 - Telnet port
$305=23
; 307 - Websocket port
$307=80
; 341 - Tool change mode
$341=0
; 342 - Tool change probing distance
$342=30.0
; 343 - Tool change locate feed rate
$343=25.0
; 344 - Tool change search seek rate
$344=200.0
; 345 - Tool change probe pull-off rate
$345=200.0
; 346 - Restore position after M6
$346=1
; 370 - Invert I/O Port inputs
$370=0
; 372 - Invert I/O Port outputs
$372=0
; 384 - Disable G92 persistence
$384=0
; 398 - Planner buffer blocks
$398=35
; 481 - Autoreport interval
$481=0
%

AronAtGit, 2023-03-31:
I tried to ping the controller:  

Pinging 192.168.1.200 with 32 bytes of data:
Reply from 192.168.1.156: Destination host unreachable.
Reply from 192.168.1.156: Destination host unreachable.
Reply from 192.168.1.156: Destination host unreachable.
Reply from 192.168.1.156: Destination host unreachable.

Ping statistics for 192.168.1.200:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),

terjeio, 2023-03-31:
Set $302, $303 and the PC gateway to the same IP - this is what works for me. And telnet was disabled - it works after enabling it.

AronAtGit, 2023-03-31:
ok, let me make sure I understand.  In this case I'll set $302, $303 to 168.192.1.200 as well as the default gateway on the windows setting? 
[image]

terjeio, 2023-03-31:
Yes

terjeio, 2023-03-31:
But the PC IP has to be different from the controller IP!

AronAtGit, 2023-03-31:
got it; yes the PC IP is a different IP than the controller.

unfortunately I'm not having much luck with this configuration either.  When I try to set the PC to the same IP as the controller it seems to not accept that and puts the controller back in DHCP mode.

[image]

terjeio, 2023-03-31:
>  When I try to set the PC to the same IP as the controller it seems to not accept that and puts the controller back in DHCP mode.

I has to be different!

terjeio, 2023-03-31:
Here are my settings:

Controller:
[image]

PC:

[image]

Are you on a machine with two ethernet ports active or wifi and ethernet? AFAIK the address ranges has to be different if so, both cannot use 192.168.1.xxx. FYI I am testing with a machine with two ethernet ports, one at 10.0.0.xxx and one at 192.162.1.xxx.

And first try with ping - your firewall may block other traffic.

AronAtGit, 2023-04-01:
Ahhh ok that makes sense.  Yes, the machine has wifi enabled and one ethernet port.  I'm obviously no networking guru but will keep trying.  Thank you!

AronAtGit, 2023-04-01:
That did the trick, thanks so much!

closed 2023-08-14 by terjeio