grblHAL/core #284: Machine pause randomly for some time and resume

muhammadelmogy, 2023-04-11:
I just have finished an upgrade to my machine , I was using grbl-mega-5x on Arduino Mega and using bCNC.
I upgraded to grblHAL  on stm32f411 (currently prototype on a bread board) and I am using bCNC also.

The machine is running just fine, Homing, jogging and G-Code from MIDI all works fine.
However, when I start a job, the machine starts and after sometime, randomly, the G-Code execution pause for several seconds maybe up to less than a minute and then resume and I mean by pause that the controller software(I tried this on bCNC and ioSender) stop sending the G-code to the hardware controller and I know this by looking on the code view in ioSender, the lines was just flowing one after the other and suddenly the line flow stops for sometime and the machine at this moment freeze and the motors still enabled.

This behavior (the pause and resume) happens as it appears randomly and there is no definitive time line for it to happen nor any specific duration, some time happen only once in a long time and sometime and sometimes may happen twice within a couple of minutes !!
All my inputs(start-halt-hold-probe) are going through opto-isolators and also all the limit switch are the same
Also all my outputs(Stepper drivers signals also run through opto-isolators)
Any suggestions?

muhammadelmogy, 2023-04-11:
I want to add some screen shots from ioSender

[image]
[image]

First one when the pause happens and the second one after the resume!
I run a program for about 25 minutes and it happened around 5 times(may be 6) and I have noticed that the pause is around less than 50 seconds always

muhammadelmogy, 2023-04-12:
I have forgotten to add my settings
[settings.txt](https://github.com/grblHAL/core/files/11207162/settings.txt)

terjeio, 2023-04-12:
I have now run the first file several times and from the _Run time_ in ioSender I cannot see that there are any random pauses unless it happens on the sender side. The _Run time_ is consistently 22:23 plus/minus a few seconds, likely from me beeing slow to act on the initial tool change. FYI I have not yet spent 22 minutes watching the sender so pauses may still be there - but if so they must be, IMO, the same for every run.

Can you check the following on your side during the pauses:
Does the _Run time_ clock stop?
In the _Console_ check _Verbose_ and _Show all realtime responses_. Are the responses still coming regularly?

Finally, have you tried with more than one sender computer?

muhammadelmogy, 2023-04-12:
As I have mentioned, I have tried this from ioSender and bCNC.
Also I the pauses are not consistent it seems random to me!
The clock is running just fine and the UI is responsive.
When I press hold after the pause action  happened the hold press does not do anything and If I try press start nothing happens also.

The pauses happens in the middle of G-code execution, there are no pauses nor any thing in the G-code to pause the machine only normal G1 and G0 movement. 

If I enable the Verbose it will make ioSender very slow and not responsive at all and I have to close it. Also the machine act very slow and it acts like it is jumping not smoothly moving! so I could not enable the Verbose.
I will open an issue for this soon in the ioSender repo.

I am using a fairly decent Laptop running windows 11 never the less I will try another Laptop but I do not think it is something related to the host machine since it was running before with grbl mega 5x and also with another files with grblHAL

terjeio, 2023-04-12:
> The clock is running just fine and the UI is responsive.

Ok, then what is your _Run time_ for the first file? Comparable to mine?

> If I enable the Verbose it will make ioSender very slow and not responsive at all and I have to close it. 

This is odd - what is the _Poll interval_ setting in _Settings: App_?

Here is the typical load (often less) with the 3D viewer visible and verbose mode enabled in the console while running the first file. Win10 desktop with 16Gb RAM.

[image]

No changes in the responsiveness of the sender that I can detect with verbose enabled. _Run time_ 22:25.

muhammadelmogy, 2023-04-13:
I have tested the whole setup from another laptop and the same issue happened again!
It is a modern windows11 Laptop. the other one is several years older!
Also, the issue with verbose is still present exactly as before but when the event happened I was able to enable the verbose and I got this
[image]

It seems to be normal or not I do not know!
Also here is my settings:app
[image]

I will try to run the first file now to the end at feed rate override 200% and give you the time

muhammadelmogy, 2023-04-13:
I just finished running the full first file on 200% feed rate override and with just a couple of pauses(which is less than before) in around 20:47

terjeio, 2023-04-13:
> at feed rate override 200%

This is what was missing....
From my initial checks it seems like it could be a silicon bug - the stepper timer randomly overflows without beeing reset as it should. Changing the timer to use downcounting instead of upcounting seems to fix it, as does changing the interrupt priority. It could be the USB code interfering with the timer, I'll check if the behaviour changes by switching to UART comms.

muhammadelmogy, 2023-04-13:
> From my initial checks it seems like it could be a silicon bug

Do you mean there could be a problem in the stm32f411 chip itself ?

I have tested the file now on a normal feed rate and the same thing happended again!

terjeio, 2023-04-13:
> Do you mean there could be a problem in the stm32f411 chip itself ? if it was true how could this be verified?

I can check with a F446... I have verified that the issue only appears when USB comms is enabled, UART comms works. This means that it _could_ be the USB code interfering with the timer. What is strange is that changing to downcounting mode or increasing the IRQ priority to 0 fixes the issue.
Do you compile locally or use the Web Builder? If compiling locally you can add a line of code to switch to downcounting mode [here](https://github.com/grblHAL/STM32F4xx/blob/0b4e6c21310c5e2ca1b4528eb30810c53a33fd64/Src/driver.c#L1981):

`    STEPPER_TIMER->CR1 |= TIM_CR1_DIR;`

muhammadelmogy, 2023-04-13:
I am compiling on my mahine , I will give it a try later and let you know about it (hopfully it is not a sillicon issue!)

muhammadelmogy, 2023-04-13:
It would be very good Idea to confirm this on other hardware(446 as you mention)

muhammadelmogy, 2023-04-14:
>Do you compile locally or use the Web Builder? If compiling locally you can add a line of code to switch to downcounting mode [here](https://github.com/grblHAL/STM32F4xx/blob/0b4e6c21310c5e2ca1b4528eb30810c53a33fd64/Src/driver.c#L1981):
> 
> ` STEPPER_TIMER->CR1 |= TIM_CR1_DIR;`

Do you want me to add it after the refered line or to replace it with the refered line?
When I replace it, the steppers does not work if I try to move them(or home) and ioSender says that it does not understand the output of the controller!

terjeio, 2023-04-14:
Add - as I wrote above. I does not matter if before or after.