grblHAL/core #286: Constant rotation of an axis by just providing RPM.

MuhammadQaisarAli, 2023-04-15:
I want to rotate `X-Axis` just like spindle. But I want to rotate it continuously with a commanded RPM.
`X10Y1Z4` means that rotate X with 10RPM , move Y to 1 and Z to 4. It would be nice if we can modify each axis. And synch the rpm with over all feed as well (Incase of feed override).

terjeio, 2023-04-15:
This is not easy/possible to implement and is not supported by standard gcode semantics. What is the use case for this?

LoganFraser, 2023-04-29:
Probably a stepper or servo driven lathe, given the context.

terjeio, 2023-05-02:
Closed due to lack of feedback.

closed 2023-05-02 by terjeio

MuhammadQaisarAli, 2023-05-02:
> This is not easy/possible to implement and is not supported by standard gcode semantics. What is the use case for this?

Yes you are right, we don't have standard g-code support and that's the challenge to be faced. A used case would be filament winding machine.

MuhammadQaisarAli, 2023-05-02:
> Probably a stepper or servo driven lathe, given the context. 

Hmm.. that could be a possible option.

MuhammadQaisarAli, 2023-05-02:
> A stepper driven spindle controlled by the S parameter is possible. Hijacking the X parameter and controlling it through the planner etc. is not something I would attempt to implement - that path is designed for synchronous motion.

Yes but that won't be an accurate solution, rather a go on