grblHAL/core #290: Homing by Z-pulse of servo

karoria, 2023-04-16:
Hi Terje,
I want to accurately home using Z-pulse of servo/easyservo drive. I also have hard limit switches for homing but want to explore homing like "First seek to limit switch and then slowly feed till Z-pulse signal received from drive. I know, I have to use 2 inputs for each axis in this case, but an example code or exact location where I need to modify code will be a great help.
Thanks,
Ravi

terjeio, 2023-04-16:
There is no need to change any code. Trap [hal.homing.get_feedrate](https://github.com/grblHAL/core/blob/b6e43d6c194fa0f830ff3987a4aa1cd78b310dd6/hal.h#L171-L181) in a plugin and switch inputs based on the [mode parameter](https://github.com/grblHAL/core/blob/b6e43d6c194fa0f830ff3987a4aa1cd78b310dd6/machine_limits.h#L29-L34) by replacing the `hal.homing.get_state` function.

karoria, 2023-04-16:
> There is no need to change any code. Trap [hal.homing.get_feedrate](https://github.com/grblHAL/core/blob/b6e43d6c194fa0f830ff3987a4aa1cd78b310dd6/hal.h#L171-L181) in a plugin and switch inputs based [...]

Does this mean, I have to write a new plugin? or should I edit machine_limits.h directly?

terjeio, 2023-04-16:
Plugin code is your best option - modifying the core means that you have to update it for new releases.

closed 2024-04-16 by terjeio