grblHAL/core #294: Strange vibration

Demitrius, 2023-04-30:
I understand, vibration is difficult qustion.

If I try slow or fast speed - looks god, minimal vibration, but vibration increase when motor works on middle speed.
Who knows, which GRBL settings may affect vibration or I need to check belt, motor, or something else? (hardware)

PS: Simple Nema23 step motor, no encoder

terjeio, 2023-04-30:
> I understand, vibration is difficult question.

Resonance is a likely culprit - search for [stepper motor damping](https://duckduckgo.com/?t=ffab&q=stepper+motor+damping&ia=web).

> Who knows, which GRBL settings may affect vibration

Microstepping (steps/mm).

Demitrius, 2023-04-30:
You mean add dampfer or remove?

I have one dampfer under motor mount

Demitrius, 2023-04-30:
[image]

terjeio, 2023-05-01:
[Here is one of the links](https://insights.globalspec.com/article/16082/how-to-reduce-stepper-motor-noise-and-vibration-in-motion-control) from the search above. Many tips in that post.
I have no idea what will work for you - however, increasing the microstepping worked for me when I recently switched to grblHAL from a Mach3/Smotstepper combo for my router/mill.

darkfibre-nl, 2023-06-20:
Here is a bit more info about what you describe,:
https://www.geckodrive.com/support/mid-band-instability/