grblHAL/core #302: Question about the maximum value of the X-axis data

CHTUZKI, 2023-05-15:
HI.
If I send the following command with iosender, will grblhal make an error

`G0 X10000000000000000000 F100`

If the data of the X-axis overflows, will it continue to run from zero, and will the X-axis controlled by grblhal stop at the moment the data overflows? Or will it keep running?

this would be a strange question😁

CHTUZKI, 2023-05-15:
I want the X axis to rotate continuously, can grblhal do it

terjeio, 2023-05-15:
No, the step counters are 32 bit signed integers - max number of revolutions is thus -2,147,483,648 to 2,147,483,647 divided by steps per revolution. And I have no idea what will happen on an under/overflow...

CHTUZKI, 2023-05-16:
Thank you for your answer.

grblhal is very powerful and can be extended very flexibly. Can grblhal export the API of a certain module or function, for example: I only want to use the motor control part of grblhal, because the motor control part of grblhal is very good, can this part be separated and made into a library or export API .

If I use the whole grblhal as a motion parser, it will make the project huge and complicated.

I saw the gcode parsing part that you isolated。Whether it is feasible to separate the motor control part of grblhal, I want to try it.

terjeio, 2023-05-16:
All functions not marked static are available already, if you provide the correct parameters for them you can call them.
Some documentation [can be found here](http://svn.io-engineering.com/grblHAL/html/index.html) (work in slow progress), this and more in the code itself.

You can control the motors directly by calling the [HAL API](http://svn.io-engineering.com/grblHAL/html/hal_8h.html) - mainly the `hal.stepper_*` functions. Next level above that will be the planner as planning and step generation is tigthly coupled.

closed 2023-05-16 by CHTUZKI