grblHAL/core #311: How to define the analog output of M68 for F4?

hanke-cnc, 2023-05-30:
Thanks to the powerful GRBLHAL, I defined many input and output pins of M62/M66 very easily. After testing the M62-M66 codes, everything worked fine. However, when I sent M68 to GRBLHAL, I received error 20. How can I define the analog output pins of M68?

hanke-cnc, 2023-05-30:
I only defined the following content in my header file for testing the input/output of M62-M66. I am not sure if I need to define the analog output pins separately for M68. It seems that analog outputs cannot be on arbitrary pins as they are output by specific pins of the ADC:

#define HAS_IOPORTS
#define AUXINPUT0_PORT          GPIOA
#define AUXINPUT0_PIN           10

#define AUXOUTPUT0_PORT         GPIOB
#define AUXOUTPUT0_PIN          0

terjeio, 2023-05-30:
I [recently added support for PWM](https://github.com/grblHAL/core/blob/master/changelog.md#20230526) type "analog" output to the iMXRT1062, RP2040 and STM32F4xx divers. Currently no board maps has this as an option though. The STM32F4xx has code in place for two outputs, `AUXOUTPUT0_PWM_PORT` and `AUXOUTPUT1_PWM_PORT`.
Available pins are limited to those defined in [driver.h](https://github.com/grblHAL/STM32F4xx/blob/5d7c277cc5af1146db466b75884f46f4d3454f57/Inc/driver.h#L278-L436) and the pins has to be defined in the map file in the same way as for spindle PWM output. Support for DAC output has to be added to the new [analog output code](https://github.com/grblHAL/STM32F4xx/blob/master/Src/ioports_analog.c) or as board specific code. I believe only two pins on the STM32F4xx can be used for DAC output, and I am not sure all variants has thhis option.

closed 2023-05-30 by hanke-cnc

hanke-cnc, 2023-06-11:
> I [recently added support for PWM](https://github.com/grblHAL/core/blob/master/changelog.md#20230526) type "analog" output to the iMXRT1062, RP2040 and STM32F4xx divers. Currently no board maps has this as an option though. The STM32F4xx has code in place for two outputs, `AUXOUTPUT0_PWM_PORT` and `AUXOUTPUT1_PWM_PORT`. Available pins are limited to those defined in [driver.h](https://github.com/grblHAL/STM32F4xx/blob/5d7c277cc5af1146db466b75884f46f4d3454f57/Inc/driver.h#L278-L436) and the pins has to be defined in the map file in the same way as for spindle PWM output. Support for DAC output has to be added to the new [analog output code](https://github.com/grblHAL/STM32F4xx/blob/master/Src/ioports_analog.c) or as board specific code. I believe only two pins on the STM32F4xx can be used for DAC output, and I am not sure all variants has this option.

Great news! M68 can now output PWM.

hanke-cnc, 2023-06-11:
> I [recently added support for PWM](https://github.com/grblHAL/core/blob/master/changelog.md#20230526) type "analog" output to the iMXRT1062, RP2040 and STM32F4xx divers. Currently no board maps has [...]

Thank you wholeheartedly for your work, which has made GRBLHAL as powerful as LinuxCNC.