grblHAL/core #316: Rotary Axis configuration

karoria, 2023-06-05:
Hi @terjeio 
Sorry for bothering you for the topic already discussed IIRC. I couldn't find any issue or wiki page so opened this new issue. I want to configure A axis as rotary without soft limits (infinity travel both sides). It should also not scale when I switch G20 and G21 as I want to use angles in degrees. What will be unit of $103 setting then? Can I assume it steps per revolution or steps per degree?

terjeio, 2023-06-05:
`$376=1` for setting A as a rotary axis, ioSender changes the unit from "mm" to "deg" for a number of related setting on the next connect. AFAIK other senders do not read settings info from the controller so will still show "mm" if any unit at all.
`$133=0` (maximum travel) for unlimited travel, no soft limits.

karoria, 2023-06-05:
Thanks. That solves my doubt.

closed 2023-08-26 by terjeio