grblHAL/core #320: Enable spindle synchronized compilation, and the firmware will crash after sending "?" about 600-1000 times.

hanke-cnc, 2023-06-10:
For the sake of testing rigor, I re-downloaded the latest core and F4 driver source code. For the sake of testing rigor, I only enabled #define SPINDLE_SYNC_ENABLE 1 and #define BOARD_CNC3040 in my_machine.h. I tested the serial port and USB virtual serial port on both F407 and F401 respectively. For the sake of testing rigor, I only used the original serial port sending tool to send "?" every 100 milliseconds. No other commands were sent after power-on.

hanke-cnc, 2023-06-10:
Testing with the spindle synchronized compilation option disabled showed no issues.

terjeio, 2023-06-10:
Do you have an encoder connected or are the encoder inputs floating?

hanke-cnc, 2023-06-11:
> Do you have an encoder connected or are the encoder inputs floating?
Currently, my STM32F407 is just a development board. After flashing the firmware, it only sends "?" for testing purposes, so there is no connection to any IO pins.

hanke-cnc, 2023-06-11:
> Do you have an encoder connected or are the encoder inputs floating?

I followed your advice and changed the spindle pulse to PD2. I tried connecting it to both GND and 3V3, but it still did not work. This problem may be quite challenging, so for now, I have only brought out their pins without enabling spindle synchronization in the compilation.

terjeio, 2023-06-11:
I have now run a F411 Blackpill with a simulator connected to the encoder pins and the spindle running (providing encoder inputs) for > 10 hours with no issues.
Try with the encoder pins shorted to ground?

hanke-cnc, 2023-06-11:
> I have now run a F411 Blackpill with a simulator connected to the encoder pins and the spindle running (providing encoder inputs) for > 10 hours with no issues. Try with the encoder pins shorted to ground?

Even when I connect both the spindle pulse port and the spindle index to GND, the spindle still dies after a certain amount of time after power-up.

hanke-cnc, 2023-06-11:
Perhaps there is nothing wrong with grblhal, and since I currently do not have a lathe for testing, I can put this problem on hold for now.

closed 2023-06-11 by hanke-cnc