grblHAL/core #326: STMH723FZT6

MrKapim, 2023-07-03:
Hi Terjeio,  I bought a SKR3 and it arrived with STM32H723FZG6. There is the driver for H743 here and I will use it. My doubt is... will it work? Any tips, help will be welcome. Questions... 1. If it doesn't work, can you tell me what to change? 2. Will be there an update for the STM32H723FZG driver?
Thanks in advance...

hanke-cnc, 2023-07-04:
hi, I'm not Terjeio, but I can share my experience, I use the drivers of H743, they work well on H743 and H750, I also tried H7B0, maybe it's the difference of running frequency, I modified the code of SystemClock_Config, GRBLHAL will act weird even though it works, I have given up on the H7 series for a long time due to the exorbitant price, the F4 series is great in every way

dresco, 2023-07-04:
Hi @MrKapim, am the author of the H7 driver.. 

I don't think it should be a problem to add support. As far as I can tell, the 723 was just used as a replacement part without any other board changes?

There is very little info in the BTT GitHub repo - am assuming it's the STM32H723**VG** part - don't know if you are able to see from the IC itself? It has less RAM and FLASH than the 743, but neither should be an issue with current configurations.

MrKapim, 2023-07-06:
Hi Dresco,

Sorry by delay to respond/communicate…

I initially tried to put the H743 firmware. USB doesn't work. I changed some parameters... It didn't work. I switched to Marlin... I loaded it as H743 and that didn't work either. I changed it to H723 and finally the USB worked. Got micro-SKR3 communication...

Can you help me? Because I understood that with the H743 driver the SKR H723 board does not communicate (USB). And there is still doubt... and the rest works?

dresco, 2023-07-06:
> Can you help me? Because I understood that with the H743 driver the SKR H723 board does not communicate (USB). And there is still doubt... and the rest works?

Yes, I should have a bit more time for a look at this next week. I'll try to create build options for the H723 variant, with and without SKR bootloader support (I still need to create a bootloader build for the H743 also).

Although the topic title is "STMH723FZT6", I don't think that can be right? Am assuming the H723 was a drop in replacement for the H743 on the same board, so should be a 100 pin LQFP? Can you tell me what the 2nd line of the writing on the chip says? I would expect it to start VET or VGT..

MrKapim, 2023-07-06:
There was a typing error. The correct one is STM32H723GZT6. For the other questions, here's a photo of the baby...

[image]

Thank you

MrKapim, 2023-07-06:
Typing error again. Sorry... 
But the pic show the correct code

dresco, 2023-07-14:
Hi @MrKapim, am making progress I think. 

What stepper drivers are you using on your board, I'll generate a test binary for you to try? I haven't sorted the bootloader support yet, presume you are okay to flash with DFU for now?

MrKapim, 2023-07-14:
Hi  @dresco, good news!!! 
About the drive, I'm using the 5160EZ. Regarding the firmware, I use platformio to compile and to burn it, I put it directly on the sdcard (I copy the firmware.bin file)...

dresco, 2023-07-14:
> I put it directly on the sdcard (I copy the firmware.bin file)...

Ah okay, I haven't added  support for the SDCard bootloader yet. Will do that next then..

MrKapim, 2023-08-07:
Hi @dresco, I hope everything is alright. I was traveling for work. Any news on the 723?

dresco, 2023-08-08:
Hi @MrKapim, apologies - got distracted with other things ;)

Can you try this test binary for me? Should be built for H723, with 5160 driver & bootloader support.

[firmware.zip](https://github.com/grblHAL/core/files/12290101/firmware.zip)

I don't have a H723 part for testing, so please let me know how it goes..

MrKapim, 2023-08-08:
Hi @dresco. I downloaded the file and installed it on the board. Unsuccessfully. It still does not identify the usb, that is, it still does not establish communication.

dresco, 2023-08-09:
Hmm okay, thanks for letting me know. Guess I may need to buy an H723 dev board to investigate further...

MrKapim, 2023-08-09:
Hi @dresco. Understood. Can you send me the binary project so that I can also try to identify, help solve the usb problem. My email is gaspar@wgaspar.com

dresco, 2023-08-10:
Sure, I'll get some updates pushed to my GitHib repo. 

Out of interest, what makes you think it's a USB problem, have you done any testing that suggests other functionality is working okay?

MrKapim, 2023-08-10:
Hi @dresco,
I put as a USB problem because without communication, how will I control my machine? In the case of the test I performed with your binary firmware, I loaded the firmware on the board and restarted.  Because there was no USB communication, I stopped the reviews.

dresco, 2023-08-10:
> Hi @dresco, I put as a USB problem because without communication, how will I control my machine? In the case of the test I performed with your binary firmware, I loaded the firmware on the board and restarted. Because there was no USB communication, I stopped the reviews.

Got it, thanks. Was just wondering if (for instance) you'd already managed to debug the issue to the USB peripheral..

MrKapim, 2023-08-10:
I haven't tried to debug it yet because it was another project, H743 controller, that I had.  They have different frequencies (and also different internal frequency dividers).  I think this is why the USB is not working and asked you the project of binary test...

dresco, 2023-08-12:
I've pushed a [723 branch](https://github.com/dresco/STM32H7xx/tree/h723) to my github repo if you would like to test with that. Commit history from the last couple of days will show everything that's changed..

There is a new PlatformIO build environment `btt_skr_30_h723_tmc5160_bl128`, which is pre-configured for the BTT bootloader, and 5160 Trinamic drivers. (This is the binary I sent previously that didn't work).

There are also some new STM32CubeIDE build entries. If using those, you'd probably want to start with the `Release H723 BTT SKR 3 (BL)` entry, and update the build settings as needed for the 5160 drivers (currently configured for 2209).

I plan to order a H723 Nucleo dev board for further testing here..

dresco, 2023-08-17:
Hi @MrKapim 

I received my H723 dev board, and found a couple of issues. Can you try this new build please?

[firmware.zip](https://github.com/grblHAL/core/files/12368609/firmware.zip)

MrKapim, 2023-08-18:
Hi @dresco 
OK. I downloaded the file. After tests, return the results...

MrKapim, 2023-08-18:
Good news!!! After connecting the USB cable, the interface appears on the VM... But it still doesn't complete the connection. Tried with IOSender and UGS. The IOSender closes informing that it was unable to connect and the UGS shows that it is connected, but also does not show any result or machine status. I opened windows systems and the communication interface is present (in my case port 5). So now we have a communication port but the connection is still not effective.

dresco, 2023-08-18:
Hmm unsure then. Have tested USB CDC successfully on my dev board here..

Have you tried attaching a terminal program to the port, no output on startup? Only significant difference should be the clock setup, but that all looks correct to me..

MrKapim, 2023-09-01:
Hi @dresco 
Forgive me for the long time in replying. My work is in another country and I had to extend my stay. I tested your suggestion and also with LighBurn. There is communication. Take a look the picture (is from the LightBurn). Him show information/answer from the board about no communicate with the motor drives...
[image]

MrKapim, 2023-09-01:
Oh yes, and in the terminal the message is the same (Could not communicate with stepper driver)

dresco, 2023-09-03:
Good news it's at least starting correctly now.

Have you looked at the [Trinamic plugin](https://github.com/grblHAL/Plugins_motor) page? While troubleshooting, it may be worth enabling just the X axis ($338 setting) for testing.

I only have the 2209 EZ driver modules for my SKR board, but do have a 5160 breakout that I used for initial porting of the Trinamic support. Will get that hooked up to my dev board this week to sanity check the 5160 code is still working here.

Cheers. Jon

MrKapim, 2023-09-04:
Hi @dresco,  I noticed that there were updates. I downloaded them, compiled them and loaded them on the board. Works correctly, USB communication, drives 5160 and all.
Thank you very much for your patience and effort to arrange what I had asked for.
I'm still new to using GRBLHAL, so I'm likely to seek help again...
Thanks again

MrKapim, 2023-10-14:
Hi @dresco , I'm here again. Sorry about the long time. Please help me. The motors don't moves and I don't have messages or error codes.

MrKapim, 2023-10-14:
I'm using the IOSender to comunicate with board btt723 (SKR723EZ)