grblHAL/core #329: Homing only moves one motor of ganged axis

hersi97, 2023-07-14:
Hello!

Currently I'm trying to use grblHAL with a small dual Z axis machine.
The board I'm using is BigTreeTech SKR mini e3. (TMC2209 drivers)

The issue I'm facing is whenever I try to home the Z axis only one motor moves, and racks the gantry before I have to EStop it.
If I manually rotate the Z axes close to the limit switches, and home the Z, I get:
```
An unexpected error was detected: (ALARM:10) Homing fail. Second dual axis limit switch failed to trigger within configured search distance after first. Try increasing trigger fail distance or check wiring.
[MSG:Emergency stop - clear, then reset to continue]
```

Weirdly, if I jog the machine both Z axis motors move without problems.

I'm using grblHAL/STM32F1xx repo as a base. (only I added fans and laser modules)

I have the following enabled in my_machine.h:
```c
#define Z_GANGED            1
#define Z_AUTO_SQUARE       1
```

I'm using the E0 output as the Z2 pins.
```c
// Z2
#define M3_STEP_PORT            GPIOB
#define M3_STEP_PIN             3  //PB3
#define STEP_OUTMODE            GPIO_BITBAND
#define M3_DIRECTION_PORT       GPIOB
#define M3_DIRECTION_PIN        4  //PB4
#define DIRECTION_OUTMODE       GPIO_BITBAND
#define M3_ENABLE_PORT          GPIOD
#define M3_ENABLE_PIN           2  //PD2
#define M3_LIMIT_PORT           GPIOC
#define M3_LIMIT_PIN            15 //PC15
#define LIMIT_INMODE            GPIO_BITBAND
```

My settings are the following:
```
>>> $$
GrblHAL 1.1f ['$' or '$HELP' for help]
$0 = 10.0    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 7    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$8 = 0   
$9 = 1   
$10 = 511    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 6   
$15 = 0   
$16 = 0   
$17 = 0   
$18 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 3    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.0    (Homing locate feed rate, mm/min)
$25 = 300.0    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$28 = 0.100   
$29 = 0.0   
$30 = 1000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$33 = 5000.0   
$34 = 0.0   
$35 = 0.0   
$36 = 100.0   
$37 = 0   
$39 = 1   
$40 = 0   
$43 = 1   
$44 = 4   
$45 = 3   
$46 = 0   
$62 = 0   
$63 = 1   
$64 = 0   
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 12000.000    (X-axis maximum rate, mm/min)
$111 = 12000.000    (Y-axis maximum rate, mm/min)
$112 = 300.000    (Z-axis maximum rate, mm/min)
$120 = 1000.000    (X-axis acceleration, mm/sec^2)
$121 = 1000.000    (Y-axis acceleration, mm/sec^2)
$122 = 100.000    (Z-axis acceleration, mm/sec^2)
$130 = 235.000    (X-axis maximum travel, millimeters)
$131 = 235.000    (Y-axis maximum travel, millimeters)
$132 = 250.000    (Z-axis maximum travel, millimeters)
$140 = 650   
$141 = 650   
$142 = 650   
$150 = 16   
$151 = 16   
$152 = 16   
$170 = 0.000   
$171 = 0.000   
$172 = 0.000   
$180 = 25.0   
$181 = 25.0   
$182 = 5.0   
$190 = 600.0   
$191 = 600.0   
$192 = 100.0   
$200 = 0.0   
$201 = 0.0   
$202 = 0.0   
$210 = 50   
$211 = 50   
$212 = 50   
$220 = 0.0   
$221 = 0.0   
$222 = 0.0   
$339 = 0   
$341 = 0   
$342 = 30.0   
$343 = 25.0   
$344 = 200.0   
$345 = 200.0   
$347 = 5.0   
$346 = 1   
$348 = 0.000   
$349 = 25.000   
$384 = 0   
$398 = 35   
$481 = 0   
```

```
>>> $i
[VER:1.1f.20230704:]
[OPT:VNMHSL2,35,512,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,ES,SED,TMC=7,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM]
[DRIVER:STM32F103RC]
[DRIVER VERSION:230702]
[BOARD:BTT SKR MINI E3 V2.0]
[PLUGIN:Trinamic v0.12]
[PLUGIN:SDCARD v1.09]
[PLUGIN:FS macro plugin v0.03]
```

terjeio, 2023-07-15:
What is the `$LIM` output when:
* no z-limit switches are triggered
* primary z motor limit switch is triggered
* secondary z motor limit switch is triggered

hersi97, 2023-07-16:
Hi, sorry for the wait

```
>>> $LIM
[LIMITS:,,,]
ok
[MSG:Reset to continue]
An unexpected error was detected: (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
>>> $LIM
[LIMITS:Z,,,]
ok
An unexpected error was detected: (ALARM:10) Homing fail. Second dual axis limit switch failed to trigger within configured search distance after first. Try increasing trigger fail distance or check wiring.
[MSG:Emergency stop - clear, then reset to continue]
>>> $LIM
[LIMITS:,,Z,]
ok
```

hersi97, 2023-07-16:
Ok, if I understand correctly, $LIM prints out `[<min>,<max>,<min2>,<max2>]`
In that case, this is wrong, I have 2 Z max limit switches.

I don't see a way to config these as max limit switches in the my_machine_map.h, if I look at motor_pins.h, I only see it defining Z_LIMIT_PIN_MAX (motor_pins.h:428):
```c
 #elif Z_GANGED_LIM_MAX && !defined(M3_LIMIT_PIN_MAX)
  #define Z_LIMIT_PORT_MAX  M3_LIMIT_PORT
  #define Z_LIMIT_PIN_MAX   M3_LIMIT_PIN
  #define Z_LIMIT_BIT_MAX   (1<<M3_LIMIT_PIN)
 #endif
```
I also can't find any mention of something like Z2_LIMIT_PIN_MAX macro in the source.

terjeio, 2023-07-16:
Homing is always towards the min limit switches. And IIRC no drivers has support for two max limit inputs on any axis as this is not (IMO) really needed in practice.

Anyway, this does not explain why only one motor is moving initially. Only an immediately triggered limit switch (after homing is started) should cause that. And since you have hard limits enabled and jogging works it is unlikely you have EMI issues.

I'll wire a F103 board up to my simulator and see if I can replicate the issue. I'll try to get it done tomorrow.

hersi97, 2023-07-16:
Ah okay, I'll remount them to be min limits then.
Thank you

Here's my config for reference:
I commented out `-D OVERRIDE_MY_MACHINE` and added this to platformio.ini
```ini
[env:BTT_SKR_MINI_E3_V20_USB_CUSTOM]
board = genericSTM32F103RC
board_build.ldscript = STM32F103RCTX_FLASH.ld
build_flags = ${common.build_flags}
  -I fans
  -I laser
  # See Inc/my_machine.h for options
  # Relocate the vector table where the boot loader expects to find them
  -D VECT_TAB_OFFSET=0x7000
  -Wl,--defsym=LD_VECT_TAB_OFFSET=0x7000
lib_deps = ${common.lib_deps}
  eeprom
  trinamic
  FatFs
  sdcard
  USB serial support
  fans
  laser
lib_extra_dirs = ${common.lib_extra_dirs}
  fans
  laser
```

```c
/*
  my_machine.h - configuration for STM32F103xx ARM processors

  Part of grblHAL

  Copyright (c) 2020-2023 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/
#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!
#define USB_SERIAL_CDC       1 // Serial communication via native USB. Comment out for UART communication.
#define SDCARD_ENABLE        1 // Run gcode programs from SD card, requires sdcard plugin.
#define FANS_ENABLE          1
#define PPI_ENABLE           1
#define Z_GANGED            1
#define Z_AUTO_SQUARE       1
#define DEFAULT_LASER_MODE  1
```

```c
/*
  my_machine_map.h - driver code for STM32F103RC ARM processors

  Part of grblHAL

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
[... 107 lines cut ...]
#define I2C_STROBE_PORT         GPIOA
#define I2C_STROBE_PIN          1 //PA1
#endif

#if SDCARD_ENABLE
#define SD_CS_PORT  GPIOA
#define SD_CS_PIN   4 //PA4
// The following defines are not used but defined for reference
// Port init is done by HAL_SPI_MspInit() in stm32f1xx_hal_msp.c
#define SD_IO_PORT  GPIOA
#define SD_SCK_PIN  5 //PA5
#define SD_MISO_PIN 6 //PA6
#define SD_MOSI_PIN 7 //PA7
#endif

// Define auxiliary outputs
#define AUXOUTPUT0_PORT     GPIOC
#define AUXOUTPUT0_PIN      6 //PC6
#define AUXOUTPUT1_PORT     GPIOC
#define AUXOUTPUT1_PIN      7 //PC7
#define AUXOUTPUT2_PORT     GPIOC
#define AUXOUTPUT2_PIN      8 //PC8
#define AUXOUTPUT3_PORT     GPIOC
#define AUXOUTPUT3_PIN      9 //PC9
```

terjeio, 2023-07-17:
Homing for auto squared Z works as it should in my test setup (using a SVN board).
You have one major issue with your board map, the M3 limit switch input is shared with Reset (E-Stop) on PC15 - not good. But again this does not explain why only one motor moves initially.

In order to debug this can you add calls to `report_current_limit_state(0, NULL)` at line 386 and before line 400 in [machine_limits.c](https://github.com/grblHAL/core/blob/0c6f3f6dd722e70bc6100ff384d252a9f5bff500/machine_limits.c#L386C1-L402)? This will output limit pin states before homing starts and when an auto-squared motor is disabled. Is the call at line 400 executed immediately after homing starts?

hersi97, 2023-07-17:
> You have one major issue with your board map, the M3 limit switch input is shared with Reset (E-Stop) on PC15 - not good. But again this does not explain why only one motor moves initially.

Oops, yeah, I did not realize that, I changed it to PC3, it's free, I don't have the safety door enabled.

And yes, this was indeed the solution to the homing problem.

Here's the output from the console:
```
>>> $HZ
[LIMITS:,,,]
[LIMITS:Z,,,]
[LIMITS:,,Z,]
[LIMITS:,,,]
[LIMITS:,,Z,]
[LIMITS:Z,,Z,]
[MSG:]
ok
```

terjeio, 2023-07-19:
> I am willing to do some more testing on my machine, if you want

Yes, please report any issues you might encounter.
Is the board map for a board you have made? Is it available, either to buy or as a public design?

hersi97, 2023-07-21:
I used the machine for a couple days now, it's running fine, without much fuss. I didn't have to move the positive Z limits either. I just had to get used to how multiple coordinate systems work (G53-G59 and G28) and how CNC machines home to a "negative" space.

The machine is basically a heavily modded Ender 3 Pro that I had spare, since I bought an other 3d printer. This Ender 3 has two independent Z axes, a diode laser head, linear rods and bearings, solid aluminium bed stud spacers (spring delete) and a silent stepper board (BigTreeTech SKR E3 Mini v2.0). I thought I would repurpose it to do some laser engraving. E.g. on photoresist with simple single or double sided PCBs that I wouldn't want to order from something like AISLER and such.