grblHAL/core #335: Axes moves together

userfuckingname666, 2023-07-23:
Hi! I have strange GRBLHAL behavior on STMF103C8T6 (128): any G1 x/y commands moves only X-axis in negative direction. At the same time Z_axis moves correctly. Firmware was created by web builder for STM/CNC3040 with following settings:

Native USB
x-single motor
y-single motor
z-single motor
probe unchecked

Compatibility: GRBL complete
Kinematics:CoreXY

Any other options are unchecked and disabled

Motors wired as:
[A0, A1, A2, A3, A4, A5]
[Xs,  Xd, Ys,  Yd,  Zs, Zd]

Also i haven't any limits or sensors connected.
What the problem may consist of?

terjeio, 2023-07-24:
> What the problem may consist of?

Bad wiring, damaged prosessor, bad stepper driver, ...
I have verified the firmware by connectiong some LEDs and the X direction pin works as it should. Are you able to measure the voltage on A1? Does it change on direction changes?

terjeio, 2023-07-24:
This is the firmware I used for testing, made by the Web Builder today:

[firmware.zip](https://github.com/grblHAL/core/files/12143099/firmware.zip)

userfuckingname666, 2023-07-24:
I've installed 6-AXIS-USBCNC-GRBL and my machine is busy now so i can't  test it for yet. There's size difference 2KB between your firmware and my one. Probably, our build settings are not identical. 

In case you are interested in solving this problem, my build setting are following:

Driver: STM32F1xx
Board: CNC3040(Red, bluepill 128k)

Connection: Native USB
Number of axes: 3
Remap ABC to UVW: unchecked
X,Y,Z axis: single motor
Probe input: unchecked
Safety door input: unchecked

EEPROM is FRAM:unchecked
Keypad: disabled
Macros: disabled
Odometer: unchecked

Compatibility level: Grbl complete(10)
TLO report: All axes
Kinematics: CoreXY

Template plugins:none
All other available checkboxes set to unchecked state

Resulting firmware size: 124324 bytes

terjeio, 2023-07-24:
> Resulting firmware size: 124324 bytes

I've tested with this - same result as previous test, X direction output changes.

```
$pins
[PIN:PB5,Reset]
[PIN:PB14,Feed hold]
[PIN:PB15,Cycle start]
[PIN:PB13,Probe]
[PIN:PB10,X limit min]
[PIN:PB11,Y limit min]
[PIN:PB12,Z limit min]
[PIN:PA0,X step]
[PIN:PA2,Y step]
[PIN:PA4,Z step]
[PIN:PA1,X dir]
[PIN:PA3,Y dir]
[PIN:PA5,Z dir]
[PIN:PB9,Steppers enable]
[PIN:PB1,Spindle on]
[PIN:PB0,Spindle direction]
[PIN:PB3,Flood]
[PIN:PB4,Mist]
[PIN:PA8,Spindle PWM]
```

userfuckingname666, 2023-07-24:
Is your X-axis moving when you send "G1 Y.."?

terjeio, 2023-07-24:
From what I can see on the LEDs, yes both motors are moving. And X andY direction signals changes on a direction change.

userfuckingname666, 2023-07-24:
Have you read my last message attentively? Both "G1 X" & "G1 Y" commands produces step signal at A0. Do you have such behavior?

userfuckingname666, 2023-07-24:
> [CoreXY kinematics](http://corexy.com/theory.html) behaves that way.
OMFG! Huge thanks, man!

I have another stupid question. Please tell me how can i unblock GrblHal made with "GrblHal compatibility" option when i haven't any sensors connected? As far as i remember, GrblHal returned alarm codes 79 but i'm not sure. I tried to invert different inputs, but didn't succeed.

terjeio, 2023-07-25:
Did you follow [this guide](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings)?