grblHAL/core #337: Incorrect HALT trigger behavoiur

0lympu5, 2023-07-30:
When the halt input is triggered, the controller resets but then I can jog the CNC around without issue, all while the HALT input remains triggered 

Controller: Flexi-HAL
GCODE Sender: IOsender
Firmware: Latest pre-built image for Flexi-HAL

I was expecting to have to input some form of input or command through IOsender after removing the HALT input to release the machine out of E-STOP. I will mention I am not using this as any form of actual E-STOP, but rather to inform the controller that E-STOP has been pressed.

terjeio, 2023-07-30:
HALT is routed as e-stop or reset?
I routed as e-stop then the input (and ALARM:10) has to be cleared before gcode is accepted:

[image]

The E "LED" is lit in the image above, this means the e-stop input is triggered. if R is lit then reset is.

0lympu5, 2023-07-30:
HALT pins on the FLEXI-HAL appear to trigger the reset light. My IOsender does not have an E LED. Is this something that can be configured within grblhal or is there something in IOsender to get the E-stop LED?

terjeio, 2023-07-30:
> Is this something that can be configured within grblhal or is there something in IOsender to get the E-stop LED?

It is a grblHAL configuration, default is dependent on compatibility mode.
You may try a [Web Builder](http://svn.io-engineering.com:8080/?driver=STM32F4xx&board=Flexi-HAL) build. I guess I should add an option for selecting halt pin routing...

0lympu5, 2023-07-30:
I am having a look at the web builder now, also mentioned it to the designer who noticed the same weird behavior. As the flexi-hal does not have a designated E-STOP input, only door/halt/pause/start

terjeio, 2023-07-30:
The reset signal is handled in the same way as legacy Grbl, it is overridable and if asserted on startup it does not cause an alarm.
I added the option of routing the (halt) input as either reset or estop signal to grblHAL, but not all drivers has this option implemented yet. The estop signal will cause an alarm on startup if asserted, and the alarm cannot be cleared before deasserted. There are other differences in reset and e-stop handling, the main one beeing that e-stop does not go through the same initialization sequence when cleared.

0lympu5, 2023-07-30:
Will have to see if the HALT pin can be re directed to trigger estop rather than reset

terjeio, 2023-07-30:
Did you try a [Web Builder](http://svn.io-engineering.com:8080/?driver=STM32F4xx&board=Flexi-HAL) build?

phomann, 2023-07-30:
To be honest, you should not be relying on IOSender or grblHAL software to implement an E-Stop. If you're system requires an E-Stop to make your machine safe, you need to to use a safety switch/ breaker, etc to remove power from the drives, spindle or whatever you are protecting the operator from. Feeding the E-Stop signal into the halt input of grblHAL is really just a curtesy to let the software know an E-Stop had occurred. 
Cheers 
Peter.

0lympu5, 2023-07-31:
@terjeio I have a build from the web builder to test this evening, as well as one in the works from the board designer over the coming days. I will update with my findings !

0lympu5, 2023-07-31:
> To be honest, you should not be relying on IOSender or grblHAL software to implement an E-Stop. If you're system requires an E-Stop to make your machine safe, you need to to use a safety switch/ [...]

As noted in the first comment, this is only being used to update the controller of the E-STOP event.

0lympu5, 2023-08-01:
The designer for the board was able to provide a beta with both E-STOP and Reset on the same HALT pin (requires both pins to be inverted the same way) 

One thing that was noticed by a few of us, when the E-STOP trigger is removed, you have to unlock and reset the machine to get out of e-stop. It feels a bit clunky having to do both. Does it need both or is there a way to customize it? In my experience with other industrial cnc, you don't have to soft restart. LEt me know your thoughts

terjeio, 2023-08-06:
> One thing that was noticed by a few of us, when the E-STOP trigger is removed, you have to unlock and reset the machine to get out of e-stop. It feels a bit clunky having to do both.

It is how legacy Grbl works.

> Does it need both or is there a way to customize it?

Currently it cannot be customized, I'll check if it is easy to add without risking side-effects.

closed 2023-08-25 by terjeio