grblHAL/core #340: A axis is not working on CoreXY mode on BTT-SKR 2.0

Avara7, 2023-08-10:
Hi,
I'm making 4 axis tangential knife CNC CoreXY machine with  BTT-SKR 2.0 controller (STM32F407). 
On usual mode of X and Y axis it's working  fine with GRBLGRU. After activating  #define CoreXY  On    in config.h file  the A axis is not working properly (when pressing jog button value of position of this axis is not changing ).
Am I doing something wrong or is there some issue with coreXY configuration?

[ver:1.1f.20230228:]
[opt:vnmslw,35,1024,4,0]
[axs:4:xyza]
[newopt:enums,rt+,tc,sed]
[firmware:grblhal]
[nvs storage:*flash]
[driver:stm32f407]
[driver version:230227]

Thank you.

terjeio, 2023-08-10:
Fix commited, issue was with corexy implementation.

Avara7, 2023-08-14:
Hello,

I appreciate your incredibly fast response time, and thank you so much for
resolving the problem!

I've tested a new version, and everything is working smoothly now.

Really grateful for your help on this!

Best regards,

Avara7.

closed 2023-08-25 by terjeio