grblHAL/core #341: Polar kinematic from Esp32_GRBL

Melkiyby, 2023-08-11:
Please help with adding this kinematics to GRBLHal. There are a lot of differences in the GRBL structure that I have not figured out.
[https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/Custom/polar_coaster.cpp](url)

terjeio, 2023-08-11:
The main difference is that the grblHAL planner requests segments from the kinematics implementation, ESP32_GRBL pushes them. The grblHAL wall_plotter kinematics implementation is similar to the polar implementation in that motion is split up into many small segments so IMO can serve as a starting point. The [wp_segment_line()](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/wall_plotter.c#L141-L199) function in grblHAL is equivalent to `cartesian_to_motors()` in ESP32_Grbl.
When a move is executed wp_segment_line() is called with the init argument set true. This signals that the function has to calculate the the number of segments (iterations) and other data needed for the move. This information is stored in local static variables to be used later when calculating the segment positions. After this call the planner will call the same function again with init set false, the function then returns the next position. The function will then be called repeatedly until all segments has been calulated and delivered.

Jog cancel handling is different: the kinematics routine subscribes to the[ jog cancel event](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/wall_plotter.c#L319) and when "fired" [sets a flag](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/wall_plotter.c#L283-L286) that causes `wp_segment_line()` to [return NULL](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/wall_plotter.c#L198). This tells the planner that there are no more segments available.

I have not looked into how homing differs. I see that ESP32_Grbl likely rehomes the machine at end of each job (when M30 is executed). This can be done in grblHAL as well by subscribing to the [grbl.on_program_completed](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/core_handlers.h#L129) event.

terjeio, 2023-08-18:
I have comitted an [intitial version](https://github.com/grblHAL/core/blob/master/kinematics/polar.c) for you to test/verify/develp further. Homing is not implemented yet. Enable in [config.h](https://github.com/grblHAL/core/blob/be9f7db33ed0c43cb1b44c0e40bde699d0ec0570/config.h#L147).

Melkiyby, 2023-09-12:
Thanks for your quick support. I tested it on two different programs and so far everything looks very good, at least there is no problem that was in Marlin with this kinematics. #https://github.com/MarlinFirmware/Marlin/issues/25330
[How its work](https://youtu.be/XUQ7XWm1a6c?si=aBVVocokKkoEGhfW)

Melkiyby, 2024-03-21:
Hi, Terjeio! After much testing, I discovered two bugs: 
1) If, when executing a CNC program, the rotary axis made many revolutions in one direction, then when executing G00 X0.0 Y0.0 F3000, the rotary axis begins to rewind the number of revolutions that it made when executing the program, and at the same time it  still will not accept the original position.  
2) There is not enough feedrate scaling to maintain the correct cutting mode depending on the position of the X axis from the center of the rotary axis.  At the moment, the closer the X axis is to the center, the slower the movements occur.

closed 2024-03-28 by terjeio