grblHAL/core #347: "Differential signal?" Does it look cool? Can grblHAL have it? Just curious.

hanke-cnc, 2023-08-17:
I was just browsing the web and came across it, then got curious and asked you. You know more.
[image]

terjeio, 2023-08-18:
> Can grblHAL have it?

Yes, why not? You will need a board that has differential output drivers that converts the single ended signals from the MCU. The drivers can be on board the controller itself or on an external board.

closed 2023-08-18 by hanke-cnc

andrewmarles, 2023-08-18:
FWIW, FlexiHAL uses differential drivers for the stepper outputs, and also has provision to use the differential encoder input to transport input signals a long way (I have tested with 30m of cable).

hanke-cnc, 2023-08-18:
> andrewmarles

Awesome, andrewmarles! I'll take a look and give your project a try when I have the time. Thank you so much