grblHAL/core #353: Add "extrusion" support for G5 command

devtanc, 2023-08-24:
I'm trying to basically run a sort of 3d printer setup with my current use-case. I currently use the A axis to support the extrusion of my filament. I'm wondering about support for either A axis or the typical "E" command support when it comes to the average command, but specifically the G5 command. Are 3d printer style commands something that this library typically tries to support, or is it focused exclusively on CNC commands? Currently it moves well, but I can't sync up movement with the A axis to be able to extrude along the bezier. Is this something that could be added? Is this something that I may be able to help in adding?
I've been able to identify the `mc_cubic_b_spline` within the `[20. Motion modes ]` section and `MotionMode_CubicSpline` case of the gcode execution code. I just don't have the context to be sure I'm not just adding something that will introduce more bugs, and I'm relatively new to C specifically.

terjeio, 2023-08-24:
>  Are 3d printer style commands something that this library typically tries to support, or is it focused exclusively on CNC commands?

The core is exclusively focused on CNC commands, some plugins have adopted Marlin M-codes/syntax but only when relevant for CNC. I have no intention to support 3D printing, but if someone would like to adopt the architecture for it then I am ok with that.

> Currently it moves well, but I can't sync up movement with the A axis to be able to extrude along the bezier.

Thats odd, does arcs works as intended? Bezier splines works the same way as arcs in that motion is split into a large number of small linear moves that are then handed to the planner.

devtanc, 2023-08-28:
The issue is that the arc command `G5` receives `XY IJ PQ` to specify the start/end and control points, and with certain firmware supports the `E` parameter for extrusion, but it doesn't support adding an additional axis to sync movements there as well. So in the case of this library it will move smoothly along the curve, but won't allow additional movement to happen at that same time, preventing me from extruding at the same time.

My current workaround has been exactly what you said. I simply use linear movement along the curve that I've already calculated, and then I can run `XYZA` at the same time.

Also, I have to say, thank you for your responsiveness. I've really appreciated your help and feedback when I've posted questions.

terjeio, 2023-08-28:
I guess I should always check the code before answering...

The [LinuxCNC spec for G5](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g5) states that it is an error if _An axis other than X or Y is specified_. The reason might be that the segment lengths pushed to the planner are not constant so likely hard/impossible to sync other axes.

kimstik, 2023-11-19:
> I guess I should always check the code before answering...
>
> The [LinuxCNC spec for G5](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g5) states that it is an error if _An axis other than X [...]

BTW what it reason of it? 
I'd like to use the G5 just like the G2 helix, with linear Z interpolation.
Seems G5 limitation is pure synthetic. 
mc_cubic_b_spline may emit dZ by scaling t-parameter.

terjeio, 2023-11-19:
> mc_cubic_b_spline may emit dZ by scaling t-parameter.

Ok, you can give it a try?

kimstik, 2023-11-19:
I made 3D analytic curve of hypotrochoidal grinding path of parabolic mirror reflector:
[image]
[hytro.nc.zip](https://github.com/grblHAL/core/files/13402929/hytro.nc.zip)

it can be nicely fitted into a spline, but unfortunately GRBL/LinuxCNC does not accept the Z axis for splines. Unlike G2/3 circular interpolation (helix).

kimstik, 2023-11-19:
> > mc_cubic_b_spline may emit dZ by scaling t-parameter.
> 
> Ok, you can give it a try?

This is a terribly interesting task. And it looks like it's been solved many times already :)

* https://gamedev.stackexchange.com/questions/5373/moving-ships-between-two-planets-along-a-bezier-missing-some-equations-for-acce/5427#5427
* https://www.geometrictools.com/Documentation/MovingAlongCurveSpecifiedSpeed.pdf
* "Hermite Spline Controller"