grblHAL/core #360: grblHal defaults to ALARM XYZ HOLD CYCLE START with pico and UGS

wyzarddoc, 2023-08-31:
RPi pico connects to ubuntu 22.04 on serial /dev/ttyACM0 but goes directly to the ALARM XYZ HOLD CYCLE START alarm statement. Advance documentation says a mask at $22 could remove alarm but no infor for setting the mask. Is it Hex,octal or binary? The pico is not connected to any inputs or outputs only serial connection for testing. Also where is the configuration file located so I could make changes to it?
thanks

terjeio, 2023-08-31:
See this [Wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).

> Advance documentation says a mask at $22 could remove alarm

`$22` is for setting homing options, `$14` is for inverting the control inputs that usually causes initial alarms. Mask settings are decimal, you can use the $$ command for info. E.g. `$$=14` will output something like this:
```
    1 - Feed hold (2)
    2 - Cycle start (4)
    6 - EStop (64)
Inverts the control signals (active low).
NOTE: Block delete, Optional stop, EStop and Probe connected are optional signals, availability is driver dependent.
```
Add the values in brackets for the options you want. 

The core config file is [found here](https://github.com/grblHAL/core/blob/master/config.h).

terjeio, 2023-08-31:
`1 - Feed hold (2)` -> First number is the bit index, starting from 0 so bit 1, value in brackets is 2, thus use `$14=2` to invert feed hold.
`$14=6` (2+4) will invert feed hold and cycle start. Some examples also in the above linked Wiki page.

wyzarddoc, 2023-08-31:
Thanks for your patients I have found $14=6 brings the alarm down to ALARM:XYZ PROBE . Where can i find the folder containing the start-up config file to make these changes permanent.?
thanks

terjeio, 2023-08-31:
Inver the probe input with `$6=1` to get rid of the last message. BTW the probe pin asserted status is _not_ an alarm state so UGS is wrong about that.

>  Where can i find the folder containing the start-up config file to make these changes permanent.?

The $-settings makes them practically permanent, that is until the core setting format is changed and you flash an update - which is not very often. Default values for the settings are in the [grbl/config.h](https://github.com/grblHAL/core/blob/master/config.h) file, which folder this is stored in on your machine I do not know. If you edit this file please take note of the IMPORTANT information in the header.

wyzarddoc, 2023-08-31:
Thanks but $6=1 did not clear the last message I tried that after you explained the $14=

wyzarddoc, 2023-09-13:
To remove all of the alarm conditions I used $14=70 that worked the best. I am still trying to gather and find all the files needed to compile grbl-hal on my local machine.

closed 2024-04-16 by terjeio