grblHAL/core #362: The strange homing, once it starts homing, it seems to think I've already touched the limit switch and starts moving away from the limit switch distance, and it alarms with code 8 after exceeding the limit switch distance.

hanke-cnc, 2023-09-10:
Manually pressing the limit switch is fine; it can trigger a hard limit alarm.

[image]

hanke-cnc, 2023-09-10:
This is my limit circuit, connecting X_XW to GND can trigger the hard limit normally.
[image]

hanke-cnc, 2023-09-10:
("0", "10.0")
("1", "25")
("2", "0")
("3", "0")
("4", "0")
("5", "0")
("6", "0")
("9", "3")
("10", "4095")
("11", "0.010")
("12", "0.002")
("13", "0")

("14", "71")
("15", "0")
("16", "0")
("17", "0")
("18", "0")

("19", "0")
("20", "1")
("21", "1")
("22", "171")
("23", "0")
("24", "25.0")
("25", "50.0")

("26", "250")
("27", "1.000")
("28", "0.100")

("29", "0.0")
("30", "1000.000")
("31", "0.000")
("32", "0")
("33", "5000.0")
("34", "0.0")
("35", "0.0")
("36", "0.0")
("37", "0")
("39", "1")
("40", "0")

("43", "1")
("44", "4")
("45", "1")
("46", "2")
("47", "0")

("48", "0")
("49", "0")
("50", "100.0")
("51", "600.0")
("52", "3000.0")
("53", "0.250")
("54", "500.0")
("55", "3000.0")
("62", "0")
("63", "3")

("64", "0")
("65", "0")
("100", "640.000")
("101", "640.000")
("102", "640.000")
("103", "640.000")
("104", "640.000")

("105", "640.000")
("110", "2333.000")
("111", "99999.000")
("112", "99999.000")
("113", "99999.000")
("114", "99999.000")

("115", "99999.000")
("120", "50.000")
("121", "50.000")

("122", "50.000")
("123", "100.000")

("124", "100.000")
("125", "100.000")
("130", "99999.000")
("131", "99999.000")

("132", "99999.000")
("133", "99999.000")
("134", "99999.000")
("135", "99999.000")

("160", "0.000")
("161", "0.000")
("162", "0.000")
("163", "0.000")
("164", "0.000")
("165", "0.000")

("341", "4")
("342", "30.0")
("343", "25.0")
("344", "200.0")
("345", "200.0")
("346", "1")
("370", "0")

("372", "0")
("376", "5")
("384", "0")

("398", "512")
("481", "150")

hanke-cnc, 2023-09-10:
Based on STM32F407VET6.

hanke-cnc, 2023-09-11:
"I conducted an experiment and noticed that the wiki mentioned the limit switch being normally closed. Now it can indeed home properly."

hanke-cnc, 2023-09-11:
"I have now connected the limit switch to the normally closed interface, and it's working fine. However, there's something strange. When I send $H for homing, it doesn't move at all. I have to press the limit switch once for it to start homing."

hanke-cnc, 2023-09-11:
"I noticed that the wiki mentioned the limit switch is a normally closed switch. Now, when I connect the limit switch to the normally closed configuration and send $HZ, there is no movement. I have to manually press the limit switch once, and then it moves away from the limit, returns to seek the limit again, and finally completes homing. It's confusing."

closed 2023-09-11 by hanke-cnc

jamesbaber, 2024-12-30:
> "I noticed that the wiki mentioned the limit switch is a normally closed switch. Now, when I connect the limit switch to the normally closed configuration and send $HZ, there is no movement. I have [...]

This is exactly the problem I am having too. How did you resolve it? Thanks

hanke-cnc, 2024-12-31:
> @hanke-cnc
> 
> > "I noticed that the wiki mentioned the limit switch is a normally closed switch. Now, when I connect the limit switch to the normally closed configuration and send $HZ, there is no movement. I have to manually press the limit switch once, and then it moves away from the limit, returns to seek the limit again, and finally completes homing. It's confusing."
> 
> This is exactly the problem I am having too. How did you resolve it? It almost seems like my homing seek rate is 1 step per second or something silly (it isn't). The motor ticks along very (very) slowly until I manually press the button myself and it carries on perfectly. The speed is not affected by the homing seek rate.
> 
> Thanks

My issue is that I entered an enormous number, 999999, for the maximum travel of the axis.

https://github.com/grblHAL/core/issues/380

jamesbaber, 2024-12-31:
That's solved my problem too!
Thank you very much for getting back to me.